z88dk/z88dk #2007: [question/bug?] How to use 256x208 graphics mode on the Galaksija Plus?

Fabrizio-Caruso, 2022-04-28:
I am trying to compile my games for the Galaksija Plus in 256x208 mode.
I am testing this with the online emulator GALe:
https://galaksija.net/
My programs always crash.
I am following the doc (passing the pragma to set the mode and the pragma to set the font).

I am compiling with:
``
gal_plus_gfx: $(ASSETS_PATH)/z88dk_xchase.asm
	$(Z88DK_PATH)$(MYZ88DK) +gal -subtype=galaxyp $(SCCZ80_OPTS) \
	$(INCLUDE_OPTS) \
	-pragma-need=ansiterminal -pragma-define:cd=1 -pragma-redirect:CRT_FONT=_font_8x8_vga_rom \
	-D_XL_SLOW_DOWN_FACTOR=900 -DXSize=32 -DYSize=26 \
	-D__GAL__ -D__GAL_PLUS__ -DFULL_GAME -DEND_SCREEN -DBETWEEN_LEVEL \
	-DCONIO -DUDG_GRAPHICS -DNO_PRINT -DNO_SLEEP -DNO_WAIT \
	-vn -lndos -create-app -o  $(BUILD_PATH)/X$(GAME_NAME)_galaksija_plus.prg \
	$(FULL_FILES) $(ASSETS_PATH)/z88dk_xchase.asm \
	$(CROSS_LIB_PATH)/display/init_graphics/z88dk/udg/udg_init_graphics.c
	rm $(BUILD_PATH)/X$(GAME_NAME)_galaksija_plus.prg
``

On the other hand, I can compile my games in pure ASCII mode.

I will try to reproduce this issue with a minimal example.

suborb, 2022-05-10:
Any joy with a minimal example? The screenmode has been tested on re-created hardware so it does work.

I think that it should be driven by `--generic-console` rather than `ansiterminal` and the mode needs to be switched on startup.

Fabrizio-Caruso, 2022-05-10:
Sorry, I will try your suggestion and test it with a minimal example.

closed 2022-08-08 by Fabrizio-Caruso