z88dk/z88dk #2008: sound.h

dikdom, 2022-04-30:
I try to figure out the `bit_` sound functions and I'm not sure if they are good for my purpose.
In my target (TVC) there is a sound device: 1 channel with a programmable volume and frequency. It doesn't seem like a one bit sound device. 
If I want to have access to that sound device I think it's better to implement its own methods for volume and pitch, and maybe some sound player. What do you think?

suborb, 2022-05-10:
Yes, it sounds like you're correct - having a separate API would be sensible.

However, you might be able to emulate it by toggling the volume (no idea what the frequency should be) - it may or may not be successful - I think @zx70 did something similar with the sn chip (which sometimes works, sometimes doesn't!)

zx70, 2022-05-11:
This is a recurrent topic sometimes the question is on graphics or display capabilities, sometimes it involves the sound stuff.
On the single machine perspective it is obviously better to have system specific APIs.
By the way the portability opportunity we have on a cross compiler makes very interesting a generic function set.
Moreover @suborb did a great job in grouping the hardware types into common drivers (YM, SN, ..).
Sadly this brings to potentially three or more function sets in parallel.
Personally I'm a portability enthusiast and I tend to focus on the bit banging, but you are both right in telling it is a bit a waste on a better hardware.

Back to my obsession, with graphics the things are even more complex, but also the sound APIs have different use cases to consider where a generic function might help in reusing the code:  a waveform output, music (MIDI, synthetized samples,,), bells and whistles, etc..

zx70, 2022-05-16:
Sorry, the virtual keyboard on my cell phone makes me lazy:  I was referring to the Yamaha PSG.
The 1bit and 4 bit stuff simply stop the program execution to play sounds, and are obviously written in assembler.   A short sound sometimes fits in a game loop, though, and the sound can be pretty good.

dikdom, 2022-05-17:
At last I decided to implement the 1 bit open/close methods and set the constants in the .inc file so I guess the 1bit sound should work - for TVC. The 1bit sound operates with actually setting the volume to 0 and to 15...
The 4 bit sound is 4 bits written sequentially to the 1bit interface, as I understood, maybe I'm wrong.
Additionally I've implemented the native sound generating routines also.

closed 2022-06-15 by zx70