z88dk/z88dk #2013: Appare's wavelength format too..  digital

zx70, 2022-05-10:
It took me a while to understand why the generated waw file weren't working on some emulator. 
The two cases I remember are FUSE and SPECTACULATOR.
I know that 44100 kHz is still a critical rate on other emulators, but in those two cases it was sufficient to use Audacity and export a file in an identical format!
A hex editor showed the difference: Audacity adds a little bit of fuzziness to the square waveform.   I tried to do the same and it solved the problem!
The only explanations I have is that perhaps the emulator converts a steady level into silence or that the tape simulations looks for a bit of roughness to be sure it is time to trigger a toggled level. 
By the way,  I chose not to insert random noise and keep a sort of pattern.   This 'digital noise' works on FUSE and still permits a good compression rate when files are zipped.

suborb, 2022-05-10:
@zx70 whilst you're in the area.... have you seen this one? https://z88dk.org/forum/viewtopic.php?p=20657#p20657

zx70, 2022-05-11:
@suborb, thankyou for the reminding me it,  I read the post and forgot it :pensive:

I found yet another problem on the file size I was declaring in the header, probably connected to the RobertK claim.
The right formula is better explained in abc80.c:
totlen = 12 - 8 + 24 + 8 + numsamp

Massimo Raffaele is testing it, in the older versions he got this message:
[image]

For a more consistent file I also removed the initial blank space which was automatically inserted, I hope it won't be a problem on other emulators.

I'll gather a bit of feedback before closing this topic .

TODO:
To extend the 22khz option to all the targets supporting it (must be tested one by one!)
To move the local wave generators to the centralized functions

zx70, 2022-05-13:
I just got the positive report by Massimo.
We had two annoying elements here, a wrong size definition in the header format and the digital data itself, where I applied a workaround to make it look less 'frozen'.
Thankyou everybody for the feedbacks.

closed 2022-05-13 by zx70