z88dk/z88dk #2016: [MSX](classic):　Incorrect memory configuration in use of expansion slots.

v9938, 2022-05-13:
The memory configuration is incorrect in 'rom.asm'.
The expansion slot processing is missing.
Therefore, when a program is running in the expansion slot, page2(0x8000-0xafff) would not be configured correctly.

The following modifications should be made according to the recommended process for MSX systems.

*.\lib\target\msx\classic\rom.asm*
**Original:**
```
; port fixing; required for ROMs
; port fixing = set the memory configuration, must be first!
    in      a,($A8)
    and     a, $CF
    ld      d,a
    in      a,($A8)
    and     a, $0C
    add     a,a
    add     a,a
    or      d
    out     ($A8),a
```

**Fixed:**
```
; port fixing; required for ROMs
; port fixing = set the memory configuration, must be first!
	call	$0138		;read primary slot #
	rrca				;move it to bit 0,1 of [Acc]
	rrca
	and		$03
	ld 		c,a
	ld		b,0
	ld		hl,$0FCC1	;see if this slot is expanded or not
	add		hl,bc
	ld		c,a			;save primary slot #
	ld		a,(hl)		;get the slot is expanded or not
	and		$80
	or		c			;set MSB if so
	ld		c,a			;save it to [c]
	inc 	hl			;Point to SLTTBL entry
	inc		hl
	inc		hl
	inc		hl
	ld		a,(hl)		;Get what  is  currently  output
						;to expansion   slot    register
	and		$0c
	or		c			;Finaly form slot address
	ld		h,$80
	call	$0024		;enable page 2
```

suborb, 2022-06-05:
Thanks for this - can you recommend an openmsx configuration to try this out on?

v9938, 2022-06-08:
Of course.
Please edit the Slot description in the OpenMSX model file. 
For the example, Slot1 expands.

The problem occurs when the ROM file generated over 16K.
It can be seen in the expansion slots ( except MSX BIOS= 1-3, C-BIOS 1-0 ).

**Example**
File path: 
*.\share\machines\\?????.xml*

Original:
```
    <primary external="true" slot="1"/>    
```

Modify:
```    
    <primary slot="1">
      <secondary external="true" slot="0">
      </secondary>
      <secondary external="true" slot="1">
      </secondary>
      <secondary external="true" slot="2">
      </secondary>
      <secondary external="true" slot="3">
      </secondary>
    </primary>
```

suborb, 2022-09-27:
Apologies, forgot to do add this one after testing it a few months ago...

closed 2022-09-27 by suborb