z88dk/z88dk #2017: z88dk-dis incorrectly disassembles LD (i[xy] + d), n

jameskirkwood, 2022-05-15:
Consider the following `example.asm`:
```
ld (ix+$11),$22
```
Executing `z80asm -b example` produces the following `example.bin` (represented here in hexadecimal):
```
dd 36 11 22
```
Disassembling `example.bin` using `z88dk-dis -mz80 example.bin` produces the following output:
```
                    ld      (ix+$22),$11                    ;[0000] dd 36 11 22
```
Note that the order of the immediate values is different from the original assembly.

suborb, 2022-05-15:
This is a repeat/same as this issue: https://z88dk.org/forum/viewtopic.php?p=19942#p19942

is this with the last release or with nightly?

jameskirkwood, 2022-05-15:
This is the latest release (2.1) via the AUR (https://aur.archlinux.org/packages/z88dk)
Yes that's the same issue.

jameskirkwood, 2022-05-15:
For now, the issue is fixed on the master branch. Thanks for the forum link suborb.

feilipu, 2022-05-16:
Closed by https://github.com/z88dk/z88dk/commit/4173ca385e985011b0b7262e701832babb010376

closed 2022-05-16 by feilipu