z88dk/z88dk #2026: Build fails on Windows/msys2

pauloscustodio, 2022-06-06:
debugger_mi2.c: In function 'debugger_mi2_console_loop':
debugger_mi2.c:1279:16: warning: implicit declaration of function 'getline' [-Wimplicit-function-declaration]
 1279 |         while (getline(&debugger_line, &debugger_line_size, stdin) == 0) ;
      |                ^~~~~~~
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: debugger_mi2.o:C:\msys64\home\T0071173\git\z88dk\src\ticks/debugger_gdb.h:20: multiple definition of `temporary_break'; debugger_gdb.o:C:\msys64\home\T0071173\git\z88dk\src\ticks/debugger_gdb.h:20: first defined here
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: debugger_mi2.o: in function `debugger_mi2_console_loop':
C:\msys64\home\T0071173\git\z88dk\src\ticks/debugger_mi2.c:1279: undefined reference to `getline'

pauloscustodio, 2022-06-06:
Maybe linenoise can be used instead of getline?

desertkun, 2022-06-06:
Does `gets` work too?

desertkun, 2022-06-06:
Docs say it does, can you check now, I've pushed d48005670bb702f5f57fc006c4fddf5fa7a7a546?

pauloscustodio, 2022-06-06:
gets passes, now a different error:  multiple definition of `temporary_break'

cc -o z88dk-gdb.exe -I../../ext/uthash/src/ -g -MMD -I../../ext/UNIXem/include cpu.o backend.o syms.o disassembler_alg.o debug.o exp_engine.o debugger.o breakpoints.o debugger_gdb.o debugger_mi2.o debugger_gdb_packets.o linenoise.o srcfile.o sxmlc.o sxmlsearch.o ../../ext/UNIXem/src/glob.o ../../ext/UNIXem/src/dirent.o ../../ext/UNIXem/src/time.o lex.yy.o expressions.tab.o -lm -lpthread -lws2_32
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: debugger_mi2.o:C:\msys64\home\T0071173\git\z88dk\src\ticks/debugger_gdb.h:20: multiple definition of `temporary_break'; debugger_gdb.o:C:\msys64\home\T0071173\git\z88dk\src\ticks/debugger_gdb.h:20: first defined here
collect2.exe: error: ld returned 1 exit status
make: *** [Makefile:38: z88dk-gdb.exe] Error 1

desertkun, 2022-06-06:
Sorry I don't have win machine at the moment, did 443cb8240619197e1e42de4d8f2dab50d514f748 fix it?

desertkun, 2022-06-06:
Haha didn't mean to cause this trouble, I gotta figure out a way to test on win32 before pushing.

pauloscustodio, 2022-06-06:
yes, all good now

desertkun, 2022-06-08:
Yeah I've pushed a supposed fix 1h ago, that didn't fix? I couldn't figure out how to include that constant on windows.

pauloscustodio, 2022-06-08:
No, latest commit fails.

desertkun, 2022-06-08:
How about `011a30f689c85b1459b0924dae485c7a52bdf8e1`?

suborb, 2022-06-08:
@desertkun if you install mingw via homebrew you can do `make CC=x86_64-w64-mingw32-gcc`

In this case, it's probably worth switching to `gettimeofday()`. However, of course, there's not one by default. There's a version floating around which uses GetFiletime which I've used on other projects.

desertkun, 2022-06-08:
I've resorted to `GetTickCount`, a lower resolution, while it doesn't block. Thanks for the tip, I'll use it for tests from now on!

desertkun, 2022-06-08:
With `make CC=x86_64-w64-mingw32-gcc EXESUFFIX=.exe` ticks does build.