z88dk/z88dk #2035: ticks build fails in Windows with the Microsoft compiler Visual Studio

pauloscustodio, 2022-06-17:
Build started...
1>------ Build started: Project: z88dk-ticks, Configuration: Debug Win32 ------
1>debugger_mi2.c
1>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\stdarg.h(23,1): warning C4005: 'va_copy': macro redefinition
1>C:\msys64\home\paulo\git\z88dk\src\ticks\debugger.h(8): message : see previous definition of 'va_copy'
1>C:\msys64\home\paulo\git\z88dk\src\ticks\debugger_mi2.c(15,10): fatal error C1083: Cannot open include file: 'pthread.h': No such file or directory

desertkun, 2022-06-17:
Ugh, sorry for another problem. 

https://stackoverflow.com/a/28036816

Some windows shenanigans. I'll fix it in few hours.

pauloscustodio, 2022-06-17:
Actually this is not on the MinGW build tools, but in the Microsoft compiler tool chain, which is used in the Visual Studio IDE.
I think you could just disable gdb integration if the system is built by Microsoft's compiler, as gdb will only work in the MinWG environment.

Use #ifdef _MSC_VER to detect the Microsoft compiler.

desertkun, 2022-06-17:
Never realized we even support win32 directly. If that's the case, I would rather get the thing working again instead of cutting stuff off.

suborb, 2022-06-17:
I've successfully used pthreads with visual studio before but I think disabling the gdb integration is probably the right approach.

The binaries in the kits are produced using mingw so that's end users sorted out.

So, the only users of the VS project files are people like Paulo who want to have a nice IDE to develop the binaries with.

desertkun, 2022-06-17:
I see. Not sure how I would do that. Could you please fix this @suborb ?

pauloscustodio, 2022-06-17:
@suborb, I found some articles about this, let me try to sort it out.

pauloscustodio, 2022-06-17:
I managed to build z88dk-gdb on both Visual Studio and MinGW with extensive changes in the Visual Studio project files; I hope I did not break anything

desertkun, 2022-06-17:
Feels like we would benefit from spending some time on extending the nightly build suite to include win32

pauloscustodio, 2022-06-17:
> Feels like we would benefit from spending some time on extending the nightly build suite to include win32

I agree, opened a new issue #2036

desertkun, 2022-06-20:
@pauloscustodio it seems nightly fails now

http://nightly.z88dk.org/report/buildreport-20220620

pauloscustodio, 2022-06-20:
I may have broken something. Let me check tonight.

Regards,
Paulo Custódio

pauloscustodio, 2022-06-21:
Another attempt in solving this submitted; let's wait for tonight's build

pauloscustodio, 2022-06-22:
Nightly still failing http://nightly.z88dk.org/report/buildreport-20220622

Now a result of some new z80asm tests that use a Perl module CPU::Z80::Assembler (I'm the author of this module)

@suborb, may I edit and commit the build script in the build repository?

suborb, 2022-06-22:
I've fixed up the environment now so the build now works.

pauloscustodio, 2022-06-22:
Thanks, you can disregard my pull request then.
Regards,

commit 45ec4cf4 (2022-06-17): Fix #2035 build z88dk-gdb with Visual Studio
Retarget the Visual Studio projects to the latest version.
Include a pthreads-w32-2-9-1 library in binary form.
Fix some warnings.
Include Unix-includes in #ifdef WIN32
Replace structure initializers from {} to {0} - the former are not accepted by the Microsoft compiler
Remove z88dk-map empty project.
Add Visual Studio projects for the missing tools in bin, except z88dk-z80svg that requires libxml2 to compile.
closed 2022-06-22 by pauloscustodio