z88dk/z88dk #2036: Extend github actiosn to also build in native Windows

pauloscustodio, 2022-06-17:
Need to check compilation with the native Microsoft compilers in Windows.

pauloscustodio, 2022-06-18:
@suborb, I am trying to setup a workflow to run a Visual Studio build in github actions and need a Microsoft developed action microsoft/setup-msbuild@v1.1

The build fails with:
microsoft/setup-msbuild@v1.1 is not allowed to be used in z88dk/z88dk. Actions in this workflow must be: within a repository owned by z88dk or created by GitHub.

Apparently the repository admin needs to select the “Allow all actions” option (see https://github.community/t/checkout-v2-error-actions-in-this-workflow-must-within-a-repository-owned-by-organization/136315).

Can you please change the setting?

suborb, 2022-06-18:
@pauloscustodio you already had the power - I'm not too keen to allow anything arbitrary to run so I've added in a wildcard for that one.

pauloscustodio, 2022-06-18:
Sorry, I did not realize I was admin... thanks.

pauloscustodio, 2022-06-21:
Todo:
[ ] solve nightly build failure
[ ] implement workflow to simulate nightly
[ ] try to run tests with code built with msbuild - challenging as no Unix tools are available
[ ] workflow for cygwin compiler - I noticed it fails compilation with msys/gcc

pauloscustodio, 2022-07-02:
@suborb, after a zillion attempts, I finally managed to compile the project in cygwin in a git workflow, but one of the tests is failing:
```
make -C testsuite
make[1]: Entering directory '/cygdrive/d/a/z88dk/z88dk/testsuite'
zcc +test -vn -a 01_cast_global_pointer.c -o 01_cast_global_pointer.opt.tmp1
diff -w 01_cast_global_pointer.opt.tmp2 results/01_cast_global_pointer.opt
16d15
< 	push	bc
make[1]: *** [Makefile:49: 01_cast_global_pointer.opt] Error 1
18d16
< 	pop	bc
22,23d19
< 	push	hl
make: *** [Makefile:224: testsuite] Error 2
< 	pop	bc
30,32c26
< 	add	hl,sp
< 	call	l_gint	;
< 	push	hl
---
> 	call	l_gintspsp	;
make[1]: Leaving directory '/cygdrive/d/a/z88dk/z88dk/testsuite'
```
It passes in my local cygwin environment.
Any ideas of what may be causing the different optimized code output?

pauloscustodio, 2022-07-03:
Tried to reproduce by cloning again and following the exact same steps as in the git workflow and the test passes. In the git workflow it fails. 
What can be causing the difference? Any ideas?

suborb, 2022-07-03:
Normally differences are due to the compiler and CPU issues, but here it just looks like some of the rules haven't fired.

Can you see what else is failing (make -k in testsuite) to see if there's a pattern?

pauloscustodio, 2022-07-03:
Failing test solved! It was due to the incorrect setting of git's core.autocrlf, the source files had CR-LF line endings. Changed to input, and the testsuite passes.
Thanks anyway.

suborb, 2022-07-03:
Grrrr - well done for tracking that one down!

pauloscustodio, 2022-07-06:
Fix https://github.com/z88dk/z88dk/issues/2036 build and test on Windows

Create github workflows that run on push and pull requests for these different cases:
- cross-compiling to Windows on Ubuntu, to simulate the nightly build
- compiling and running the tests on Ubuntu, MacOS, Windows/MinGW and Windows/Cygwin
- compiling and running only the z80asm tests on Visual Studio x86 and x64 (AFAIK the z80asm tests are the only ones that rely only on Perl and are cross-platform)

In the process the following changes were necessary:
- create the different workflow files (obviously)
- work around a bug in UNIXem - you need to include both <unixem/glob.h> and <glob.h> on Windows, but only <glob.h> on Unix or Windows/Cygwin - had to define a new -D_CYGWIN to be passed to the C compiler if the architecture is Windows/Cygwin; this was done in src/Make.common, which is then included in the projects that need glob.h

It was a difficult and long task, and some workflows take more than one hour to run, but I think this will help us detect errors sooner, hopefully before merging to master.

pauloscustodio, 2022-07-24:
Tests are failing after compile; this was not noticed because of the way they they were being called, not propagating the error status.

desertkun, 2022-07-24:
Appreciate the effort; it's already been useful on my PR as it caught a windows error I would have missed otherwise.

desertkun, 2022-07-25:
@pauloscustodio build-on-windows-cygwin job, it seems, takes forever to build the libraries. 
https://github.com/z88dk/z88dk/runs/7496003144?check_suite_focus=true
3 hours. Is than an error? ~30m on other jobs

pauloscustodio, 2022-07-25:
No, not an error, just Cygwin being very slow at file I/O. For people running windows I recommend msys2 instead

pauloscustodio, 2022-07-26:
Extend the build-on-windows-msbuild jobs to build also the Debug configuration, to detect failures in the compiler flags.

commit 2794c1fa (2022-07-05): Fix #2036 build and test on Windows

Create github workflows that run on push and pull requests for these different cases:
- cross-compiling to Windows on Ubuntu, to simulate the nightly build
- compiling and running the tests on Ubuntu, MacOS, Windows/MinGW and Windows/Cygwin
- compiling and running only the z80asm tests on Visual Studio x86 and x64 (AFAIK the z80asm tests are the only ones that rely only on Perl and are cross-platform)

In the process the following changes were necessary:
- create the different workflow files (obviously)
- work around a bug in UNIXem - you need to include bo [... cut]
commit b2b483d0 (2022-07-25): Fix #2036: detect failures in z80asm tests

Tests were failining and errors were not detected.
Use nightly build to copy prebuilt libraries and executables not built
by Visual Studio.
commit 7eae1e71 (2022-07-26): Fix #2036 build Debug and Release configurations
commit 5c9084a8 (2022-07-27): Revert "Fix #2036 build Debug and Release configurations"

This reverts commit 7eae1e716a057aad05183b2f77fc15af7ede5c76.
closed 2022-07-27 by pauloscustodio