z88dk/z88dk #2037: Extend github actions to build in native ARM64

feilipu, 2022-06-18:
Related to #2036, and dependent on https://github.com/actions/virtual-environments/issues/2552 (and others).

Is it worth adding an ARM64 (aarch64) binary distribution?

- Cost?
- Complexity?
- Usefulness?

dbolli, 2022-06-18:
In the near future, when Apple Silicon starts to replace Apple Intel, I think the macOS nightly build and macOS binaries will need to move to Apple Silicon/ARM64. For now, Apple Silicon users are probably happy to compile from nightly build source in my opinion.

feilipu, 2022-06-18:
> For now, Apple Silicon users are probably happy to compile from nightly build source in my opinion.

Perhaps the question is whether a build test should be done to ensure that building from source doesn't become broken?
Or, is it sufficient to rely on an issue being raised when / if building on ARM64 becomes broken at some stage?

feilipu, 2022-06-19:
> Perhaps the question is whether a build test should be done to ensure that building from source doesn't become broken?

So I went back to the original hardware / software. A Clockwork DevTerm A06 to test, and the compile was working error free on Armbian 21.04 which was the original test platform.

But, I've just upgraded the system to Armbian 22.04 and there are now errors appearing in the z80asm tests at the end of the build. Everything else through to the last section of z80asm testing is working as previously.

I'm going to check whether something is missing (caused by packages not properly being upgraded in the transition to 22.04), and then update this comment.

feilipu, 2022-07-09:
Reopen at the right opportunity.

closed 2022-07-09 by feilipu