z88dk/z88dk #2042: GFX: more fixed and improvements

zx70, 2022-06-28:
I spotted a number of issues regarding the graphics functions.
In this record I'm recapping the enhancement areas.

[ ] circle()  in narrow mode fixed
[ ] draw line in narrow mode fixed

zx70, 2022-07-13:
[image]

feilipu, 2022-07-16:
I've added some instructions on how to use `math32` floating point for the relevant graphics demonstrations.

The comparisons between `genmath` and `math32` are shown in below videos on SpecNext hardware.
- [fern with `genmath`](https://youtu.be/8fis7hTsrgE)
- [fern with `math32`](https://youtu.be/RO3pZce0G5k)
- [coswave with `genmath`](https://youtu.be/StUawzW3Q3I)
- [coswave with `math32`](https://youtu.be/DiSp4soCk_A)

zx70, 2022-07-16:
@feilipu, those hints are very useful.
Sometimes I look for them as reminders  :D
... the zx81 math library was the only way to save enough space with the bgi demo

feilipu, 2022-07-16:
Just realised I was looking at `coswave.c` and building `sinwave.c`... doh!
Anyway, the video is still useful.

zx70, 2022-07-17:
In the examples header I try to put a bit of build possibilities. 
Maths, etc seems to be a good hint, please feel free to add  😉

feilipu, 2022-07-21:
Been playing with [savage](https://github.com/z88dk/z88dk/master/examples/embedded/savage.c#L22) and other examples to check performance and accuracy.

The punchline is that when using SDCC there's no advantage to using `math48` floating point, as its additional accuracy is lost in compiler conversion to IEEE standard single precision. Use `math32`. 

When using SCCZ80 for accuracy `math48` is best, but for graphics (placing a pixel onto a screen) `math32` is substantially faster.

zx70, 2022-07-22:
Your videos show the difference in speed very clearly. 
I didn't know that math48 was generally useless with SDCC, it is very interesting. 
I usually don't care much for the FP precision,  but there are cases where the precision matters.
In most of the cases I'm not sure  :D 
https://github.com/z88dk/z88dk-ext/tree/master/misc/fir

Yet z88dk is currently the best kit to address all the cases  ;)

feilipu, 2022-07-23:
I've added a few notes to the [wiki](https://github.com/z88dk/z88dk/wiki/Classic--Maths-Libraries#size-of-floating-point-number) to draw attention to the relative merit of 6 byte vs 4 byte floating point formats, and their use with the two compilers. I hope that helps someone to build a more responsive game.

Just for interest on FIR, I had cause to use it for a [real-time low pass filter](https://github.com/feilipu/Goldilocks_Analogue_DAC_Library) on AVR hardware and did the calculations in fixed point 24.8 format. But again speed was more important than accuracy [for that case](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/examples/GoldilocksAnalogueTestSuite/GoldilocksAnalogueTestSuite.ino#L234).

closed 2022-07-27 by zx70