z88dk/z88dk #2047: [linux] WSL build fails with "Can't locate Modern/Perl.pm in @INC"

Galbi3000, 2022-07-21:
Ok, I have tried this in both Ubuntu 20.04 LTS and 22.04 LTS versions on WSL under Windows 11 Pro Version 22H2 (22622.290). I have checked for this being reported before but the only mention I found was in response to another error where someone said they got my error and found it was due to the permissions of ~/.cpan/ being set to root and not the user however I do not have a .cpan directory in my home directory so it can not be that!

Here are the exact commands I entered:

`wget http://nightly.z88dk.org/z88dk-latest.tgz
tar -zxvf z88dk-latest.tgz
sudo apt install build-essential dos2unix libboost-all-dev texinfo texi2html libxml2-dev subversion bison flex zlib1g-dev m4 libtemplate-perl libtemplate-plugin-yaml-perl libfile-slurp-perl ragel re2c
cd z88dk/
export BUILD_SDCC=1
export BUILD_SDCC_HTTP=1
chmod +x build.sh
./build.sh
`
After a few seconds of building it stops with these lines:

`install -m 755 sccz80 /home/galbi/z88dk/bin/z88dk-sccz80
make[1]: Leaving directory '/home/galbi/z88dk/src/sccz80'
make -C src/z80asm PREFIX=`pwd` PREFIX_SHARE=`pwd` install
make[1]: Entering directory '/home/galbi/z88dk/src/z80asm'
gcc -std=gnu11 -MMD -Wall -O3 -Isrc -Isrc/c -It -I../common -I../../ext/optparse -I../../ext/regex -I../../ext/uthash/src  -g -O2 -c -o src/c/common.o src/c/common.c
cd dev && perl make_reloc_code.pl && dos2unix -q reloc_code.[ch] && \
        mv -f reloc_code.[ch] ../src/c/
Can't locate Modern/Perl.pm in @INC (you may need to install the Modern::Perl module) (@INC contains: /etc/perl /usr/local/lib/x86_64-linux-gnu/perl/5.30.0 /usr/local/share/perl/5.30.0 /usr/lib/x86_64-linux-gnu/perl5/5.30 /usr/share/perl5 /usr/lib/x86_64-linux-gnu/perl/5.30 /usr/share/perl/5.30 /usr/local/lib/site_perl /usr/lib/x86_64-linux-gnu/perl-base) at make_reloc_code.pl line 3.
BEGIN failed--compilation aborted at make_reloc_code.pl line 3.
make[1]: *** [Makefile:140: src/c/reloc_code.c] Error 2
make[1]: Leaving directory '/home/galbi/z88dk/src/z80asm'
make: *** [Makefile:142: bin/z88dk-z80asm] Error 2
`

I should note that I was following instructions from a video that was created in February this year and the guy in the video was using WSL Ubuntu 20.04 LTS and did not run into this issue. He did get another error, the same one as reported in the results looking for my error, but he was expecting that one. Either something was removed from Ubuntu in the months after his video or something was added to the build but not included in the list of dependencies.

So, any ideas how I fix this and get it to build?

desertkun, 2022-07-21:
cpan install Modern::Perl

Galbi3000, 2022-07-21:
Hehe, now I get the same error, different module! I think the instructions need updating! lol

pauloscustodio, 2022-07-21:
Updated  https://github.com/z88dk/z88dk/wiki/installation , Linux/Unix section

closed 2022-07-21 by pauloscustodio

Galbi3000, 2022-07-21:
> Updated https://github.com/z88dk/z88dk/wiki/installation , Linux/Unix section

Thank you, though you might want to put the cpan install in front of the line of modules for people who don't know how to install them :)

pauloscustodio, 2022-07-21:
I've used ```cpanm --local-lib=~/perl5 ``` instead of ```cpan install``` because it allows the installation of the modules in the user's directory without the need for sudo