z88dk/z88dk #2049: z80asm: inconsistent output with -oFILE

pauloscustodio, 2022-07-22:
While porting the z80asm command line parsing code to C++, I detected the currently inconsistent behavior of the ```-oFILE``` option. I think this should be corrected, but scripts that depend on this will break.

1) Code without sections:
1.1) Command without ```-oFILE``` generates ```FILE.bin```, where ```FILE``` is the first object file
1.2) Command with ```-oFILE``` without extension generates ```FILE```
1.3) Command with ```-oFILE.EXT``` generates ```FILE.EXT```
2) Code with a ```CODE``` section at a given ORG:
2.1) Command without ```-oFILE``` generates ```FILE.bin``` and ```FILE_CODE.bin```, where ```FILE``` is the first object file
2.2) Command with ```-oFILE``` without extension generates ```FILE``` and ```FILE_CODE.bin```
2.3) Command with ```-oFILE.EXT``` generates ```FILE.EXT``` and ```FILE_CODE.bin```

My proposed solution is:
- if ```-oFILE``` has no extension, all output files have the ```.bin``` extension
- if ```-oFILE.EXT``` has an extension, all output files have the ```.EXT``` extension

Appreciate your comments.

feilipu, 2022-07-23:
> I detected the currently inconsistent behavior of the `-oFILE` option.

And I thought it was just me... :clown_face:

pauloscustodio, 2022-07-23:
appmake.c:273 and onwards assumes that the files are named file_CODE.bin. 
This with the above change causes the sms example to fail to compile:
```
zcc +sms -vn libpsgtest.c music.asm -o libpsgtest.sms -create-app
sms: ERROR: File libpsgtest_DATA.bin not found for a rom model compile
```

Shall we ignore the extension and always produce ```.bin``` files?

pauloscustodio, 2022-07-24:
No fix, to avoid breaking too many things:
1) ```-oFILE``` generates single binary ```FILE```
2) ```-oFILE.EXT``` generates single binary file ```FILE.EXT```
3) section outputs are always ```FILE_CODE.bin```

We'll keep inconsistent as is today.

commit 0dd1bd1e (2022-07-23): Fix #2049: z80asm: inconsistent output with -oFILE

if -oFILE has no extension, all output files have the .bin extension
if -oFILE.EXT has an extension, all output files have the .EXT extension
commit 424781a7 (2022-07-24): Fix #2049: z80asm: inconsistent output with -oFILE

if -oFILE has no extension, all output files have the .bin extension
if -oFILE.EXT has an extension, all output files have the .EXT extension
closed 2022-07-24 by pauloscustodio