z88dk/z88dk #2057: Site http://nightly.z88dk.org/ is down

pauloscustodio, 2022-07-26:


suborb, 2022-07-26:
Sorted - thanks for the headup.

pauloscustodio, 2022-08-02:
And now up again.

suborb, 2022-08-02:
Apologies, first issue was a power cut and a failed fsck, the second one was unknown - I'm guessing temperature related so I've given it a good dust inside and out.

pauloscustodio, 2022-08-16:
It appears to be down again.

suborb, 2022-08-16:
Well, it's probably not heat related, just crappy Dell hardware it seems. I guess I need to find some old/new hardware to replace it.

Anyway, I've toggled the remote power switch and it should be back up.

pauloscustodio, 2022-08-22:
Down again, detected because our github workflows download from it and fail.

feilipu, 2022-09-02:
> Down again, detected because our github workflows download from it and fail.

Seems still down, as it was last night. Causing build failures.

closed 2022-11-04 by feilipu