z88dk/z88dk #2058: Linker Scripts?

RetroRevivalGames, 2022-07-27:
Hello there,

I was just wondering if z88dk supports anything like linker scripts to organize my binaries.

Thank you!

RetroRevivalGames, 2022-07-27:
And for that matter, is there anything like gcc's -ffunction-sections and --gc-sections to eliminate dead code?

feilipu, 2022-07-28:
`zcc` and `z80asm` support list files where source files and build options are listed (one per line). They can work out what needs to happen from that.

`z80asm` only links necessary objects, but it is limited to the resolution of a file. There are old issues to fix this, but it is not critical.

Because of this libraries are usually written one function per file. You can see this reflected in all the code, and I guess everyone is used to that.

closed 2022-07-28 by RetroRevivalGames