z88dk/z88dk #2059: zcc - -v too verbose

feilipu, 2022-07-28:
I'm not sure if I missed when it happened, but `zcc` has started to spit the whole listing of assembled files in its verbose output.
Removing the `-v` option removes the listing, but it also removes the information about what `zcc` is doing.

Is it possible to revert to `-v` producing just information about `zcc` actions?

Or.. have I something broken somewhere?

eg.
`zcc +rc2014 -subtype=hbios -clib=sdcc_iy -SO3 -v -m --list --max-allocs-per-node100000 -llib/hbios/time -llib/hbios/diskio_hbios -llib/hbios/ff fileCopyTest.c -o fileCopyTest -create-app`

produces...

```
$ zcc +rc2014 -subtype=hbios -clib=sdcc_iy -SO3 -v -m --list --max-allocs-per-node100000 -llib/hbios/time -llib/hbios/diskio_hbios -llib/hbios/ff fileCopyTest.c -o fileCopyTest -create-app

PROCESSING fileCopyTest.c
z88dk-zsdcpp -iquote"." -D__Z88DK -D__RC2014 -D__Z80 -D__SDCC -D__SDCC_IY -isystem"/home/phillip/Z80/z88dk/lib/config/../..//include/_DEVELOPMENT/sdcc"  -D__SDCC  "fileCopyTest.c" "/tmp/tmpXXp1F5R6.i2"
fileCopyTest.c:19:2: warning: #warning No timer calculation possible.
z88dk-zpragma -zcc-opt="/tmp/tmpzccXXulSOA3/zcc_opt.def" < "/tmp/tmpXXp1F5R6.i2" > "/tmp/tmpXXp1F5R6.i"
z88dk-zsdcc --constseg rodata_compiler   --max-allocs-per-node100000 --allow-unsafe-read --reserve-regs-iy -mz80 --no-optsdcc-in-asm --c1mode --emit-externs  --no-c-code-in-asm   --no-peep --peep-file "/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/sdcc_peeph.3"  < "/tmp/tmpXXp1F5R6.i" -o "/tmp/tmpXXp1F5R6.opt"
z88dk-copt  "/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/sdcc_opt.1" < "/tmp/tmpXXp1F5R6.op1" > "/tmp/tmpXXp1F5R6.opt"
z88dk-copt  "/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/sdcc_opt.9" < "/tmp/tmpXXp1F5R6.opt" > "/tmp/tmpXXp1F5R6.op1"
z88dk-copt  "/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/sdcc_opt.2" < "/tmp/tmpXXp1F5R6.op1" > "/tmp/tmpXXp1F5R6.asm"
cat "/tmp/tmpXXp1F5R6.asm" >> "/tmp/tmpXXp1F5R6.tmp"
cat "/tmp/tmpXXp1F5R6.tmp" >> "/tmp/tmpXXp1F5R6.asm"
z88dk-z80asm -D__SDCC_IY -I"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/rc2014" -I"/tmp/tmpzccXXulSOA3" -D__SDCC -l   -s  -mz80 -I"/home/phillip/Z80/z88dk/lib/config/../..//lib" "/tmp/tmpXXp1F5R6.asm"

PROCESSING /home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/rc2014/rc2014_crt.asm.m4
m4 -I "/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/rc2014" -I "/tmp/tmpzccXXulSOA3" -I "/home/phillip/Z80/z88dk/lib/config/../..//src/m4"  --define=__STARTUP=8  < "/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/rc2014/rc2014_crt.asm.m4" > "/tmp/tmpXX1QiKD2.asm"
cat "/tmp/tmpXX1QiKD2.asm" >> "/tmp/tmpXX1QiKD2.tmp"
cat "/tmp/tmpXX1QiKD2.tmp" >> "/tmp/tmpXX1QiKD2.asm"
WILL ACT AS CRT

GENERATING OUTPUT
/tmp/tmpXXp1F5R6.asm:
     1                          MODULE fileCopyTest_c
     2                          LINE 0, "fileCopyTest.c"
fileCopyTest.c:
                                
     1                          ;--------------------------------------------------------
     2                          ; File Created by SDCC : free open source ANSI-C Compiler
     3                          ; Version 4.2.0 #13131 (Linux)
     4                          ;--------------------------------------------------------
     5                          ; Processed by Z88DK
     6                          ;--------------------------------------------------------
     7                          
     8                          	EXTERN __divschar
     9                          	EXTERN __divschar_callee
    10                          	EXTERN __divsint
    11                          	EXTERN __divsint_callee
    12                          	EXTERN __divslong
    13                          	EXTERN __divslong_callee

--- clip out a very long asm listing that fills my console history with junk ---

  1430   0002b5 7263ff2575fb4652	DEFM "rc=%u FR_%s"
                f82573          
  1431   0002c0 0d              	DEFB 0x0d
  1432   0002c1 0a              	DEFB 0x0a
  1433   0002c2 00              	DEFB 0x00
  1434                          	SECTION IGNORE
  1435                          
cat "/tmp/tmpXXp1F5R6.lis" >> "fileCopyTest.c.lis"

PROCESSING CRT
z88dk-z80asm -D__SDCC_IY -I"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/rc2014" -I"/tmp/tmpzccXXulSOA3" -D__SDCC -l   -s  -mz80 -I"/home/phillip/Z80/z88dk/lib/config/../..//lib" "/tmp/tmpXX1QiKD2.asm"

z88dk-z80asm -mz80 -b -d  -o"fileCopyTest" -m -s -L.  -L"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/lib/sdcc_iy"   -D__SDCC_IY -I"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/rc2014" -I"/tmp/tmpzccXXulSOA3" -D__SDCC -l  -llib/hbios/time -llib/hbios/diskio_hbios -llib/hbios/ff -lrc2014    "/tmp/tmpXX1QiKD2.o" "/tmp/tmpXXp1F5R6.o"
cat "/tmp/tmpXX1QiKD2.lis" >> "fileCopyTest.lis"
z88dk-appmake +rom --warn --ihex  -b "fileCopyTest" -c "fileCopyTest"
```

feilipu, 2022-07-28:
@desertkun do you need this verbosity to appear on the console, when it is already piped to a file?

For me, I'd prefer to see the "workings" not disguised by the "output".

```

                if (verbose) {
                    FILE* f = fopen(ptr, "r");
                    if (f) {
                        static char buf[1024];
                        unsigned long nread;
                        while ((nread = fread(buf, 1, sizeof(buf), f)) > 0)
                            fwrite(buf, 1, nread, stdout);
                        fclose(f);
                    }
                }

```

pauloscustodio, 2022-07-28:
I understand the need; when zcc finishes it deletes all temporary files, eliminating the option to debug them.

For this reason I've re-added the zcc ```-no-cleanup``` option that keeps all the temporary files after exit. This way the code to dump the generated source is not needed.

feilipu, 2022-07-28:
But, aren't those files available with `--list` anyway?

Certainly the assembly files are available. But perhaps not the intermediate files?

pauloscustodio, 2022-07-28:
Yes, the ```-no-cleanup``` option keeps the intermediate files.

desertkun, 2022-07-28:
Hope fc30d125b44196190a195bb94b81a84e640e6bca settles it. I only needed that with combination on -E -dD to mimic gcc behaviour, so IDEs (CLion in that case) would recognise gcc when processing CMake project.

feilipu, 2022-07-28:
Thank you. Works for me.

closed 2022-07-28 by feilipu