z88dk/z88dk #2063: Bad parameter splitting on CP/M

zx70, 2022-08-05:
From time to time I pick an working program written in C and try to run it on z88dk.
In this way I have a simple and funny way to test z88dk and spot those hidden bugs which are difficult to find by writing test programs from scratch (which would obviously be limited by my skills and programming "stile", or rather habits).

This one shows a limit in the command line paramenters parsing:
https://github.com/z88dk/z88dk-ext/blob/master/os-related/change.c

The program works but both the "<infile", ">outfile" file redirections need to be used to get the program reacting as expected.
The original CP/M program does the right thing, as well as a GCC compiled version.

To be honest, after the /lib folder sorting I tried to adjust it a couple of times already, but the TRS80 way to parse the command line, the CP/M and the OSCA ones now depend one each other and in my past attempts I broke something.

zx70, 2022-08-18:
Yet on CP/M, I think that the __bdos() workaround is not necessary, IX is saved also on the original bdos() call.
I'd like to update it and insert also the __callee opt but the build tree is slightly messy.

feilipu, 2026-11-08:
Fixed in 72b1c6ee91 (`(crt0) Improve command line parsing a little`).

`cpm_crt0.asm` NUL-terminates the DMA tail, and `crt_command_line.inc` drops each redirect from the argument count. A line with both a redirect and several arguments now has the right `argc`.

closed 2026-11-08 by feilipu