z88dk/z88dk #2064: [bug][Aquarius] On real hardware 3 out of 5 of my Aquarius games are broken

Fabrizio-Caruso, 2022-08-07:
Roy Templeman (@LoudScotsBloke) has tested recent versions of my Cross-Lib games 
(compiled with ZCC v19435-c8d2246b6-20220311) on his real Mattel Aquarius (S2 release).

3 out of 5 either don't load or misbehave (random characters on the screen):
- XSnake: does not load 
- XShoot crashes back to OK prompt.
- XBomber: corrupted chars while firing a bomb

The very same .caq files are correctly loaded and run inside both Virtual Aquarius and Aqualite emulators.

Needless to say, the game code cannot be the culprit as these games run fine on all other targets including the native PC. 

I am suspecting a stack issue. What should I do?

I am attaching some of the .caq files.
[XBomber_random_chars_when_fire.zip](https://github.com/z88dk/z88dk/files/9277628/XBomber_random_chars_when_fire.zip)
[XSnake_does_not_load_on_real_hardware.zip](https://github.com/z88dk/z88dk/files/9277635/XSnake_does_not_load_on_real_hardware.zip)

[image]
[image]
[image]

suborb, 2022-08-29:
What's interesting here is that the binary sizes of faulty programs are quite different, 8k and 16k so it's not a simple "program too big issue"

You can play around with the stack point by using `-pragma-define:REGISTER_SP=NNNN`

* `REGISTER_SP=-10000` -> `ld sp,(10000)`
* `REGISTER_SP=10000` -> `ld sp,10000`

The default on the aquarius is `-1` which means don't change it.

Fabrizio-Caruso, 2022-08-29:
Maybe it is related to CONIO/VT100.

pjshumphreys, 2022-09-21:
Are you able to do a memory test on your Aquarius? Maybe is could have bad ram somewhere in the address space? Also Doesn't the Aquarius have only 4K or ram by default? So for games that are 8k or 16k you'd need a ram expansion or put them on rom (at 0xc000 and up perhaps) right?

Fabrizio-Caruso, 2022-09-21:
@pjshumphreys I am not running the tests myself. Some else has done the test on his machines. His computers work fine. The problem is in Z88DK.

Fabrizio-Caruso, 2022-09-21:
@zx70 this issue may be related to CONIO as a problem with it is also mentioned in the Z88DK documentation.

suborb, 2023-07-08:
@Fabrizio-Caruso Did this one get sorted?

Fabrizio-Caruso, 2023-07-08:
I need someone with the real hardware to tell us...
@pjshumphreys  would you please re-test new versions of my games? There is a two more games for a total of 7.

@suborb My guess is still that CONIO is broken. In Cross-Lib I have implemented several ways to do graphics. For this target I can choose between CONIO, z88dk sprites (mono and low res) and "memory mapped" (same resolution and colors as CONIO but implemented with POKE's screen ram). I suspect that if I build my games with "memory mapped" graphics, they would work on the real hardware.

suborb, 2023-07-08:
All conio does is poking as well, so it's weird if it doesn't work.

```
; c = x
; b = y
; a = character to print
; e = raw
generic_console_printc:
        call    xypos
        ld      (hl),a
        inc     h
        inc     h
        inc     h
        inc     h
        ld      a,(__aquarius_attr)
        ld      (hl),a
        ret

xypos:
        ld      hl,DISPLAY - CONSOLE_COLUMNS
        ld      de,CONSOLE_COLUMNS
        inc     b
generic_console_printc_1:
        add     hl,de
        djnz    generic_console_printc_1
generic_console_printc_3:
        add     hl,bc                   ;hl now points to address in display
        ret

```

Fabrizio-Caruso, 2023-07-10:
I thought CONIO was relying on rom routines and the issue observed by @LoudScotsBloke on the real machines was related to this:
[image]

zx70, 2023-07-14:
The code I pulled was meant to deal with tape formats

zx70, 2023-08-18:
@LoudScotsBloke, this is the opien topic I was referring to.
If you all don't mind I'll try to revive it.
My last update in the Aquarius world was on the loader/packager.
The most interesting improvement was probably the direct support of the WAV format (including the usual tuning tricks like a TURBO option and the 22khz mode).   Also the memory position of the loader should have been tuned a little IIRC.
I remember I saw an article online explaining how to deal with two different Aquarius versions, I'm not sure I addressed such difference already..     @Fabrizio-Caruso, could you focus on a program which was failing and try to rebuild it, basing on appmake to generate the final wave file ?

Fabrizio-Caruso, 2023-08-19:
@zx70 
I can rebuild the 3 games that had issues:

    XSnake: does not load
    XShoot crashes back to OK prompt.
    XBomber: corrupted chars while firing a bomb

possibly without using CONIO.

Fabrizio-Caruso, 2023-08-19:
In the attached zip file you will find:
 - bomber (+16k is safe but +4k may work)
 - snake (+16k minimum)
 - shoot (+32k required)
in 2 versions:
 - standard (compiled using CONIO)
 - alt (compiled using my own code)
 
 In particular I use Cross-Lib makefile:
 https://github.com/Fabrizio-Caruso/CROSS-LIB/blob/master/src/Makefile_common
 (makefile targets used are: aquarius_exp_16k for CONIO and aquarius_alt for non-CONIO)

With Cross-Lib scripts these binaries are build by the following commands:
```
xl <game_name> aquarius
xl <game_name> aquarius_alt
```

[aquarius_binaries_to_test.zip](https://github.com/z88dk/z88dk/files/12385025/aquarius_binaries_to_test.zip)

zx70, 2023-08-25:
I understand the problem is not related to the ROM version because the loader would otherwise fail, but I'm pasting the link to an interesting article just not to loose it again

https://forums.atariage.com/topic/338765-aquarius-the-inufuto-collection-s1-s2-and-micro-expander-compatible-pack-inside/