z88dk/z88dk #2065: Compilation error - windows.h: No such file or directory

waylon531, 2022-08-08:
Hi, I'm having a bit of trouble getting z88dk compiled on my laptop. For some reason everything compiled fine on my desktop, but I'm getting an error now.

```
make[1]: Entering directory '/home/uelen/build/z88dk/src/z88dk-lib'
ccache gcc -g -O2   -c -o z88dk-lib.o z88dk-lib.c
In file included from z88dk-lib.c:19:
./dirent.h:20:10: fatal error: windows.h: No such file or directory
   20 | #include <windows.h>
      |          ^~~~~~~~~~~
compilation terminated.
```

I'm not sure if I'm missing a library or something. I think I have every library installed from the docs, so maybe the docs need updating? I'm on linux, I could use a pointer to the right library to grab. Is it windows.h from mingw?

pauloscustodio, 2022-08-08:
It seems you have somehow WIN32 defined.
Could you check the output of ```gcc -E z88dk-lib.c```?

waylon531, 2022-08-09:
Here's a snippet:

```
# 251 "/usr/include/ctype.h" 3 4
extern int isalnum_l (int, locale_t) __attribute__ ((__nothrow__ , __leaf__));
extern int isalpha_l (int, locale_t) __attribute__ ((__nothrow__ , __leaf__));
extern int iscntrl_l (int, locale_t) __attribute__ ((__nothrow__ , __leaf__));
extern int isdigit_l (int, locale_t) __attribute__ ((__nothrow__ , __leaf__));
extern int islower_l (int, locale_t) __attribute__ ((__nothrow__ , __leaf__));
extern int isgraph_l (int, locale_t) __attribute__ ((__nothrow__ , __leaf__));
extern int isprint_l (int, locale_t) __attribute__ ((__nothrow__ , __leaf__));
extern int ispunct_l (int, locale_t) __attribute__ ((__nothrow__ , __leaf__));
extern int isspace_l (int, locale_t) __attribute__ ((__nothrow__ , __leaf__));
extern int isupper_l (int, locale_t) __attribute__ ((__nothrow__ , __leaf__));
extern int isxdigit_l (int, locale_t) __attribute__ ((__nothrow__ , __leaf__));

extern int isblank_l (int, locale_t) __attribute__ ((__nothrow__ , __leaf__));

extern int __tolower_l (int __c, locale_t __l) __attribute__ ((__nothrow__ , __leaf__));
extern int tolower_l (int __c, locale_t __l) __attribute__ ((__nothrow__ , __leaf__));

extern int __toupper_l (int __c, locale_t __l) __attribute__ ((__nothrow__ , __leaf__));
extern int toupper_l (int __c, locale_t __l) __attribute__ ((__nothrow__ , __leaf__));
# 327 "/usr/include/ctype.h" 3 4

# 8 "z88dk-lib.c" 2
# 19 "z88dk-lib.c"
# 1 "./dirent.h" 1 3
In file included from z88dk-lib.c:19:
./dirent.h:20:10: fatal error: windows.h: No such file or directory
   20 | #include <windows.h>
      |          ^~~~~~~~~~~
compilation terminated.
```

I tried commenting out `<windows.h>` from dirent.h, but I get a ton of compilation errors when I do. The code is definitely using the stuff from that header file.

If WIN32 is an environment variable it isn't set, I get an empty string back when I echo it:
```
 $ echo $WIN32

```

pauloscustodio, 2022-08-09:
Problem cause found, solution under way.

commit 278b9ab9 (2022-08-09): Fix #2065: gcc including local dirent.h

From the output of gcc -E in a user's report, the line "#include <dirent.h>" is including the local "dirent.h" instead the system one, causing compilation errors in Linux.

Moved the local dirent.h to a subdirectory win32 and rever to it in the include statement.
closed 2022-08-09 by pauloscustodio