z88dk/z88dk #2066: [bug][windows] Windows binary cannot locat dynamic link library

Fabrizio-Caruso, 2022-08-09:
With the latest snapshot I can no longer compile anything under Windows/Cygwin.
I get an error about not being able to find iteratordeEv.

It is trivially reproduced by running the Windows version of "zcc" under Cygwin on any C file, for example:
```
zcc +ace test.c
```

[image]

pauloscustodio, 2022-08-10:
Do you have g++ installed in Cygwin? The referred entry point is from the C++ std::filesystem library.

pauloscustodio, 2022-08-10:
You need the following Cygwin packages:
gcc-g++ bison flex libxml2-devel libiconv-devel zlib-devel liblzma-devel wget unzip dos2unix autoconf automake libtool ragel perl perl-Regexp-Common perl-Test-Harness perl-Text-Diff perl-YAML-Tiny perl-Module-Build perl-List-MoreUtils perl-Params-Util Perl-B-COW Perl-Clone

Fabrizio-Caruso, 2022-08-10:
@pauloscustodio this is not a Cygwin issue. I get the same error under Windows if I use the the standard Windows command line.

Fabrizio-Caruso, 2022-08-10:
@pauloscustodio This is an issue about the Windows snapshot, built for Windows. The problem appears under Windows without Cygwin and also appears under Cygwin. Previous versions still work correctly under Windows.

Fabrizio-Caruso, 2022-08-10:
@pauloscustodio I am removing Cygwin from the description of the problem. I have double checked this on a different computer. The Windows version of Z88DK is no longer usable.

pauloscustodio, 2022-08-10:
@suborb, maybe we need to tell gcc in the build snapshot script to link staticaly? See https://stackoverflow.com/questions/6578484/telling-gcc-directly-to-link-a-library-statically

suborb, 2022-08-10:
There’s meant to be a c++ dll in the kit.

pauloscustodio, 2022-08-10:
Depends.exe outputs: 
```
Error: At least one required implicit or forwarded dependency was not found.
Error: At least one module has an unresolved import due to a missing export function in an implicitly dependent module.
Warning: At least one delay-load dependency module was not found.
```

On LIBSTDC++-6.DLL all the std::filesystem functions are undefined. std::filesystem was introduced in C++17. Maybe we need a newer version of LIBSTDC++-6.DLL?

pauloscustodio, 2022-08-11:
I've checked the released version 2.2, and it is OK. The error was introduced in commit 8a0f54f8a3e4ee06816a0c43ad9dd865ad3f80be, just after the release of 2.2, when I ported the command line parsing of z80asm to C++, and starting using std::filesystem from C++17.

Fabrizio-Caruso, 2022-08-11:
Once this issue has been solved, would it be possible to have a sort of automatic test of the dependencies in the continuous integration?

suborb, 2022-08-11:
@pauloscustodio I think it needs a static link to the fs library, all that's on the system is the following (no libstdc++7 on the machine):

```
./gcc/x86_64-w64-mingw32/9.3-posix/libstdc++fs.a
./gcc/x86_64-w64-mingw32/9.3-posix/libstdc++-6.dll
./gcc/x86_64-w64-mingw32/9.3-posix/libstdc++.dll.a
./gcc/x86_64-w64-mingw32/9.3-posix/libstdc++.dll.a-gdb.py
./gcc/x86_64-w64-mingw32/9.3-posix/libstdc++.a
./gcc/x86_64-w64-mingw32/9.3-win32/libstdc++fs.a
./gcc/x86_64-w64-mingw32/9.3-win32/libstdc++-6.dll
./gcc/x86_64-w64-mingw32/9.3-win32/libstdc++.dll.a
./gcc/x86_64-w64-mingw32/9.3-win32/libstdc++.dll.a-gdb.py
./gcc/x86_64-w64-mingw32/9.3-win32/libstdc++.a
./gcc/i686-w64-mingw32/9.3-posix/libstdc++fs.a
./gcc/i686-w64-mingw32/9.3-posix/libstdc++-6.dll
./gcc/i686-w64-mingw32/9.3-posix/libstdc++.dll.a
./gcc/i686-w64-mingw32/9.3-posix/libstdc++.dll.a-gdb.py
./gcc/i686-w64-mingw32/9.3-posix/libstdc++.a
./gcc/i686-w64-mingw32/9.3-win32/libstdc++fs.a
./gcc/i686-w64-mingw32/9.3-win32/libstdc++-6.dll
./gcc/i686-w64-mingw32/9.3-win32/libstdc++.dll.a
./gcc/i686-w64-mingw32/9.3-win32/libstdc++.dll.a-gdb.py
./gcc/i686-w64-mingw32/9.3-win32/libstdc++.a
```

pauloscustodio, 2022-08-12:
Added fix to llink libc and libc++ statically when cross-compiling so that we can use C++17 features on older systems.
Also added cron job to github workflows to test the win32 daily build at 6 AM every day.

Let's see if tomorrow the problem is solved.

Fabrizio-Caruso, 2022-08-13:
Thanks @pauloscustodio and @suborb ! The issue is fixed for me.

pauloscustodio, 2022-08-24:
z88dk-win32-20220824-7db629d7e-19857\z88dk\bin\z88dk-z80asm.exe fails with error:
[image]

pauloscustodio, 2022-08-25:
Solved by linking libc and libc++ statically (https://github.com/z88dk/z88dk/commit/a30b0b4f849a4244c5da9dd784808b761406e9d3)

pauloscustodio, 2023-03-17:
I have removed the static libraries needed to include std::experimental::filesystem, and included boost::filesystem instead.
To link with Boost, ```-lboostfilesystem-mt``` is needed in ```msys2```, I don't know if the same library name is used in other platforms. If yes, ```src/z80asm/build_ldflags.pl``` may need some tweaking

closed 2023-03-17 by pauloscustodio

suborb, 2023-03-17:
@pauloscustodio I think you may have reintroduced the original problem?

I thought the static link on windows was there so that the windows binary (built from nightly) could run on older mingw installations?

pauloscustodio, 2023-03-19:
I have missed this message and saw in your changes in the build script; they seem fine. Let's wait for tonight's nightly

suborb, 2023-03-20:
Sorry, I didn't meant to check that in!

I think we now solve for 1,3,4 case 2 is solved for Windows, but not for macOS.

For macOS, and nightly (i.e. CROSS=1) I think we need to force using boost regardless of the autodetection (this PR https://github.com/z88dk/z88dk/pull/2205 but adjusted to the perl script)

commit a8084a02 (2023-03-17): Fix #2066: Use boost::filsystem as fallback to std::filesystem from C++17
commit c7b0982c (2023-03-17): Fix #2066: Use boost::filesystem as fallback to std::filesystem from C++17