z88dk/z88dk #2068: z80asm - directory traversal / reference for crt0 for rc2014-8085 startups

feilipu, 2022-08-15:
Something has changed in the directory traversal mechanism of z80asm which is causing the rc2014_8085 target to be broken.

The rc2014_8085 target is a hybrid, and has most of its code in new library, but some in classic.
In the startup 2 and 32 (both for 8085), there are references across the directory structure to [`crt_init_amalloc.asm`](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/rc2014/startup/rc2014_crt_2.asm.m4#L273) and [`crt_runtime_selection.asm`](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/rc2014/startup/rc2014_crt_2.asm.m4#L374).

The directory traversal is no longer working as it was, so these files are not being found.

Perhaps the directory from which the script is being executed is no longer the same?
But that doesn't explain it as the other relative reference are still working.
And the `crt_init_malloc` and `crt_runtime_selection` files haven't moved either.

Anyway, there have been no other changes to the startup code since this was working.

The below [build of CP/M-IDE for 8085](https://github.com/RC2014Z80/RC2014/tree/master/ROMs/CPM-IDE#building-software-from-source) shows the issue.

``` sh
zcc +rc2014 -subtype=acia85 -O2 --opt-code-speed=inlineints -m -D__CLASSIC -DAMALLOC -l_DEVELOPMENT/lib/sccz80/lib/rc2014/ff_85_ro @cpm22.lst -o ../rc2014-acia85-cpm22 -create-app
```

``` sh
PROCESSING cpm22preamble.asm
cat "cpm22preamble.asm" >> "/tmp/tmpXXFmRoTf.asm"
cat "/tmp/tmpXXFmRoTf.asm" >> "/tmp/tmpXXFmRoTf.tmp"
cat "/tmp/tmpXXFmRoTf.tmp" >> "/tmp/tmpXXFmRoTf.asm"
z88dk-z80asm -I"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/rc2014" -I"/tmp/tmpzccXX6RPAvf" -D__SCCZ80 -l   -s  -m8085 -I"/home/phillip/Z80/z88dk/lib/config/../..//lib" -I"/home/phillip/Z80/RC2014/ROMs/CPM-IDE/acia85"  "/tmp/tmpXXFmRoTf.asm"

PROCESSING cpm22bios.asm
cat "cpm22bios.asm" >> "/tmp/tmpXXEjcDij.asm"
cat "/tmp/tmpXXEjcDij.asm" >> "/tmp/tmpXXEjcDij.tmp"
cat "/tmp/tmpXXEjcDij.tmp" >> "/tmp/tmpXXEjcDij.asm"
z88dk-z80asm -I"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/rc2014" -I"/tmp/tmpzccXX6RPAvf" -D__SCCZ80 -l   -s  -m8085 -I"/home/phillip/Z80/z88dk/lib/config/../..//lib" -I"/home/phillip/Z80/RC2014/ROMs/CPM-IDE/acia85"  "/tmp/tmpXXEjcDij.asm"

PROCESSING cpm22.asm
cat "cpm22.asm" >> "/tmp/tmpXXf27x9i.asm"
cat "/tmp/tmpXXf27x9i.asm" >> "/tmp/tmpXXf27x9i.tmp"
cat "/tmp/tmpXXf27x9i.tmp" >> "/tmp/tmpXXf27x9i.asm"
z88dk-z80asm -I"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/rc2014" -I"/tmp/tmpzccXX6RPAvf" -D__SCCZ80 -l   -s  -m8085 -I"/home/phillip/Z80/z88dk/lib/config/../..//lib" -I"/home/phillip/Z80/RC2014/ROMs/CPM-IDE/acia85"  "/tmp/tmpXXf27x9i.asm"

PROCESSING main.c
z88dk-ucpp -iquote"." -D__Z88DK -D__RC2014 -D__Z80 -D__CLASSIC -DAMALLOC -D__SCCZ80 -D__8080__ -D__8085__ -isystem"/home/phillip/Z80/z88dk/lib/config/../..//include"  -DSCCZ80 -DSMALL_C -D__SCCZ80 -isystem"/home/phillip/Z80/z88dk/lib/config/../..//include"    "main.c" "/tmp/tmpXXj63Sih.i2"
z88dk-zpragma -sccz80 -zcc-opt="/tmp/tmpzccXX6RPAvf/zcc_opt.def" < "/tmp/tmpXXj63Sih.i2" > "/tmp/tmpXXj63Sih.i"
z88dk-sccz80  -ext=opt -m8085 -zcc-opt="/tmp/tmpzccXX6RPAvf/zcc_opt.def" --opt-code-speed=inlineints -standard-escape-chars   "/tmp/tmpXXj63Sih.i" -o "/tmp/tmpXXj63Sih.opt"
z88dk-copt -m8085 "/home/phillip/Z80/z88dk/lib/config/../..//lib/z80rules.9" < "/tmp/tmpXXj63Sih.opt" > "/tmp/tmpXXj63Sih.op1"
z88dk-copt -m8085 "/home/phillip/Z80/z88dk/lib/config/../..//lib/z80rules.2" < "/tmp/tmpXXj63Sih.op1" > "/tmp/tmpXXj63Sih.opt"
z88dk-copt -m8085 "/home/phillip/Z80/z88dk/lib/config/../..//lib/z80rules.1" < "/tmp/tmpXXj63Sih.opt" > "/tmp/tmpXXj63Sih.op1"
z88dk-copt -m8085 "/home/phillip/Z80/z88dk/lib/config/../..//lib/z80rules.8" < "/tmp/tmpXXj63Sih.op1" > "/tmp/tmpXXj63Sih.opt"
z88dk-copt -m8085 /home/phillip/Z80/z88dk/lib/config/../..//lib/arch/8085/8085_rules.1 < "/tmp/tmpXXj63Sih.opt" > "/tmp/tmpXXj63Sih.asm"
cat "/tmp/tmpXXj63Sih.asm" >> "/tmp/tmpXXj63Sih.tmp"
cat "/tmp/tmpXXj63Sih.tmp" >> "/tmp/tmpXXj63Sih.asm"
z88dk-z80asm -I"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/rc2014" -I"/tmp/tmpzccXX6RPAvf" -D__SCCZ80 -l   -s  -m8085 -I"/home/phillip/Z80/z88dk/lib/config/../..//lib" "/tmp/tmpXXj63Sih.asm"

PROCESSING /home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/rc2014/rc2014_crt.asm.m4
m4 -I "/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/rc2014" -I "/tmp/tmpzccXX6RPAvf" -I "/home/phillip/Z80/z88dk/lib/config/../..//src/m4"  --define=__STARTUP=2  < "/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/rc2014/rc2014_crt.asm.m4" > "/tmp/tmpXXVxpVTg.asm"
cat "/tmp/tmpXXVxpVTg.asm" >> "/tmp/tmpXXVxpVTg.tmp"
cat "/tmp/tmpXXVxpVTg.tmp" >> "/tmp/tmpXXVxpVTg.asm"
WILL ACT AS CRT

GENERATING OUTPUT
cat "/tmp/tmpXXFmRoTf.lis" >> "cpm22preamble.asm.lis"
cat "/tmp/tmpXXEjcDij.lis" >> "cpm22bios.asm.lis"
cat "/tmp/tmpXXf27x9i.lis" >> "cpm22.asm.lis"
cat "/tmp/tmpXXj63Sih.lis" >> "main.c.lis"

PROCESSING CRT
z88dk-z80asm -I"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/rc2014" -I"/tmp/tmpzccXX6RPAvf" -D__SCCZ80 -l   -s  -m8085 -I"/home/phillip/Z80/z88dk/lib/config/../..//lib" "/tmp/tmpXXVxpVTg.asm"
/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/rc2014/rc2014_crt.asm.m4:1181: error: file open: ../../../../../lib/crt/classic/crt_init_amalloc.asm
  ^---- include "../../../../../lib/crt/classic/crt_init_amalloc.asm"
/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/rc2014/rc2014_crt.asm.m4:1282: error: file open: ../../../../../lib/crt/classic/crt_runtime_selection.asm
  ^---- include "../../../../../lib/crt/classic/crt_runtime_selection.asm"
phillip@SilverQuad:~/Z80/RC2014/ROMs/CPM-IDE/acia85$ 
```

feilipu, 2022-08-15:
Resolved by changing the traversal depth.

pauloscustodio, 2022-08-15:
The directory transversal was changed right after version 2.2 by porting to C++ and using std::filesystem instead of POSIX glob(). Did this cause a change in behavior?

feilipu, 2022-08-15:
Yes. Reduced one level. But fixed now.

Probably won't affect anything else, as this is the only hybrid target as far as I know.

feilipu, 2022-08-16:
Got it. [Thanks. Hadn't added it previously](https://github.com/RC2014Z80/RC2014/commit/1e56d24f40095bf7cefc2c77de3239713ca0419d).

The build line is below from a terminal in the [acia85](https://github.com/RC2014Z80/RC2014/tree/master/ROMs/CPM-IDE/acia85) directory. More [info](https://feilipu.me/2022/03/23/cpm-ide-for-rc2014/) here.

```sh
zcc +rc2014 -subtype=acia85 -O2 --opt-code-speed=inlineints -m -D__CLASSIC -DAMALLOC -l_DEVELOPMENT/lib/sccz80/lib/rc2014/ff_85_ro @cpm22.lst -o ../rc2014-acia85-cpm22 -create-app
```

pauloscustodio, 2022-08-16:
I'm getting assembly errors in the current master. Am I missing any setting?

```
$ zcc +rc2014 -subtype=acia85 -O2 --opt-code-speed=inlineints -m -D__CLASSIC -DAMALLOC -l_DEVELOPMENT/lib/sccz80/lib/rc2014/ff_85_ro @cpm22.lst -o ../rc2014-acia85-cpm22 -create-app
cpm22bios.asm:352: error: illegal identifier
  ^---- ld (sektrk),bc
cpm22bios.asm:356: error: illegal identifier
  ^---- ld (seksec),bc
cpm22bios.asm:364: error: illegal identifier
  ^---- ld (dmaadr),bc ;save the address
      ^---- ld(dmaadr),bc
cpm22bios.asm:534: error: illegal identifier
  ^---- srl h ;shift right
      ^---- srl h
cpm22bios.asm:536: error: illegal identifier
  ^---- srl h ;shift right
      ^---- srl h
cpm22bios.asm:591: error: illegal identifier
  ^---- srl h
cpm22bios.asm:592: error: illegal identifier
  ^---- rr l
```

feilipu, 2022-08-16:
AFK. 3hr. Sorry.

pauloscustodio, 2022-08-16:
Sorry, forgot your time zone.

feilipu, 2022-08-16:
Ok sorry about delay. Looks like your terminal might be in the acia directory (for z80).
Those z80 instructions don't appear in the `cpm22bios.asm` in the [acia85](https://github.com/RC2014Z80/RC2014/tree/master/ROMs/CPM-IDE/acia85) directory.

Sorry the integration / automation is pretty poor. :man_shrugging:

pauloscustodio, 2022-08-17:
Now in the correct directory (RC2014/ROMs/CPM-IDE/acia85) a new error happens:
```
$ zcc +rc2014 -subtype=acia85 -O2 --opt-code-speed=inlineints -m -D__CLASSIC -DAMALLOC -l_DEVELOPMENT/lib/sccz80/lib/rc2014/ff_85_ro @cpm22.lst -o ../rc2014-acia85-cpm22 -create-app
main.c: line 23: file '_DEVELOPMENT/sccz80/lib/rc2014/ff.h' not found
```

feilipu, 2022-08-17:
Probably you've still got to install the newlib RC201 `ff` library automatically with `z88dk-lib` and then manually add the `ff_85` and `ff_85_ro` 8085 libraries manually.

From [here](https://github.com/feilipu/z88dk-libraries), use `z88dk-lib +rc2014 -f ff` to do the automatic install of newlib headers (including your missing `ff.h`) and library files.

Then [the extra files](https://github.com/feilipu/z88dk-libraries/tree/master/ff/rc2014/lib/newlib/sccz80) `ff_ro`, `ff_85`, and `ff_85_ro` need to be [manually placed into the right directory](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/lib/sccz80/lib/rc2014) for sccz80 use.

Sorry the process is a bit janky, but `z88dk-lib` can only handle one header file and one library file each. So building with different options or different subtypes mean manual handling.

pauloscustodio, 2022-08-17:
The culprit is my first suspect: [8a0f54f8a3e4ee06816a0c43ad9dd865ad3f80be] Move command line options processing to the C++ code

pauloscustodio, 2022-08-17:
The current master branch is OK; you are including ```../../../../lib/crt/classic/crt_init_amalloc.asm``` with 
```-I"~/git/z88dk_issue_2068/lib/config\..\..\/libsrc/_DEVELOPMENT/target/rc2014"``` which resolves to a valid path 
```"~/git/z88dk_issue_2068/lib/config\..\..\/lib/crt/classic/crt_init_amalloc.asm```

pauloscustodio, 2022-08-17:
There was an old bug in str_path_canon() that was not processing sequences of ```/../../``` correctly, causing the need to include one more level in the includes. This was uncovered by the now correct implementation using C++ std::filesystem and no longer using str_path_canon()

commit 35665d4a (2022-08-17): Fix #2068 fix str_path_canon()

str_path_canon() was not handling series of "/../../" correctly; needs to recheck string each time a "word/../" is eliminated.

Added test to detect failure.
closed 2022-08-17 by pauloscustodio