z88dk/z88dk #2069: z80asm - UNDEFINE

feilipu, 2022-08-17:
I'm trying to do something a bit odd, and would like to undefine a value that was previously defined. And, I'm not sure how the [syntax documented](https://github.com/z88dk/z88dk/wiki/Tool---z80asm---directives#undefine-name) for this is supposed to work.

I'm trying to build a [classic CP/M application for 8085](https://github.com/z88dk/z88dk-ext/blob/master/misc/yash.c), but would like to have access to RC2014 hardware constants to access the IDE hardware directly.

So in the command line I'm declaring the target as CP/M and the library as 8085, and linking RC2014 8085 libraries. No problem.

```sh
zcc +cpm -clib=8085 -O2 -v -m --list -DAMALLOC -l../../libsrc/_DEVELOPMENT/lib/sccz80/lib/rc2014/ff_85 -l../../lib/clibs/rc2014-8085_clib yash.c -o yash -create-app
```

And within the main file, I'm using [this bit of assembly](https://github.com/z88dk/z88dk-ext/blob/master/misc/yash.c#L201) to include the PUBLIC file from RC2014_8085 target, which is working fine.

```c
static void dsk0_helper(void) __naked
{
    __asm
;       UNDEFINE __CPU_CLOCK ; <- this is not working
        INCLUDE "../libsrc/_DEVELOPMENT/target/rc2014/config_rc2014-8085_public.inc"
;       defc _cpm_dsk0_base = 0xF700    ; XXX Uncomment for RC2014 SIO Build
        defc _cpm_dsk0_base = 0xF800    ; XXX Uncomment for RC2014 ACIA & ACIA 8085 Build
    __endasm;
}
```

The problem is that `__CPU_CLOCK` is defined in the CP/M crt0 and also in the RC2014 `config_rc2014-8085_public.inc`. I can successfully (and have tested with) manually remove `__CPU_CLOCK` from one of the files. But rather than doing that manual thing, I was hoping that `UNDEFINE` would get the same result more easily.

```sh
yash.c::dsk0_helper::0::0:202: error: syntax error
  ^---- UNDEFINE 7372800
```

Should this work?

feilipu, 2022-08-17:
For interest, I'm trying to do this odd thing because I've a transient failure with [f_open](https://github.com/z88dk/z88dk-ext/blob/master/misc/yash.c#L298).

It never works first time, but always works second time, so I'm trying to dig into why this happens, and then fix it.
Probably it will be somewhere in the 8085 string functions.

pauloscustodio, 2022-08-17:
It should work. The fact that the assembler is seeing ```UNDEFINE 7372800``` is odd, maybe the C preprocessor is replacing __CPU_CLOCK by the number? Could you look at the generated files (options -v -no-cleanup to zcc)?

feilipu, 2022-08-17:
It looks like `z88dk-ucpp` preprocessor has already converted it to 7372800, in the first step

``` sh
PROCESSING yash.c
z88dk-ucpp -iquote"." -D__Z88DK -D__CPM -DCPM -D__CPM__ -DZ80 -D__Z80 -DAMALLOC -D__8085__  -DSCCZ80 -DSMALL_C -D__SCCZ80 -isystem"/home/phillip/Z80/z88dk/lib/config/../..//include"    "yash.c" "/tmp/tmpXXWwFLjB.i2"
z88dk-zpragma -sccz80 -zcc-opt="/tmp/tmpzccXXHbdiGy/zcc_opt.def" < "/tmp/tmpXXWwFLjB.i2" > "/tmp/tmpXXWwFLjB.i"
z88dk-sccz80  -ext=opt -m8085 -zcc-opt="/tmp/tmpzccXXHbdiGy/zcc_opt.def" -standard-escape-chars   "/tmp/tmpXXWwFLjB.i" -o "/tmp/tmpXXWwFLjB.opt"
z88dk-copt -m8085 "/home/phillip/Z80/z88dk/lib/config/../..//lib/z80rules.9" < "/tmp/tmpXXWwFLjB.opt" > "/tmp/tmpXXWwFLjB.op1"
z88dk-copt -m8085 "/home/phillip/Z80/z88dk/lib/config/../..//lib/z80rules.2" < "/tmp/tmpXXWwFLjB.op1" > "/tmp/tmpXXWwFLjB.opt"
z88dk-copt -m8085 "/home/phillip/Z80/z88dk/lib/config/../..//lib/z80rules.1" < "/tmp/tmpXXWwFLjB.opt" > "/tmp/tmpXXWwFLjB.asm"
cat "/tmp/tmpXXWwFLjB.asm" >> "/tmp/tmpXXWwFLjB.tmp"
cat "/tmp/tmpXXWwFLjB.tmp" >> "/tmp/tmpXXWwFLjB.asm"
z88dk-z80asm -I"/tmp/tmpzccXXHbdiGy" -D__SCCZ80 -l   -s  -m8085 -I"/home/phillip/Z80/z88dk/lib/config/../..//lib" "/tmp/tmpXXWwFLjB.asm"
yash.c::dsk0_helper::0::0:202: error: syntax error
  ^---- UNDEFINE 7372800
phillip@SilverQuad:~/Z80/z88dk-ext/misc$ ^C

```

resulting in `/tmp/tmpXXWwFLjB.i2` containing.

```c

static void dsk0_helper(void) __naked
{
    __asm
        UNDEFINE  7372800 
        INCLUDE "../libsrc/_DEVELOPMENT/target/rc2014/config_rc2014-8085_public.inc"
;       defc _cpm_dsk0_base = 0xF700    ; XXX Uncomment for RC2014 SIO Build
        defc _cpm_dsk0_base = 0xF800    ; XXX Uncomment for RC2014 ACIA & ACIA 8085 Build
    __endasm;
}

```

So somehow this `__CPU_CLOCK` needs to get past the `z88dk-ucpp` in order for `z88dk-z80asm` to undefine it.

closed 2022-08-17 by feilipu

feilipu, 2022-08-18:
Removed the `PUBLIC __CPU_CLOCK` definition. It is only used for building libraries, and for CP/M newlib builds was incorrect most of the time anyway. https://github.com/z88dk/z88dk/commit/066b276b586c0c3ca16431d3367ea4aca5b7e4b4