z88dk/z88dk #2071: Little opt for CP/M fcntl

zx70, 2022-08-19:
This one started as a small intervention on the CP/M build, but I under-estimated the file bindings.
Sorry for the recent build warnings, I think it is all sorted now.

- Reduced number of warnings in the fcntl library build
- quick and dirty insertion of the CALLEE variant of the bdos() function, together with the "__bdos" centralization it reduces the library size
- __bdos entry usage extended to all the CP/M fcntl lib to centralize the IX preservation, this means that an equivalent label is now declared also for the MSX
- bios() doubled to biosh(), the latter has different return values, picking HL in place of the A register

feilipu, 2022-08-20:
Not sure if you're up for some more optimisation, but...

Some time ago I tried to optimise the `read.c` and `write.c` to use `BDOS` sequential read and write where the request matched the sector alignment and size (128 Bytes). It did work most of the time, but there were corner cases (eg. `lseek`) where it didn't work, so it [had to be backed out](https://github.com/z88dk/z88dk/commit/ccea56eafdbdfab59e59ec1b5357544636e3dab6).

The reason that I thought it could work is that CP/M `PIP` exclusively uses BDOS sequential read and write calls, and `PIP` remains, to my knowledge, the fastest way of copying a file in CP/M.

The PIP PLM source is available [from here](http://www.cpm.z80.de/download/cpm2-plm.zip). I've added it to this issue. [PIP.PLM](https://github.com/z88dk/z88dk/files/9386090/PIP.TXT)

Perhaps you can see how it manages to use the BDOS sequential read and write calls?
(and entirely avoids using BDOS random read and write calls)?

Don't know. I couldn't solve it. :man_shrugging: 

```plm
DISKREAD: PROCEDURE(FCB) BYTE;

    DECLARE FCB ADDRESS;
    RETURN MON2(20,FCB);
    END DISKREAD;

DISKWRITE: PROCEDURE(FCB) BYTE;
    DECLARE FCB ADDRESS;
    RETURN MON2(21,FCB);
    END DISKWRITE;

  FILLSOURCE: PROCEDURE;
    /* FILL THE SOURCE BUFFERS */
    DECLARE (I,J) BYTE;
    NSOURCE = 0;
    CALL SELECT(SDISK);
    CALL SETSUSER; /* SOURCE USER NUMBER SET */
        DO I = 0 TO NBUF;
        /* SET DMA ADDRESS TO NEXT BUFFER POSIITION */
        CALL SETDMA(.SBUFF(NSOURCE));
        IF (J := DISKREAD(.SOURCE)) <> 0 THEN
            DO; IF J <> 1 THEN
                CALL ERROR(.('DISK READ ERROR$'));
            /* END - OF - FILE */
            HARDEOF = NSOURCE; /* SET HARD END-OF-FILE */
            SBUFF(NSOURCE) = ENDFILE; I = NBUF;
            END; ELSE
        NSOURCE = NSOURCE + 128;
        END;
    NSOURCE = 0;
    CALL SETCUSER; /* BACK TO CURRENT USER NUMBER */
    END FILLSOURCE;

  WRITEDEST: PROCEDURE;
    /* WRITE OUTPUT BUFFERS UP TO BUT NOT INCLUDING POSITION
    NDEST - THE LOW ORDER 7 BITS OF NDEST ARE ZERO */
    DECLARE (I, J, N) BYTE;
    DECLARE DMA ADDRESS;
    DECLARE DATAOK BYTE;
    IF (N := LOW(SHR(NDEST,7)) - 1) = 255 THEN RETURN ;
    NDEST = 0;
    CALL SELECT(DDISK);
    CALL SETRANDOM(.DEST); /* SET BASE RECORD FOR VERIFY */
        DO I = 0 TO N;
        /* SET DMA ADDRESS TO NEXT BUFFER */
        DMA = .DBUFF(NDEST);
        CALL SETDMA(DMA);
        IF DISKWRITE(.DEST) <> 0 THEN
            CALL ERROR(.('DISK WRITE ERROR$'));
        NDEST = NDEST + 128;
        END;
    IF VERIF THEN /* VERIFY DATA WRITTEN OK */
        DO;
        NDEST = 0;
        CALL SETDMA(.BUFF); /* FOR COMPARE */
            DO I = 0 TO N;
            DATAOK = READRANDOM(.DEST) = 0;
            DESTR = DESTR + 1; /* NEXT RANDOM READ */
            J = 0;
                /* PERFORM COMPARISON */
                DO WHILE DATAOK AND J < 80H;
                DATAOK = BUFFER(J) = DBUFF(NDEST+J);
                J = J + 1;
                END;
            NDEST = NDEST + 128;
            IF NOT DATAOK THEN
                CALL ERROR(.('VERIFY ERROR$'));
            END;
        DATAOK = DISKWRITE(.DEST);
        /* NOW READY TO CONTINUE THE WRITE OPERATION */
        END;
    NDEST = 0;
    END WRITEDEST;
```

zx70, 2022-08-20:
Yes,  I remember the topic. 
Sadly I don't have the exact test case anymore, nor I'm able to test the resulting performances,  but I'll have a look. 
My major concerns at the moment were related to the facts that the resulting program size sometimes is much bigger than the ones got by building the same programs with historical compilers in native cp/m environment.   Moreover the tricky way the BDS C had to deal with file redirection seems to give more predictive results than our current "parameter grab" from the current command line.

zx70, 2022-08-20:
About the historical sources,  I'm looking at the few survived files of the "portable cp/m" project by DR.   It seems that there were PL/M alternatives in Pascal and C for CPUs like Motorola or Zilog Z8000..

feilipu, 2022-08-20:
Would be useful to have `PIP` too, if it were available in C.

zx70, 2022-08-20:
I was surprised by the C conversion of the SUBMIT tool, I think it was created earlier by the creator of MESCC.  It looks like a faithful equivalent of the PL/M tool, probably written starting from it.
I wonder how difficult would be that approach,   if I find time for that I'll compare an earlier version of STAT with the later C conversion. 
Only a part of the tools is available,  though, the only ones which exfiltered from Zilog/Olivetti for the M20 computers.  At the moment it is believed that the original DR source code is mostly lost.

zx70, 2022-08-26:
I succeeded in compiling STAT.C with z88dk!
It doesn't work,  but the program is able to understand the command line parameters, correctly uses BDIS to deal with the console i/o, and does part of the expected work.
I had to extend the version checks and insert our interface to the BIOS to reach the DPB table on CP/M versions older than v3.
It does everything with no need of argv or stdio.h.
There's a lot of work left but my changes to the original code are very limited,  at the moment. 
I think I'll put the unfinished results online to share it.

feilipu, 2022-08-27:
> Sadly my computer is now broken, I'm not sure I can recover that experiment.

Must be the weather... Just recovering my work from a hard drive failure on `/home`. Surprised that a NAS drive I purchased specifically for reliability (over speed) would fail after just 4 years of desktop use.

Good luck with the recovery.

zx70, 2022-08-27:
I made it !
I saved the unfinished backport of stat.c together with the mostly untouched original ed.c, which I haven't tried to build yet.

The code included runtime conditional calls depending on CP/M version checks rather than conditional code included at build time, which is very interesting IMO.
Another curious thing is that bdos.h seems to have been created by Zilog, which makes me think that this C variant of the "Portable CP/M" project was a joint effort.   It is believed that the high level language components in the M68K version relied on the Pascal language.

By the way, the this version of stat command was not able to support the older CP/M versions like 2.x, but it is clear that the original programmers had it in mind:
https://github.com/z88dk/z88dk-ext/blob/51153dec8a37ab04afa17a852c658d3ccd969e78/os-related/CPM/DR/stat.c#L201

The more invasive change I've done at the moment:
https://github.com/z88dk/z88dk-ext/blob/51153dec8a37ab04afa17a852c658d3ccd969e78/os-related/CPM/DR/stat.c#L528

zx70, 2023-05-12:
I'm closing it because I'm not planning to do more on this subject at the moment. 
@felipu, the programs you mentioned work correctly it the files are never accessed randomly. 
The only way I see to make your idea work is to prepare a new library in alternative to the existing one, probably it is a good idea to exclude or alter fseek(), etc.

closed 2023-05-12 by zx70