z88dk/z88dk #2072: classic - possible memory non-initialisation

feilipu, 2022-08-21:
Been hunting this one for some time, but since it was not critical it has taken some time.

I have replicated the issue into an application, which makes tracking it easier, and it looks strongly like there's something wrong with both Z80 and I8085 string initialisation in classic library. Let's see if I can find it.

This issue first identified in CP/M-IDE for 8085 in the [cpm](https://github.com/RC2014Z80/RC2014/blob/master/ROMs/CPM-IDE/acia85/main.c#L148) function.

The issue is that a string match consistently doesn't work first time, but does match the second time.
But in certain circumstances it does work first time, possibly where something has been properly initialised previously elsewhere.

This below result from compiling [yash.c](https://github.com/z88dk/z88dk-ext/blob/master/misc/yash.c#L281) for both I8085 sccz80/classic and Z80 using sccz80/new.

The [ff library](https://github.com/feilipu/z88dk-libraries/tree/master/ff) is compiled for both I8085 and for Z80 using sccz80.

The first time `"turbop.cpm"` is captured and printed correctly, but doesn't match the file using functions within the ff library.
The second time it does match (and is correctly mounted).

```sh
B>yash85

> ls
----A 2020/04/27 19:55   8388608  BBCBASIC.CPM
----A 2022/06/12 15:42   8388608  HITECHC.CPM
----A 2020/04/27 19:55   8388608  MSBASCOM.CPM
----A 2020/04/27 19:55   8388608  MSCOBOL.CPM
----A 2020/01/01 00:00     65536  MULTHI.BIN
----A 2020/01/01 00:00     65536  MULTLO.BIN
----A 2020/05/13 18:03   8388608  NZCOM.CPM
----A 2020/04/27 19:55   8388608  NZSYS.CPM
----A 2020/05/10 21:04   8388608  PLI.CPM
----A 2020/05/10 21:05   8388608  PLISRC.CPM
----A 2020/05/03 14:18   1048576  RANDOM1.TXT
----A 2020/04/27 19:55   8388608  SYS.CPM
----A 2020/04/27 19:55   8388608  TEMPLATE.CPM
----A 2020/05/10 22:03   8388608  UNDRSTND.CPM
----A 2020/04/28 17:22   8388608  USER.CPM
----A 2020/04/28 17:22   8388608  USER2.CPM
----A 2020/04/28 17:22   8388608  USER3.CPM
----A 2020/04/27 19:55   8388608  ZORK.CPM
----A 2021/03/01 13:30   8388608  SLRTOOLP.CPM
----A 2022/07/10 19:52   8388608  TURBOP.CPM
----A 2021/03/01 13:30   8388608  SLRTOOL.CPM
----A 2022/07/10 20:07     53474  TURBOP~1.ZIP
D---- 2022/06/12 14:41         0  TRASH-~1
  22 File(s), 152228066 bytes total
   1 Dir(s), 1883230208 bytes free

> dmount c turbop.cpm
Mounting "turbop.cpm" on drive C:

rc=6 FR_INVALID_NAME

> dmount c turbop.cpm
Mounting "turbop.cpm" on drive C: from LBA 411616

> exit

B>c:

C>dir

C: MC       HLP : READ     ME  : TINST    COM : TINST    DTA
C: TINST    MSG : TURBO    COM : TURBO    MSG : TURBO    OVR
C>
```

Isolation process.

- [x] __present__ in sccz80/classic/I8085
- [x] __present__ in sccz80/classic/Z80
- [x] not in sccz80/newlib
- [x] not in sdcc/newlib

Something is causing `ff` [`f_open`](https://github.com/feilipu/z88dk-libraries/blob/master/ff/source/ff.c#L3689) to return with `FR_INVALID_NAME` which is an unusual result. It should be easy to find when this return is generated, and work out which string functions are involved in the process.

feilipu, 2022-08-21:
It seems to point back to the [`ya_get_line()`](https://github.com/z88dk/z88dk-ext/blob/master/misc/yash.c#L861) implementation, as I'm suspicious that a space is being left as the first character in tokenisation `' '`. This is indicated by [`follow_path()`](https://github.com/feilipu/z88dk-libraries/blob/master/ff/source/ff.c#L3077) in ff.

feilipu, 2022-08-21:
It is odd though, that [`__string_strtok_p`](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/string/z80/__string_strtok_p.asm) is [read](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/string/z80/asm_strtok.asm#L54), but never written...?

suborb, 2022-08-21:
https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/string/z80/asm_strtok.asm

feilipu, 2022-08-22:
Checking the output of `strtok()` provides no further clues.

feilipu, 2022-08-25:
Possibly, this is an issue related to [`ya_ls`](https://github.com/z88dk/z88dk-ext/blob/master/misc/yash.c#L423) and the process of reading a directory corrupting something, rather than `ya_split_line` being faulty. So, the search continues there...

feilipu, 2022-08-27:
Isolating the functions in `ya_ls` leads to the ["unusual" call](https://github.com/z88dk/z88dk-ext/blob/master/misc/yash.c#L433) of `res = f_opendir(dir, (const TCHAR*)".");` where [`dir` is previously allocated with `malloc()`](https://github.com/z88dk/z88dk-ext/blob/master/misc/yash.c#L927).

Replacing allocated `dir` with a stack version, removes the fault. So now the fault is isolated. :+1: 
Is it pointer passing? Or `malloc()` initialisation?

```c
int8_t ya_ls(char ** args)
{
    FRESULT res;
    DIR dir;                   /* Get work area for the directory */
    uint32_t p1;
    uint16_t s1, s2;

    res = f_mount(fs, (const TCHAR*)"0:", 0);
    if (res != FR_OK) { put_rc(res); return 1; }

    if(args[1] == NULL) {
        res = f_opendir(&dir, (const TCHAR*)".");
    } else {
        res = f_opendir(&dir, (const TCHAR*)args[1]);
    }
    if (res != FR_OK) { put_rc(res); return 1; }

    return 1;
}
```

suborb, 2022-08-27:
I think that’s using the classic Malloc? So needs setting up in crt using the Amal lox style code.

feilipu, 2022-08-27:
I thought that `malloc()` would already be covered by the `amalloc()` initialisation code in the [CP/M initialisation crt0](https://github.com/z88dk/z88dk/blob/master/lib/target/cpm/classic/cpm_crt0.asm#L155)?

suborb, 2022-08-27:
It should do as long as you specify the appropriate option.

feilipu, 2022-08-27:
I think the correct option is to add `-DAMALLOC` to the build command, like [the example command here](https://github.com/z88dk/z88dk-ext/blob/master/misc/yash.c#L116).
Is that all?

feilipu, 2022-08-29:
Somehow the program is scribbling into the file structure of [`fs`](https://github.com/z88dk/z88dk-ext/blob/master/misc/yash.c#L926) on the heap, if `dir` is defined on the heap. Not sure why yet.

AND

Separately, if [`Finfo`](https://github.com/z88dk/z88dk-ext/blob/master/misc/yash.c#L157) is defined in the function the `ls` function in the program fails, unless `Finfo` is defined in the function as a `static`.
It is pointing to either issues with [ff](https://github.com/feilipu/z88dk-libraries/tree/master/ff), or issues with local stack access into structs working differently to static access into structs.

feilipu, 2022-08-30:
Very odd.

The error is gone is [`DIR dir`](https://github.com/z88dk/z88dk-ext/blob/master/os-related/yash.c#L422) is defined on the stack, but persists is `DIR * dir` is global and malloc'ed. Or, if it is defined `static DIR dir` function local. 100% reproducible.

AND

Directory listing only works if [`static FILEINFO Finfo`](https://github.com/z88dk/z88dk-ext/blob/master/os-related/yash.c#L427) is either global or function static. On the stack, it won't work and no files are listed in the directory.

Two problems. Reading the resulting code provides no clues.

closed 2022-09-04 by feilipu