z88dk/z88dk #2081: [KC][getk]

Hojoe42, 2022-09-18:
Hi,

the getk() function for target KC on KC 85/4 sometimes return (random) values.

The following program prints a number and the current pressed key code. If it runs and **no** key is pressed, the variables x and ch contains the same value! If a key is pressed ch shows the correct key code.

It can compiled with `zcc +kc -g --list -o GetkTest-l.o -create-app GetkTest.c`.
It runs with JKCEmu with System 85/4.

```
#include <stdio.h>
#include <stdlib.h>
#include <stdbool.h>
#include <conio.h>

char START[] = {0x7f, 0x7f, 'G', 'E', 'T', 'K', 0x01, 0xcd, 0x00, 0x10};

uint8_t getSomething();
uint8_t something;

int main()
{
    something = 0;
    while(true)
    {
        uint8_t x = getSomething();
        uint8_t ch = getk();
        gotoxy(0, 0);
        cprintf("something: %d  und ch: %d  ", x, ch);
        if( ch == 'X')
        {
            gotoxy(0, 30);
            cputs("ENDE");
            break;
        }
        else if( ch == 'H')
        {
            gotoxy(0, 1);
            cputs("Hello World");
        }
    }
    return 0;
}
uint8_t getSomething()
{
    return something++;
}

```
The os documentation for function FNKBDS in [target/kc/stdio/getk.asm](https://github.com/z88dk/z88dk/blob/master/libsrc/target/kc/stdio/getk.asm) says:
> Name: . . . KBDS . . . . . . . . . . . . . UP-Nr. 0CH
> FKT.: Tastenstatusabfrage ohne Quittierung der Taste
> PE: -
> PA: CY = 0 => keine Taste gedrückt, dann Register A unverändert
> CY = 1 => Taste gedrückt, dann A = Zeichencode (ASCII)
> VR: AF
CY == 0 means no key is pressed and CY == 1 means a key is pressed. Currently the CY flag isn't checked.

[getk.patch.txt](https://github.com/z88dk/z88dk/files/9594180/getk.patch.txt) contains a patch

Holger

feilipu, 2022-09-21:
I've added the patch, but note I've no way to test the patch.
Please check and reopen this issue if I've messed up.

closed 2022-09-21 by feilipu

Hojoe42, 2022-09-22:
Thank you. 👍  
I used the nightly build and now it works on KC 85/3 and KC 85/4 :-)

commit 4e95b8df (2022-09-21): KC - patch getk - closes #2081
commit 40865519 (2022-09-21): KC - patch getk typo - closes #2081