z88dk/z88dk #2090: sccz80 - long right shift inconsistent

feilipu, 2022-09-30:
The testing of arithmetic right shift is inconsistent with the implementations.

The implementation for both [8080 long](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/l/sccz80/8-8080/i32/l_long_asr.asm#L36) and [8080 integer](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/l/sccz80/8-8080/l_asr.asm#L17) (and derivative CPUS) and [z80](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/l/z80/long/small/l_small_asr_dehl.asm#L31) assumes that the MSB is replicated.

But the testing assumes that it is not replicated.
[RHS Variable](https://github.com/z88dk/z88dk/blob/master/test/suites/sccz80/rshift.c#L49)
[RHS Constant](https://github.com/z88dk/z88dk/blob/master/test/suites/sccz80/rshift.c#L165)

Which one are we going with, because AFAIK it is undefined in C?

feilipu, 2022-09-30:
Think this may be a long eq comparison issue rather than an arithmetic right shift issue.

feilipu, 2022-09-30:
I've fixed / split out the 8085 code, but there's still an issue when the MSW is zero.
Neither z80 nor 8085 code paths pass the ASR >> 16 tests when `--opt-code-speed=longcompare` is set.
I'm missing something, but I can't see what it is currently.

feilipu, 2022-09-30:
Possibly the shorter code path should never be used unless the type is `unsigned`?
i.e. `value > 0`. Otherwise the comparison to zero for the MSW fails.

suborb, 2022-09-30:
The "expected" behaviour is:

- Signed = the sign bit is preserved
- Unsigned = sign bit is reset

I think these routines were tested across all CPU variants so the "don't touch any option" behaviour is good?

So, any issue must lie in the speed optimisations?

feilipu, 2022-09-30:
Yes. It is only the inline code 32-bit compare that fails on both z80 and 8085.

I've fixed the 8085 code to be the "same" jp lengths as z80. But there's something else wrong where the shorter code path is used. i.e. `asr >> 16` fails because compare is failing (not the shift failing).

suborb, 2022-09-30:
The result of 0x80000000L >> 16 should be 0xFFFF8000 so the short path shouldn't be being taken.

suborb, 2022-09-30:
It's to do with some fun with unsigned/signed operations.

```
    int32_t value = (int32_t)value64;

     if ( value / 65536 == 0 ) {
```

Consider the value is -32768, that condition ends up being true so it heads down the short path.

When considering what to do, and outputting the word components we need to treat the number as unsigned. There's a few other places that do extra casts:

```
uint32_t highword = ((uint32_t)value) / 65536;
```

feilipu, 2022-09-30:
Also affects `zne` function too.

And the `set_carry(lval);` call is not consistent between `zeq` and `zne`. I'm not sure which way is correct. 🤷‍♂️

suborb, 2022-09-30:
The zeq() variant looks correct - if we're leaving the true/false in the carry flag (which we are with these optimisations) then `set_carry()` should be called.

feilipu, 2022-09-30:
OK. Fixed it. The `(uint32_t)` cast needs to be on both the code path selection, and on the `bc` loading.

commit 263ce0c5 (2022-09-30): sccz80 - fix opt-code-speed=longcompare z80 8085 - closes #2090
closed 2022-09-30 by feilipu