z88dk/z88dk #2099: lib3d - add 4d and matrices

feilipu, 2022-10-13:
Looking at the path to integrate the __`lib3d`__ with pieces of a [homogenous coordinate](https://en.wikipedia.org/wiki/Homogeneous_coordinates) [__`3d`__](https://github.com/feilipu/z88dk-libraries/blob/master/3d/source/include/sccz80/3d.h) library.

Quoting from Wikipedia on why to do this: "Homogeneous coordinates are ubiquitous in computer graphics because they allow common vector operations such as [translation](https://en.wikipedia.org/wiki/Translation_(geometry)), [rotation](https://en.wikipedia.org/wiki/Rotation_(mathematics)), [scaling](https://en.wikipedia.org/wiki/Scaling_(geometry)) and [perspective projection](https://en.wikipedia.org/wiki/Perspective_projection) to be represented as a matrix by which the vector is multiplied. By the chain rule, any sequence of such operations can be multiplied out into a single matrix, allowing simple and efficient processing. By contrast, using Cartesian coordinates, translations and perspective projection cannot be expressed as matrix multiplications, though other operations can."

The key things needed are.

1. vectors with an added 4th  ___w___ dimension.
2. matrix operations for 4x4 order. (I updated `3d` library yesterday to use an intermediate matrix, so that the matrix operations can be chained without costing RAM. e.g. see [`rotx_m()`](https://github.com/feilipu/z88dk-libraries/blob/master/3d/source/rotx_m.c#L63) ).
3. options to choose the existing fixed point functions (multiply, sin, and cosine) or any of the sccz80 supported floating point library functions.

@zx70 My `3d` library is not very comprehensive. Really just the matrix operations and the x,y,z,w vector operations. Perhaps I can just add the needed functions and add the dimension to the vector structures of `lib3d`? But, I don't want to mess with other machines' outcomes that I can't test.

zx70, 2022-10-13:
Feel free to extend, I think the possible impacts are limited, if you are evolving the real 3D stuff.
I reused isin() and icos() from time to time wich brought me to put the functions to draw polygons, arcs and turtle graphics in it, but the rotation/scaling functions were used very little.
A grep scan in the "examples" subfolders will suffice.

feilipu, 2022-10-17:
I can start to move some files in, now that I'm happy with them.

But I don't think I'll be able to write the makefile to build them for the different math libraries and machine types. Lots of combinations to support, even just for CP/M. Perhaps I'll have a go and call out if I need some help.

zx70, 2022-10-17:
Good point, I'd first look at the transformation function (rotation, etc..).   I *think* we already have a way to do it with integers.

zx70, 2022-10-17:
to clarify, 
https://github.com/z88dk/z88dk/blob/master/libsrc/lib3d/ozrotatepointx.c

could you replace the Floating Point functions with isin/icos?  The values will be approximated but it should be faster.
the next step will be deciding on the data structures, whether to merge the two libraries, keep  both the structures or chose the best one (probably yours) .

feilipu, 2022-10-18:
> Could you replace the floating point with `icos` / `isin`?

That's actually one of the points that I'm hung up on right now. That, and the treatment of "native" math operations by `SCCZ80`.

I've written my functions to make use of the `SCCZ80` capability to do a multiply in whatever is the current math format. So for example, where a multiply is `_Float16` or `float` then it selects the right code, rather than having to call specific functions for every math operation.

So I'd have to rewrite code to use fixed point `imult()`, or 16-bit floating `l_mul_f16` or 32-bit floating `l_mul_f32` or of course genmath `l_mul_f48` (I can't remember the exact function names, but you get the point) depending on which build was being done. Now `SCCZ80` handles that all automatically, except of course the fixed point code.

So, it is quite a bit of effort to translate the functions to not use the `SCCZ80` style.

---

The other question that I have is regarding the function style. I've gone with first parameter is also the result. i.e. `x *= y` or `mult_m( &x, &y)`. That allows me to use the compiler autogenerated inline `memcpy()` to put the result in the the right place. And, since most 3D transformation operations are chained it reduces the memory usage.

But the existing operations use 3 pointers.

I'm can be ok with using 3 pointers, as I can make `mult_m(&x, &x, &y)` work by using an interim matrix as I do currently. But, would you consider the 3 pointer style important to maintain?

For the (rot and translate, etc) transformations, I've set them up to take scalars to operate on the provided target matrix, rather than setting up a transformation matrix which is later multiplied with the target matrix. Again, this is to reduce memory usage. Is it ok to keep this method?

zx70, 2022-10-18:
> But, would you consider the 3 pointer style important to maintain?
Not at all.  As I said the I used the 3D library mostly for adding graphics functions which required SIN/COS.   The code examples really depending on 3D vectors are so few.   If we have the opportunity to get a smarter function set we probably should do it.

On the other side, doing everything with integers was a good way to save memory and CPU time, which to be honest is the main reason for me proposing the merge.
I don't know how the 8080 and 8085 deal with long types, though.

feilipu, 2023-08-06:
Stale.

closed 2023-08-06 by feilipu