z88dk/z88dk #2110: __z88dk_inline & __z88dk_no_reentrant

desertkun, 2022-10-22:
I've been thinking to optimize code on my project and I more often than node, I had to resort to ugly fixes, which should be able to be handled by z88dk directly, so I need a feature z88dk is currently missing (to my knowledge):

`__z88dk_non_reentrant`
Local variables and function arguments are not located on stack, instead, declared as global variables. When non-reentrant function is called, instead of pushing arguments on stack, corresponding global "arguments" variables are modified. Should be able to coexist with `__z88dk_fastcall`.

`__z88dk_inline`
When a function is marked with that, the body of function is copied instead of CALL instruction. Function arguments instead declared as local variables on stack for the inline code. If that function is also `__z88dk_non_reentrant` those arguments are global.

What do you think?

suborb, 2022-10-26:
`__z88dk_non_reentrant` should be fairly easy to do.

inline is a bit tricker - can you do what you need to do with macros?

desertkun, 2022-10-26:
Can't debug macros. Also, I am doing fair share of cross-compiled code, so defines for z88dk would force defines for PC too. Eh.

desertkun, 2022-10-31:
@suborb so I tried to make a first attack on __z88dk_non_reentrant, having limited success https://github.com/desertkun/z88dk/commit/e71a300e37580725377ab50bd14f335fe27802af

On a __z88dk_non_reentrant function, I declare any local as a static global. And while initializing it with cost value worked fine (remember unlike static, a `char a = 10` has to be reset on entry every time), initializing it with some other variable has led to issues. 

Could you help me put right code on the `// ????????` block on my change on my branch __z88dk_non_reentrant at the link above? 

```c
#include <stdio.h>
char check_a() __z88dk_non_reentrant
{
        char a = 20;
        a++;
        return a;
}
char check_b(char v) __z88dk_non_reentrant
{
        char b = v;
        b += 2;
        return b;
}
int main() {
        printf("a = %d\n", check_a());
        printf("b = %d\n", check_b(5));
}
```

To test `zcc +test main.c && z88dk-ticks a.bin`

It prints out 
```
a = 21
b = 2
```
because b is not initialized properly. Because hl contains address of b and something else should contain address of v and it has to be copied over somehow. I've only prepared b.

suborb, 2022-10-31:
I think you need a `gen_store_static(sym)` call?

desertkun, 2022-10-31:
Haha it worked :D Thanks. This change is aggressive, but it shaves off solid 200-300 bytes depending on the function. I feel like it has to be unit tested. Not sure how UT framework works here. I'll revisit that once I move all the code I can to non-reentrant one and test it on a field!

Also additional question: should I also make function arguments be also global? and when it's being called, it's not "push stuff" but rather "fill that variable up", that could shave some bytes too.

suborb, 2022-10-31:
As a note, don't forget about scope frames! The naming convention in your branch assumes just a single variable of that name in a function. You should include `sym->level` in the name.

suborb, 2022-11-11:
@desertkun that branch has a first stab at making function arguments static in case you've not figured out the way to do it.

desertkun, 2022-11-11:
Oh wow. I've lost steam for the moment, but looks very interesting. This could shave legit bytes if it works. Also z88dk has to support it, since this ain't not stack (right?).

Cool cool. I'll revisit it sometime. But feel free to steal the moment for the time being.

desertkun, 2022-11-13:
@suborb I've cherry-picked your changes. Seems like so far we do not gain any bytes :(

code
```
#include <stdio.h>

char sum(char a, char b)
{
	return a + b;
}

char check(char a, char b, char c, char d) __z88dk_non_reentrant
//char check(char a, char b, char c, char d)
{
        return sum(a, b * 2) + c / 2 + d % 2;
}

int main() {
        printf("b = %d\n", check(10, 11, 12, 13));
}
```

footprint of code WITH `__z88dk_non_reentrant` is 2056 bytes, WITHOUT is 2048

generated __z88dk_non_reentrant
```
_check:
                    call    l_debug_push_frame              ;[014b] cd de 06
                    ld      hl,$07f3                        ;[014e] 21 f3 07
                    call    l_gchar                         ;[0151] cd ef 06
                    push    hl                              ;[0154] e5
                    ld      hl,$07f2                        ;[0155] 21 f2 07
                    call    l_gchar                         ;[0158] cd ef 06
                    add     hl,hl                           ;[015b] 29
                    push    hl                              ;[015c] e5
                    call    _sum                            ;[015d] cd 31 01
                    pop     bc                              ;[0160] c1
                    pop     bc                              ;[0161] c1
                    push    hl                              ;[0162] e5
                    ld      hl,$07f1                        ;[0163] 21 f1 07
                    call    l_gchar                         ;[0166] cd ef 06
                    bit     7,h                             ;[0169] cb 7c
                    jr      z,i_2                           ;[016b] 28 01
                    inc     hl                              ;[016d] 23
i_2:
                    sra     h                               ;[016e] cb 2c
                    rr      l                               ;[0170] cb 1d
                    pop     de                              ;[0172] d1
                    add     hl,de                           ;[0173] 19
                    push    hl                              ;[0174] e5
                    ld      hl,$07f0                        ;[0175] 21 f0 07
                    call    l_gchar                         ;[0178] cd ef 06
                    bit     7,h                             ;[017b] cb 7c
                    jr      z,i_3                           ;[017d] 28 04
                    ld      a,l                             ;[017f] 7d
                    cpl                                     ;[0180] 2f
                    inc     a                               ;[0181] 3c
                    ld      l,a                             ;[0182] 6f
i_3:
                    ld      a,l                             ;[0183] 7d
                    and     $01                             ;[0184] e6 01
                    ld      l,a                             ;[0186] 6f
                    ld      h,$00                           ;[0187] 26 00
                    pop     de                              ;[0189] d1
                    add     hl,de                           ;[018a] 19
                    ld      a,l                             ;[018b] 7d
                    call    l_gchar_sxt                     ;[018c] cd f0 06
                    call    l_debug_pop_frame               ;[018f] cd d6 06
                    ret                                     ;[0192] c9

[... 83 lines cut ...]
                    ld      hl,$000b                        ;[01a2] 21 0b 00
                    push    hl                              ;[01a5] e5
                    ld      hl,$000c                        ;[01a6] 21 0c 00
                    push    hl                              ;[01a9] e5
                    ld      hl,$000d                        ;[01aa] 21 0d 00
                    push    hl                              ;[01ad] e5
                    call    _check                          ;[01ae] cd 4b 01
                    pop     bc                              ;[01b1] c1
                    pop     bc                              ;[01b2] c1
                    pop     bc                              ;[01b3] c1
                    pop     bc                              ;[01b4] c1
```

It seems we do a lot of dumb
```
                    ld      hl,$000b                        ;[01a1] 21 0b 00
                    ld      a,l                             ;[01a4] 7d
                    ld      (_st_check_b),a                 ;[01a5] 32 f2 07
```

Where it could be
```
                    ld      a, $0b                        
                    ld      (_st_check_b),a               
```

desertkun, 2022-11-13:
OK I was able to fix that with COPT
```
	ld	hl,%1	;const
	ld	a,l
	ld	(%2),a
=
	ld	a,%eval(%1 256 %%)	;const
	ld	(%2),a
```

It generates nice
```
                    ld      a,$0a                           ;[019a] 3e 0a
                    ld      (_st_check_a),a                 ;[019c] 32 eb 07
                    ld      a,$0b                           ;[019f] 3e 0b
                    ld      (_st_check_b),a                 ;[01a1] 32 ea 07
                    ld      a,$0c                           ;[01a4] 3e 0c
                    ld      (_st_check_c),a                 ;[01a6] 32 e9 07
                    ld      a,$0d                           ;[01a9] 3e 0d
```

But this is still problems, seems like there's a lot to copt.

suborb, 2022-11-13:
That rule is too broad - I think you need to match a ld hl afterwards as well to ensure that the hl value isn’t used. 

I thought there was already a rule to do that check though.

desertkun, 2022-11-13:
I don't follow, not sure how far do I look for unused hl.

spth, 2023-02-04:
What is the motivation for __z88dk_inline? Why not just use normal inline?

suborb, 2023-03-19:
Aging out this experiment.

closed 2023-03-19 by suborb