z88dk/z88dk #2118: cygwin build - chocolatey failure

feilipu, 2022-11-04:
This has been broken for some time.

```sh
Cygwin v3.3.6 [Approved]
cygwin package files upgrade completed. Performing other installation steps.
Download site: http://mirrors.kernel.org/sourceware/cygwin/
Installing 64-bit Cygwin...
Cygwin has been installed.
Added C:\ProgramData\chocolatey\bin\Cygwin.exe shim pointed to 'c:\tools\cygwin\cygwin.bat'.
Copying cygwin package manager (setup) to C:\tools\cygwin
Environment Vars (like PATH) have changed. Close/reopen your shell to
 see the changes (or in powershell/cmd.exe just type `refreshenv`).
 The upgrade of cygwin was successful.
  Software installed to 'C:\tools\cygwin'

Chocolatey upgraded 1/1 packages. 
 See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).
Chocolatey v1.2.0

cabal has been installed as a side by side installation.
Side by side installations are deprecated and is pending removal in v2.0.0.
Installing the following packages:
make;gcc-g++;bison;flex;libxml2-devel;libiconv-devel;zlib-devel;liblzma-devel;wget;unzip;dos2unix;autoconf;automake;libtool;ragel;perl;perl-Regexp-Common;perl-Test-Harness;perl-Text-Diff;perl-YAML-Tiny;perl-Module-Build;perl-List-MoreUtils;perl-Params-Util;Perl-B-COW;Perl-Clone
By installing, you accept licenses for the packages.

Chocolatey installed 0/0 packages. 
 See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).
The system cannot find the file specified
Error: Process completed with exit code 1.
```

pauloscustodio, 2022-11-06:
This is problem is reported here https://github.com/egor-tensin/setup-cygwin/issues/9

jkroepke, 2022-11-07:
I have the same error, here is the error I got from the logs:

```
2022-11-07T22:35:19.3222967Z  See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).
2022-11-07T22:35:19.3223386Z 2022-11-07 22:35:07,866 4196 [DEBUG] - Sending message 'PostRunMessage' out if there are subscribers...
2022-11-07T22:35:19.3223687Z 2022-11-07 22:35:07,866 4196 [ERROR] - The system cannot find the file specified
2022-11-07T22:35:19.3224116Z 2022-11-07 22:35:07,897 4196 [ERROR] - More Details: System.ComponentModel.Win32Exception (0x80004005): The system cannot find the file specified
2022-11-07T22:35:19.3224416Z    at System.Diagnostics.Process.StartWithCreateProcess(ProcessStartInfo startInfo)
2022-11-07T22:35:19.3225014Z    at chocolatey.infrastructure.commands.CommandExecutor.execute_static(String process, String arguments, Int32 waitForExitInSeconds, String workingDirectory, Action`2 stdOutAction, Action`2 stdErrAction, Boolean updateProcessPath, Boolean allowUseWindow, Boolean waitForExit)
2022-11-07T22:35:19.3225565Z    at chocolatey.infrastructure.commands.CommandExecutor.execute(String process, String arguments, Int32 waitForExitInSeconds, String workingDirectory, Action`2 stdOutAction, Action`2 stdErrAction, Boolean updateProcessPath, Boolean allowUseWindow)
2022-11-07T22:35:19.3226040Z    at chocolatey.infrastructure.app.services.CygwinService.install_run(ChocolateyConfiguration config, Action`1 continueAction)
2022-11-07T22:35:19.3226552Z    at chocolatey.infrastructure.app.services.ChocolateyPackageService.perform_source_runner_function[T](ChocolateyConfiguration config, Func`2 function)
2022-11-07T22:35:19.3226948Z    at chocolatey.infrastructure.app.services.ChocolateyPackageService.install_run(ChocolateyConfiguration config)
2022-11-07T22:35:19.3227360Z    at chocolatey.infrastructure.app.runners.GenericRunner.run(ChocolateyConfiguration config, Container container, Boolean isConsole, Action`1 parseArgs)
2022-11-07T22:35:19.3227757Z    at chocolatey.infrastructure.app.runners.ConsoleApplication.run(String[] args, ChocolateyConfiguration config, Container container)
2022-11-07T22:35:19.3227961Z    at chocolatey.console.Program.Main(String[] args)
```

https://github.com/jkroepke/helm-secrets/actions/runs/3414624012/jobs/5682825355

feilipu, 2022-12-07:
[Reported that the project is dead](https://github.com/egor-tensin/setup-cygwin/issues/9#issuecomment-1329370279), and a [resolution is offered](https://github.com/cygwin/cygwin-install-action).

closed 2022-12-11 by pauloscustodio