z88dk/z88dk #2119: "Undefined symbol: pic" when using math lib for Amstrad CPC

ColinPitrat, 2022-11-04:
Hello,

I've been trying to do a simple program using floating points on Amstrad CPC but compiling even the simplest thing fails with:

```
z80/init_floatpack.asm:127: error: undefined symbol: pic
  ^---- (pic)
z80/init_floatpack.asm:182: error: undefined symbol: pic
  ^---- (pic)
```

For example the following:

```
#include <cpc.h>

main()
{
  double_t a;
  a = 2.0;
}
```

generates the error when compiled with:

```
zcc +cpc -lmz -I${INCLUDE_PATH} -o test.bin test.c
```

feilipu, 2022-11-05:
I think the firmware Pi and half Pi routines were [removed in this commit](https://github.com/z88dk/z88dk/commit/830e51579c9b1e2fd5f9552780c2f199148258a5), and replaced with other routines.

This means that the initialisation of the PIC (Pi Constant) location becomes irrelevant.
I'll remove it.

closed 2022-11-05 by feilipu

feilipu, 2022-11-05:
Please re-open if this is not actually a fix.

commit 16236302 (2022-11-05): cpcmath Pi constant - closes #2119