z88dk/z88dk #2121: z88dk-ticks: set register terminates ticks

raps500, 2022-11-05:
A "set" command terminates ticks if a nexti or stepi command was previously issued. Both release 2.2, on windows 10 64 bit and the nightly build z88dk-win32-20221103-b018259ae8-20290 fail in the same way.

`

D:\Projects\HEXCalc\T80_XO2_DL1414\firmware>c:\tools\z88dk-win32-2.2\bin\z88dk-ticks.exe -d fw_main.bin
                    di                                      ;[0000] f3

    $0000    ()> set hl $1234
                    ld      sp,$1ffe                        ;[0001] 31 fe 1f

    $0001    ()> reg
af $0040   bc $0000   de $0000   hl $1234   ix $0000    S:0 Z:1 H:0 P/V:0 N:0 C:0
af'$0040   bc'$0000   de'$0000   hl'$0000   iy $0000
pc $0001  [pc]$  31   sp $0000  [sp]$31F3

    $0001    ()> nexti
                    ld      hl,$0162                        ;[0004] 21 62 01

    $0004    ()> set hl $1235
100000000

D:\Projects\HEXCalc\T80_XO2_DL1414\firmware>
`

desertkun, 2022-11-05:
Thanks, I'll look into it.

desertkun, 2022-11-10:
Fixed by `7e380553dd1bbaf44768df616ff81c6c8e80fa3e`

closed 2022-11-10 by desertkun

raps500, 2022-11-11:
Great thanks ! it works now, and -x mapfile too !