z88dk/z88dk #2122: Build fails on Raspberry Pi OS (undefined reference to `std::filesystem::)

mrgw454, 2022-11-06:
Getting the following undefined reference error when trying to compile:

i@raspberrypi:~/source/z88dk $ ./build.sh
Checking if submodule dependencies are met...
Module dependencies are present
make -C src/sccz80 PREFIX=`pwd` install
make[1]: Entering directory '/home/pi/source/z88dk/src/sccz80'
install -m 755 sccz80 /home/pi/source/z88dk/bin/z88dk-sccz80
make[1]: Leaving directory '/home/pi/source/z88dk/src/sccz80'
make -C src/z80asm PREFIX=`pwd` PREFIX_SHARE=`pwd` install
make[1]: Entering directory '/home/pi/source/z88dk/src/z80asm'
g++ -o z88dk-z80asm src/c/reloc_code.o src/c/str.o src/c/libfile.o src/c/z80asm.o src/c/sym.o src/c/modlink.o src/c/dbg.o src/c/opcodes.o src/c/list.o src/c/zobjfile.o src/c/z80pass.o src/c/common.o src/c/scan.o src/c/strhash.o src/c/parse.o src/c/minunit.o src/c/directives.o src/c/symtab1.o src/c/codearea.o src/c/array.o src/c/module1.o src/c/class.o src/c/expr1.o src/c/alloc.o ../common/objfile.o ../common/die.o ../common/strutil.o ../common/fileutil.o ../common/option.o ../common/zutils.o ../common/dirname.o ../common/optparse.o ../../ext/regex/regfree.o ../../ext/regex/regcomp.o ../../ext/regex/regexec.o ../../ext/regex/regerror.o src/cpp/utils.o src/cpp/errors.o src/cpp/preproc.o src/cpp/args.o src/cpp/lstfile.o src/cpp/lex.o src/cpp/float.o src/cpp/main.o src/cpp/defines.o
/usr/bin/ld: src/cpp/utils.o: in function `expand_glob_1(std::set<std::filesystem::__cxx11::path, std::less<std::filesystem::__cxx11::path>, std::allocator<std::filesystem::__cxx11::path> >&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
utils.cpp:(.text+0x1dd8): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: utils.cpp:(.text+0x2148): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: utils.cpp:(.text+0x2154): undefined reference to `std::filesystem::status(std::filesystem::__cxx11::path const&)'
/usr/bin/ld: utils.cpp:(.text+0x216c): undefined reference to `std::filesystem::status(std::filesystem::__cxx11::path const&)'
/usr/bin/ld: utils.cpp:(.text+0x2228): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: utils.cpp:(.text+0x2314): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: utils.cpp:(.text+0x231c): undefined reference to `std::filesystem::__cxx11::path::has_root_directory() const'
/usr/bin/ld: utils.cpp:(.text+0x23a8): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: utils.cpp:(.text+0x2478): undefined reference to `std::filesystem::__cxx11::recursive_directory_iterator::recursive_directory_iterator(std::filesystem::__cxx11::path const&, std::filesystem::directory_options, std::error_code*)'
...
...
/usr/bin/ld: args.cpp:(.text+0x856c): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: src/cpp/args.o: in function `std::filesystem::__cxx11::path::operator=(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&)':
args.cpp:(.text._ZNSt10filesystem7__cxx114pathaSEONSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZNSt10filesystem7__cxx114pathaSEONSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x7c): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: args.cpp:(.text._ZNSt10filesystem7__cxx114pathaSEONSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZNSt10filesystem7__cxx114pathaSEONSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x170): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:59: z88dk-z80asm] Error 1
make[1]: Leaving directory '/home/pi/source/z88dk/src/z80asm'
make: *** [Makefile:142: bin/z88dk-z80asm] Error 2

Is there a directive like '-lstdc++fs' that may be needed?

Thanks for any help you can provide!

suborb, 2022-11-06:
I have a hunch, but can you check you've installed the dependencies? There's a list here: 

https://github.com/z88dk/z88dk/wiki/installation#building-from-sources

mrgw454, 2022-11-06:
Hello,

I thought I had completed those steps.  Here's the 'apt install' step
mentioned in the docs:

***@***.***:~/source $ sudo apt install build-essential bison flex
libxml2-dev subversion zlib1g-dev m4 ragel re2c dos2unix texinfo texi2html
gdb curl perl cpanminus ccache libboost-all-dev libmodern-perl-perl
libyaml-perl liblocal-lib-perl libcapture-tiny-perl libpath-tiny-perl
libtext-table-perl libdata-hexdump-perl libregexp-common-perl libclone-perl
libfile-slurp-perl
Reading package lists... Done
Building dependency tree
Reading state information... Done
bison is already the newest version (2:3.3.2.dfsg-1).
build-essential is already the newest version (12.6).
ccache is already the newest version (3.6-1).
cpanminus is already the newest version (1.7044-1).
curl is already the newest version (7.64.0-4+deb10u3).
dos2unix is already the newest version (7.4.0-1).
flex is already the newest version (2.6.4-6.2).
gdb is already the newest version (8.2.1-2).
libboost-all-dev is already the newest version (1.67.0.1+b1).
libcapture-tiny-perl is already the newest version (0.48-1).
libclone-perl is already the newest version (0.41-1+b1).
libdata-hexdump-perl is already the newest version (0.02-1).
libfile-slurp-perl is already the newest version (9999.26-1).
liblocal-lib-perl is already the newest version (2.000024-1).
libmodern-perl-perl is already the newest version (1.20180901-1).
libpath-tiny-perl is already the newest version (0.108-1).
libregexp-common-perl is already the newest version (2017060201-1).
libtext-table-perl is already the newest version (1.132-1).
libxml2-dev is already the newest version (2.9.4+dfsg1-7+deb10u5).
libyaml-perl is already the newest version (1.27-1).
m4 is already the newest version (1.4.18-2).
perl is already the newest version (5.28.1-6+deb10u1).
ragel is already the newest version (6.10-1).
re2c is already the newest version (1.1.1-1).
subversion is already the newest version (1.10.4-1+deb10u3).
texi2html is already the newest version (1.82+dfsg1-5).
texinfo is already the newest version (6.5.0.dfsg.1-4+b1).
zlib1g-dev is already the newest version (1:1.2.11.dfsg-1+deb10u2).
0 upgraded, 0 newly installed, 0 to remove and 128 not upgraded.

As for the perl modules, I believe those were taken care of as well.  Is
there a specific package you may think is missing I could check?

Thanks for your help!

-Ron

mrgw454, 2022-11-06:
Quick update,

libboost is 1.67 and my perl is at 5.28.  Both of these have versions a bit
lower than is required.  I apologize for not catching this sooner.

suborb, 2022-11-07:
No worries, I've just checked some of the auto builds and it looks like they're all using gcc 9 and above, so your initial guess is right, a `-lstdc++fs` looks like it's needed with gcc8.