z88dk/z88dk #2124: Simplest floating point program fails (hangs) on Amstrad CPC

ColinPitrat, 2022-11-07:
I'm trying to use floating points on Amstrad CPC with a simple program like:

```
#include <cpc.h>                                                                                                                                                                                                                
                                                                                                                                                                                                                                
main()                                                                                                                                                                                                                 
{                                                                                                                                                                                                                               
  double_t a;                                                                                                                                                                                                                   
  a = 0;                                                                                                                                                                                                                        
}
```

And compiling with:

```
zcc +cpc -lndos -lmz -zorg=16384 -I<include_path> -o quadratic.bin quadratic.c
```

The resulting executable hangs. I tried on 2 different emulators (Caprice32 and xcpc) to verify that it was not an emulator issue. I also tried with different models (464, 664, 6128). But it hangs in all cases.

Looking at the diff of the assembly with & without the assignment:

```
$ diff with_assign.asm without_assign.asm
6c6
< ;     Module compile time: Mon Nov  7 13:34:22 2022
---
> ;     Module compile time: Mon Nov  7 13:33:43 2022
125,137d124
<       push    bc
<       push    bc
<       push    bc
<       ld      hl,0    ;const
<       add     hl,sp
<       push    hl
<       ld      hl,0    ;const
<       call    l_int2long_u_float
<       pop     hl
<       call    dstore
<       pop     bc
<       pop     bc
<       pop     bc
```

I tried to trace this in the debugger in Caprice32 but couldn't find exactly what the issue is apart that it seems to be looping infinitely.

ColinPitrat, 2022-11-07:
Ah then maybe it's just a doc to be updated:
https://github.com/z88dk/z88dk/wiki/Platform---Amstrad-CPC#floating-point

feilipu, 2022-11-08:
Been playing a little with Caprice32 (nice emulator), and the call into space is fixed.

But, I get varying results with some math libraries. The CPC library produces a wildly inaccurate trigonometric result.
Is there a configuration mismatch between the default library entry points for cpcmath and the default machine type of the Caprice32 emulator?

```c
// zcc +cpc -lcpcfs -lmz -subtype=dsk -create-app -o main.dsk main.c
// zcc +cpc -lcpcfs -lm -subtype=dsk -create-app -o main.dsk main.c
// zcc +cpc -lcpcfs --math32 -subtype=dsk -create-app -o main.dsk main.c

// Correct number is 0.540302

// CPC

#include <cpc.h>
#include <math.h>
#include <stdio.h>

#pragma printf = "%s %c %d %f"     // enables %s, %c, %d, %f only

main()
{
    double_t a;
    a = 0;
    a = cos(a+1);
    printf("number is %f",a);
    while(1);
}
```
[image]

ColinPitrat, 2022-11-08:
I observed weird results with `pow` too when I tried to use it for computing square roots but I didn't dig yet.

feilipu, 2022-11-08:
> It seems with `-lm`, pow returns -0.000000 for every exponent apart from 1 for which it returns a large negative number.

I assume you mean with `-lmz`?

I just checked with both genmath (`-lm`) and with math32 (`--math32`) and they're producing nice tables of correct results.
Whereas `-lmz` does as you suggest.

ColinPitrat, 2022-11-08:
> I assume you mean with `-lmz`?

No I really mean `-lm`. But looking deeper into it, that's because I was missing an include. Adding:

```
#include <math/math_cpc.h>
```

solves the issue.

> I'm not convinced that anyone actually used CPCmath after it was written.

I'm new to z88dk. I see that there are some tests in the code but I didn't look deep yet on how they are implemented. Do you think it would be possible to have some tests running on a CPC emulator? Caprice32 supports that through printer output and breakpoints. This is used in TRSE to run some unit tests.

feilipu, 2022-11-08:
> I'm missing an include.

Adding `math.h` is enough, and then it automatically adapts to different targets and subtypes:

```c
#include <math.h>
```

suborb, 2023-04-22:
Looks like I'm going through a bit of a +cpc phase so I'm looking at this in a bit more depth.

I've tracked down a couple of problems with the library - `l_f48_ldexp` was modifying random bits of memory. The integer to float conversion was incorrect. So fp operations are no longer crashing, but I've still got an issue with the float -> integer conversion.

suborb, 2023-04-23:
Ironically, the simpler the program, the more unlikely it was to work - the bit of work memory we use for floating point operations was being paged out and replaced with ROM.

closed 2023-04-23 by suborb

suborb, 2023-04-24:
It looks like there's an issue generating some floating point constants, adding `-doublestr` as an option works around it. 

It's probably an exponent bias problem which I'll track down...

suborb, 2023-04-24:
I've added a general --math-cpc option to use the code extracted from the firmware.

Performance wise, it's not great with n-body coming at about 2.2 - so about 25% slower than bbc math

commit 956ae893 (2023-04-23): Fix #2124: Fix CPC maths (again)

A few problems were encountered:

- The fp48 interface requires float to take a long, but the code
  only handled an int
- The fp48 interface requires ifix to return a long, but the code
  only returned an int
- ftoa/ftoe were "special" for the CPC, swithc to generic
- Delete pow10 since it's not overly useful
- Firmware call pokery destroyed alt registers

But the main issue was the fa value was usually paged out when
fp routines were called (unless program size was sufficiently large)
so move fa to fixed address in high memory.