z88dk/z88dk #2126: .lis file address field

maziac, 2022-11-09:
Hi,

sorry to open a bug for this but there is no discussion forum.

I'm the author of [DeZog](https://github.com/maziac/DeZog) (Z80 debugger) which parses the z88dk list file to get file/line number associations to the asm source files.
A user brought up that the z88dk-z80asm list file format has changed that's why I raised the other issue about 4 and 6 digits.

Anyhow I have a question:

The address used in the .lis file is difficult to parse as it is not the real physical address but something else.

Would it be possible to use the physical address instead in that field.
Or could you explain what the address field really shows?

Here is a short snippet:
~~~
    13                          	ORG 0x8000
    14
    15                          label1:
    16   0000 00                nop
    17
    18   0001 3e05            label2:	ld a,5
~~~

Here the 0000 is actually the physical address 0x8000.

suborb, 2022-11-09:
> In my tests the symptoms file did not contain the physical address either:

Sorry, I meant the .map file. If you assemble with -debug then it will insert a label corresponding to each line in the source file.

Thus, you can read the .map file to get the address, then go to the source file to get the opcode/source assembly.

maziac, 2022-11-10:
OK, I see. Good hint.
That would mean in total I would need to analyze all 3 files:
1. The map file for the labels.
2. The sym file for the physical address and file/line <-> address association
3. The .lis file for the EQU labels (or can I get the EQUs somewhere else?) and the size of the data/opcode for a specific line

Seems feasible, but is still quite complex.
Do you know if this debug option worked already with the 1.x version or when this was introduced?

suborb, 2022-11-10:
I can only reproduce that if the EQU is neither used nor exported from a file which is a little oddity, but doesn't seem like it should cause any trouble.

maziac, 2022-11-10:
OK, I see. The EQU need to be used, then it appears in the map file.

I will try to change my parser to use the map file instead of the lis file. 
Seems to be really easier.

Thanks for the help.

closed 2022-11-10 by maziac

desertkun, 2022-11-10:
Random question, are you aware that z88dk-ticks and z88dk-gdb now support mi2 protocol and can be debuggable in Visual Studio Code?

maziac, 2022-11-11:
No, I wasn't aware of that.
I'll give it a try. If it works out fine I could drop support for z88dk for DeZog.

desertkun, 2022-11-11:
@maziac I've pushed an example repository https://github.com/speccytools/z88dk-gdb-vscode-test
Feel free to make a fork it doesn't work for you right away, and submit a pull request.

maziac, 2022-11-12:
Thanks. But... what  vscode extension is used?

maziac, 2022-11-12:
Here is how far I got:
1. the vscode executable is probably "Native Debug" by WebFreak. At least I was able to start the z88dk-gdb and it was able to connect to mame.
2. The mame command line i used was: ```./mame spectrum -resolution 640x480  -window  -debug -debugger gdbstub -debugger_port 1337```
3. To build the program I tried make/make:
~~~
> z88dk-gdb-ide-test % cmake .        
-- The C compiler identification is AppleClang 14.0.0.14000029
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /Volumes/SDDPCIE2TB/Projects/Z80/asm/dezog_issues/desertkun/z88dk-gdb-ide-test
bob@macmini z88dk-gdb-ide-test % make
[ 50%] Building C object CMakeFiles/idetest.dir/main.c.o
clang: warning: argument unused during compilation: '-debug' [-Wunused-command-line-argument]
[100%] Linking C executable idetest
ld: library not found for -lndos
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [idetest] Error 1
make[1]: *** [CMakeFiles/idetest.dir/all] Error 2
make: *** [all] Error 2
> z88dk-gdb-ide-test % 
~~~

But as you can see the make did not succeed.

Could you simply just the the project  including binaries, sources and map file.
At the moment I'm not interested in compiling/building issues, I just would like to debug.

desertkun, 2022-11-12:
> cmake . 
That's because you're not doing project generation from IDEs and doing it on your own, so cmake chosen default compiler, `cc`. See [this wiki article](https://github.com/z88dk/z88dk/wiki/CMake) on how to use cmake manually within z88dk, it's relatively new feature.

maziac, 2022-11-12:
Couldn't you simply zip your project together with the binaries and map files and attach it here.
Would make my life easier.

desertkun, 2022-11-12:
Can do, see `build/` subfolder. Hope mame works for you I haven't tested it for a while, my fork of Fuse with gdbstub on https://speccytools.org should work too. 

[z88dk-gdb-ide-test.zip](https://github.com/z88dk/z88dk/files/9996026/z88dk-gdb-ide-test.zip)

--

You also can build a +test platform and run it in z88dk-ticks, which is a gdb and terminal-only emulator in one package.
Here's pre-built +test platform, which won't work on speccy, but would work on ticks, just go into `build` folder and do `z88dk-ticks -d -x idetest.map idetest.bin`:

[z88dk-gdb-ide-test.zip](https://github.com/z88dk/z88dk/files/9996031/z88dk-gdb-ide-test.zip)