z88dk/z88dk #2128: nexti command terminates ticks

raps500, 2022-11-13:
The nexti command sometimes terminates ticks after reg or stepi.
This is on nigthly build z88dk-win32-20221111-71a8b6787a-20302

```
    $0135    (key_enter_func+3)> stepi
                    or      a                               ;[0138] b7

    $0138    (key_enter_func+6)>
                    jr      nz,key_enter_func_1             ;[0139] 20 00

    $0139    (key_enter_func+7)>
key_enter_func_1:
                    jp      main_loop_redraw                ;[013b] c3 ba 00

    $013b    (key_enter_func_1+0)>
main_loop_redraw:
                    ld      a,(input_buffer_len)            ;[00ba] 3a 09 1e

    $00ba    (main_loop_redraw+0)>
                    or      a                               ;[00bd] b7

    $00bd    (main_loop_redraw+3)>
                    jr      z,main_loop_redraw_stack        ;[00be] 28 0d

    $00be    (main_loop_redraw+4)>
main_loop_redraw_stack:
                    call    prepare_tos                     ;[00cd] cd db 00

    $00cd    (main_loop_redraw_stack+0)>
prepare_tos:
                    ld      hl,(ASTACK_PTR)                 ;[00db] 2a 02 1f

    $00db    (prepare_tos+0)>
                    ld      de,$1e1e                        ;[00de] 11 1e 1e

    $00de    (prepare_tos+3)>
                    call    copy_i32                        ;[00e1] cd b2 03

    $00e1    (prepare_tos+6)> re
The following commands could match, which one is ambiguous:
    reg
    restore
    restore_pc

    $00e1    (prepare_tos+6)> reg
af $0044   bc $001C   de $1E1E   hl $1F84   ix $0000    S:0 Z:1 H:0 P/V:1 N:0 C:0
af'$0040   bc'$0000   de'$0000   hl'$0000   iy $0000
pc $00E1  [pc]$  CD   sp $2000  [sp]$00D0

    $00e1    (prepare_tos+6)> nexti
28509

D:\Projects\HEXCalc\T80_XO2_DL1414\firmware>debug_fw.bat
```
```
    $0000    (reset_vector+0)> cont
Hit breakpoint 1: @0135 (key_enter_func+3)
                    ld      a,(input_buffer_len)            ;[0135] 3a 09 1e

    $0135    (key_enter_func+3)> stepi
                    or      a                               ;[0138] b7

    $0138    (key_enter_func+6)>
                    jr      nz,key_enter_func_1             ;[0139] 20 00

    $0139    (key_enter_func+7)>
key_enter_func_1:
                    jp      main_loop_redraw                ;[013b] c3 ba 00

    $013b    (key_enter_func_1+0)>
main_loop_redraw:
                    ld      a,(input_buffer_len)            ;[00ba] 3a 09 1e

    $00ba    (main_loop_redraw+0)>
                    or      a                               ;[00bd] b7

    $00bd    (main_loop_redraw+3)>
                    jr      z,main_loop_redraw_stack        ;[00be] 28 0d

    $00be    (main_loop_redraw+4)>
main_loop_redraw_stack:
                    call    prepare_tos                     ;[00cd] cd db 00

    $00cd    (main_loop_redraw_stack+0)>
prepare_tos:
                    ld      hl,(ASTACK_PTR)                 ;[00db] 2a 02 1f

    $00db    (prepare_tos+0)>
                    ld      de,$1e1e                        ;[00de] 11 1e 1e

    $00de    (prepare_tos+3)>
                    call    copy_i32                        ;[00e1] cd b2 03

    $00e1    (prepare_tos+6)> nexti
28509

D:\Projects\HEXCalc\T80_XO2_DL1414\firmware>
```

desertkun, 2022-11-13:
Current implementation of `nexti` for ticks is not perfect: since `nexti` is a step-over, it essentially just places a breakpoint after the call instruction. But that call might never return to caller for some reason, e.g. a stack got messed up. I should put a warning when ticks exits and there are unresolved temporary breakpoints.

desertkun, 2022-11-13:
I've pushed 8c86065e1e0688e2c6355a969d6d82f86a08a195 to do exactly that. Please make sure your stack is in good condition and there is no problems with your program.

here's an example of program that could repro the warning:

test.c:
```
void sub()
{
#asm
pop bc
#endasm
}

int main() {
        sub();
}
```

then
```
zcc +test -debug test.c
z88dk-ticks -d -x a.map a.bin
b main
c
ni
ni
```

closed 2024-05-11 by suborb