z88dk/z88dk #2131: "make test" fails due to missing include directory (-I)

danfe, 2022-11-17:
Doing `make test` for version 2.2 fails for me in the following way:
```
zcc +rc2014 -vn  -I../../framework -DNO_LOG_RUNNING -DNO_LOG_PASSED ../../framework/test.c main.c strchr.c strcmp.c stricmp.c strlcat.c strlcpy.c strncat.c strncmp.c strrev.c strrstr.c strstr.c -o test_rc2014.bin  -clib=new -subtype=basic -m
../../framework/test.c: line 7: file 'setjmp.h' not found
../../framework/test.c: line 8: file 'stdarg.h' not found
../../framework/test.c: line 9: file 'stdio.h' not found
gmake[4]: *** [Makefile:35: test_rc2014_CODE.bin] Error 1
gmake[4]: Leaving directory '/tmp/usr/ports/devel/z88dk/work/z88dk/test/suites/string'
```
The following patch to the `test/suites/make.config` file fixes it:
```patch
@@ -23,7 +23,7 @@ SOURCES = $(wildcard $(FRAMEWORK_DIR)/*.c)
 # Rules for building suites
 FRAMEWORK_DIR ?= ../../framework
 SOURCES = $(wildcard $(FRAMEWORK_DIR)/*.c)
-CFLAGS = -I$(FRAMEWORK_DIR)
+CFLAGS = -I$(FRAMEWORK_DIR) -I../../../include
 
 MACHINE = z88dk-ticks$(EXESUFFIX) -w 30
 ```
This is on FreeBSD, but I can't exactly see why it should be system-specific.

suborb, 2022-12-14:
Given that it's building in CI without this change, there must be something else going on - it escapes me at the moment as to the cause though.

feilipu, 2026-11-08:
Reopen if still current.

closed 2026-11-08 by feilipu