z88dk/z88dk #2132: z80asm - expression test sets value

feilipu, 2022-11-19:
Testing an expression in asm seems to set the value, so that on later tests the value is changed.

Resolved for me here https://github.com/z88dk/z88dk/commit/dd13b0b3d0fc0e8eeed44bbd0df5639324aefd02.
But it would be good to know why.

If `__IO_CF_8_BIT` is configured to `0x00` in the configuration, the first test will set it to `0x01`.

```asm
IF __IO_CF_8_BIT = 1
```
I also tried `==`, but the value is also set during the test

```asm
IF __IO_CF_8_BIT == 1
```
The [z80asm documentation](https://github.com/z88dk/z88dk/wiki/Tool---z80asm---expressions) seems to suggest that either form should be acceptable (and not have the side effect of setting the value).

pauloscustodio, 2022-11-26:
Could not reproduce the bug. Assembling this code:
```
	defb 0
IF __IO_CF_8_BIT = 1
	defb 1
ENDIF
IF __IO_CF_8_BIT = 1
	defb 2
ENDIF
```
with __IO_CF_8_BIT undefined results in a binary file with just 0x00. Assembling with ```-D__IO_CF_8_BIT``` , results in a binary file with 0x00,0x01,0x02.

Could you provide an example to exercise the error?

feilipu, 2022-11-28:
I've tracked this down a little more. There seems to be two different things in this issue.

1. The `rc2014-8085_clib.lib` library is [compiled with classic libraries](https://github.com/z88dk/z88dk/blob/master/libsrc/Makefile#L1824), before the `rc2014-8085_private.inc` file is [generated by newlib](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/Makefile#L60). So the classic library collects the "old" status of any configuration. The work around is to ensure that the headers are regenerated before compiling the library, by compiling twice. But this doesn't seem to be sufficient. I had worked this out previously, and had worked around it.
2. As far as I know it should be sufficient to regenerate the `rc2014-8085_clib.lib` to collect the correct setting of the configuration in the header file. But, this doesn't seem to work properly. So there's something else that requires a full rebuild of z88dk to make the newlib config change work.

feilipu, 2022-11-28:
This is the test file I'm using.

```c
// zcc +rc2014 -subtype=acia85 -O2 -m -D__CLASSIC -DAMALLOC -l_DEVELOPMENT/lib/sccz80/lib/rc2014/ff_85 main.c -o bug2132 -create-app

#include <unistd.h>
#include <stdlib.h>
#include <stdio.h>
#include <sys/compiler.h>

#include <../libsrc/_DEVELOPMENT/target/rc2014/config_rc2014-8085.h>

#include <_DEVELOPMENT/sccz80/lib/rc2014/ff.h>

// GLOBALS
static FATFS * fs;              /* Pointer to the filesystem object (on heap) */
                                /* FatFs work area needed for each volume */

/**
   @brief Main entry point.
   @param argc Argument count.
   @param argv Argument vector.
   @return status code
 */
int main(int argc, char ** argv)
{
    (void)argc;
    (void *)argv;

    fs = (FATFS *)malloc(sizeof(FATFS));                    /* Get work area for the volume */

    f_mount(fs, (const TCHAR*)"0:", 1);

    free(fs);

    return 0;
}
```
Even with the configuration for `__IO_CF_8_BIT = 0x01` in `rc2014-8085_private.inc`

```asm
defc __RC2014 = 1

defc __CPU_CLOCK = 7372800

defc __IO_DIO_PORT = 0x00
defc __IO_CF_8_BIT = 0x01
defc __IO_CF_PORT_BASE = 0x10
defc __IO_PIO_PORT_BASE = 0x20
defc __IO_ACIA_PORT_BASE = 0x80
defc __IO_SIO_PORT_BASE = 0x80
defc __IO_SIO_PORT_OFFSET_A = 0x00
defc __IO_SIO_PORT_OFFSET_B = 0x02
defc __IO_SIO_PORT_OFFSET_C = 0x00
defc __IO_SIO_PORT_OFFSET_D = 0x01
```

it generates this map file using a newly compiled `rc2014-8085_clib.lib`, with the correct `.inc` file, known because `__CPU_INTEL__` is correctly set.

```sh
__CPU_INTEL__                   = $0001 ; const, local, , ide_hard_reset, , _DEVELOPMENT/target/rc2014/driver/ide/asm/ide_hard_reset.asm:0
__IO_CF_8_BIT                   = $0000 ; const, local, , ide_hard_reset, , _DEVELOPMENT/target/rc2014/config_rc2014-8085_private.inc:40
__IO_PIO_IDE_CTL                = $0022 ; const, local, , ide_hard_reset, , _DEVELOPMENT/target/rc2014/config_rc2014-8085_private.inc:689
__IO_PIO_IDE_CONFIG             = $0023 ; const, local, , ide_hard_reset, , _DEVELOPMENT/target/rc2014/config_rc2014-8085_private.inc:690
__IO_PIO_IDE_RD                 = $0092 ; const, local, , ide_hard_reset, , _DEVELOPMENT/target/rc2014/config_rc2014-8085_private.inc:691
__IO_PIO_IDE_RST_LINE           = $0080 ; const, local, , ide_hard_reset, , _DEVELOPMENT/target/rc2014/config_rc2014-8085_private.inc:701
ide_rst_dly                     = $01EE ; addr, local, , ide_hard_reset, code_driver, _DEVELOPMENT/target/rc2014/driver/ide/asm/ide_hard_reset.asm:47
ide_rst_dly2                    = $01F4 ; addr, local, , ide_hard_reset, code_driver, _DEVELOPMENT/target/rc2014/driver/ide/asm/ide_hard_reset.asm:54
__CPU_INTEL__                   = $0001 ; const, local, , ide_wait_ready, , _DEVELOPMENT/target/rc2014/driver/ide/asm/ide_wait_ready.asm:0
__IO_CF_8_BIT                   = $0000 ; const, local, , ide_wait_ready, , _DEVELOPMENT/target/rc2014/config_rc2014-8085_private.inc:40
[... 36 lines cut ...]
__IO_PIO_IDE_LSB                = $0020 ; const, local, , 8255_read_block, , _DEVELOPMENT/target/rc2014/config_rc2014-8085_private.inc:687
__IO_PIO_IDE_MSB                = $0021 ; const, local, , 8255_read_block, , _DEVELOPMENT/target/rc2014/config_rc2014-8085_private.inc:688
__IO_PIO_IDE_CTL                = $0022 ; const, local, , 8255_read_block, , _DEVELOPMENT/target/rc2014/config_rc2014-8085_private.inc:689
__IO_PIO_IDE_CONFIG             = $0023 ; const, local, , 8255_read_block, , _DEVELOPMENT/target/rc2014/config_rc2014-8085_private.inc:690
__IO_PIO_IDE_RD                 = $0092 ; const, local, , 8255_read_block, , _DEVELOPMENT/target/rc2014/config_rc2014-8085_private.inc:691
__IO_PIO_IDE_RD_LINE            = $0040 ; const, local, , 8255_read_block, , _DEVELOPMENT/target/rc2014/config_rc2014-8085_private.inc:700
__IO_PIO_IDE_DATA               = $0008 ; const, local, , 8255_read_block, , _DEVELOPMENT/target/rc2014/config_rc2014-8085_private.inc:703
ide_rdblk                       = $0337 ; addr, local, , 8255_read_block, code_driver, _DEVELOPMENT/target/rc2014/device/8255/8080/8255_read_block.asm:23
__CPU_INTEL__                   = $0001 ; const, local, , ide_wait_drq, , _DEVELOPMENT/target/rc2014/driver/ide/asm/ide_wait_drq.asm:0
__IO_CF_8_BIT                   = $0000 ; const, local, , ide_wait_drq, , _DEVELOPMENT/target/rc2014/config_rc2014-8085_private.inc:40
__IO_PIO_IDE_ALT_STATUS         = $0016 ; const, local, , ide_wait_drq, , _DEVELOPMENT/target/rc2014/config_rc2014-8085_private.inc:715
__CPU_INTEL__                   = $0001 ; const, local, , ide_cache_flush, , _DEVELOPMENT/target/rc2014/driver/ide/asm/ide_cache_flush.asm:0
__IO_CF_8_BIT                   = $0000 ; const, local, , ide_cache_flush, , _DEVELOPMENT/target/rc2014/config_rc2014-8085_private.inc:40
__IO_PIO_IDE_COMMAND            = $000F ; const, local, , ide_cache_flush, , _DEVELOPMENT/target/rc2014/config_rc2014-8085_private.inc:711
__IDE_CMD_CACHE_FLUSH           = $00E7 ; const, local, , ide_cache_flush, , _DEVELOPMENT/target/rc2014/config_rc2014-8085_private.inc:728
sta_nodisk                      = $0377 ; addr, local, , asm_disk_status, code_driver, _DEVELOPMENT/target/rc2014/driver/diskio/8085/asm_disk_status.asm:28
dresult_error                   = $039C ; addr, local, , asm_disk_read, code_driver, _DEVELOPMENT/target/rc2014/driver/diskio/8085/asm_disk_read.asm:53
[... cut ...]
__IO_CF_8_BIT                   = $0000 ; const, local, , ide_read_sector, , _DEVELOPMENT/target/rc2014/config_rc2014-8085_private.inc:40
__IO_PIO_IDE_SEC_CNT            = $000A ; const, local, , ide_read_sector, , _DEVELOPMENT/target/rc2014/config_rc2014-8085_private.inc:706
__IO_PIO_IDE_COMMAND            = $000F ; const, local, , ide_read_sector, , _DEVELOPMENT/target/rc2014/config_rc2014-8085_private.inc:711
__IDE_CMD_READ                  = $0020 ; const, local, , ide_read_sector, , _DEVELOPMENT/target/rc2014/config_rc2014-8085_private.inc:722
```

feilipu, 2022-12-03:
Added a [commit ](https://github.com/z88dk/z88dk/commit/bd24465bd1a4a0b84b7e264c024c8797ab5d11fb)to address the first issue.

But the second issue is not fixed. The configuration change is properly available in the header files, but a rebuild of the `rc2014-8085_clib.lib` does not get the configuration changed in the example code classic build. Whereas the newlib example build is correct.

feilipu, 2022-12-04:
Something is being weirdly cached in library (re)builds in classic.

I've tried inserting a dummy definition `__IO_MY_TEST` into the newlib configuration files, and then testing it in one of the library assembly files. Building and rebuilding the library with a value toggle will correctly echo that toggle in the map file in some places, but NOT where it is being tested in the library file.

Below is snipped from the `bug2131.map` file (source see above), when built using the classic library.

```asm
__CPU_INTEL__                   = $0001 ; const, local, , ide_hard_reset, , _DEVELOPMENT/target/rc2014/driver/ide/asm/ide_hard_reset.asm:0
__IO_MY_TEST                    = $0000 ; const, local, , ide_hard_reset, , _DEVELOPMENT/target/rc2014/config_rc2014-8085_private.inc:41
__IO_PIO_IDE_CTL                = $0022 ; const, local, , ide_hard_reset, , _DEVELOPMENT/target/rc2014/config_rc2014-8085_private.inc:323

__IO_MY_TEST                    = $0001 ; const, public, , rc2014_crt_asm_m4, , /home/phillip/Z80/z88dk/lib/config/../../libsrc/_DEVELOPMENT/target/rc2014/config_rc2014-8085_public.inc:59
__IO_CF_8_BIT                   = $0000 ; const, public, , rc2014_crt_asm_m4, , /home/phillip/Z80/z88dk/lib/config/../../libsrc/_DEVELOPMENT/target/rc2014/config_rc2014-8085_public.inc:61
```

This means that even if the configuration files are being properly rebuilt, the new configuration files are not being read by the classic build.

@suborb I'm not sure where to look to fix this.
Perhaps somewhere in the classic `Makefile`?

suborb, 2022-12-04:
I can see we've got this:

```
rc2014-8085_clib.lib:
        @echo ''
        @echo '--- Building RC2014-8085 Library ---'
        @echo ''
        $(call buildgeneric,rc2014,portable)
        TARGET=rc2014 TYPE=8085 $(LIBLINKER) -DSTANDARDESCAPECHARS -m8085 -DFORrc2014 -x$(OUTPUT_DIRECTORY)/rc2014-8085_clib @$(TARGET_DIRECTORY)/rc2014/rc2014-8085.lst
```

and rc2014-8085.lst has this:

```
@_DEVELOPMENT/target/rc2014/device/device_8085_sccz80.lst
@_DEVELOPMENT/target/rc2014/driver/driver_8085_sccz80.lst
```

What I'm not seeing is a clean of those files within `_DEVELOPMENT`. 

It might be worth doing the same thing as we do for other bits of classic that pull in from newlib, assemble those files via makefile and store the object files out of the _DEVELOPMENT tree. You can then add on the dependency on the config file to trigger a rebuild.

feilipu, 2022-12-05:
> ```sh
> @_DEVELOPMENT/target/rc2014/device/device_8085_sccz80.lst
> @_DEVELOPMENT/target/rc2014/driver/driver_8085_sccz80.lst
> ```
> What I'm not seeing is a clean of those files within _DEVELOPMENT.

And manually deleting the generated `*.o` files confirms that this is indeed the issue.
I'll have a fumble around (weak make-fu) to try to fix it, but may end up passing it along.

feilipu, 2022-12-05:
Ok, it was not particularly flexible or smart to just delete the offending `*.o` files, to force them to be rebuilt.
But, it works.

Case closed.

closed 2022-12-05 by feilipu