z88dk/z88dk #2137: CP/M mode on Nascom

zx70, 2022-11-29:
I'm in trouble adapting the NASCOM libraries to CP/M
I'm trying to keep the same codebase but with a shifted display address (it should be 0xf800 rather than 0x800).
@suborb, please confirm that base_graphics is never initialized. 
I'm attempting to extend the makefile myself,  but a help would be welcome.   I understood the way it works on the MicroBee, but I'm still goofy on passing the right build options (-Dforcpm...)

suborb, 2022-11-29:
Unless I'm missing something, the +nascom target just uses gencon and characters to do graphics. `generic_console.asm` has this in it: `defc  DISPLAY = 0x0800`

Take a look at +homelab - for that the .cfg file is passing in `-pragma-export:CONSOLE_DISPLAY=0xc000` to adjust the location of the screen memory in generic_console.asm

zx70, 2022-11-29:
Thanks,  it helps a lot.
There was still a bit of code for the NASCOM here and there referring to base_graphics,  but it was probably rarely hit.

closed 2022-11-30 by zx70