z88dk/z88dk #2143: [info/bug?][mtx] 1-bit sound support?

Fabrizio-Caruso, 2022-12-18:
I have not understand if 1-bit sound is supposed to work in the MTX target as the doc is ambiguous about it (it is shown in the table but not in the target doc: https://github.com/z88dk/z88dk/wiki/Platform---Memotech-MTX)

Anyway, if I try it I get a continuous sound. So it seems broken.

[image]

suborb, 2022-12-18:
This is the old sn76489 issue. In theory there's a 1 bit emulation layer over the sound chip. In practice it only works for a very few machines.

As a result, it's marked as not present in `features.h`:  https://github.com/z88dk/z88dk/blob/master/include/features.h#L2223 - I'll update the table.

Fabrizio-Caruso, 2022-12-19:
I see:
```
__HAVE_PSG_SN76489 1
```
Does it mean that I should be able to produce some sounds with the PSG_SN76489 specific API's ?

I have not succeeded so far. I can produce a minimal example but I am not sure I am using the API's correctly.

feilipu, 2026-11-09:
psg_init writes the channel 1 silence byte and then reads the MTX latch port. MTX 1-bit sound uses that init. __HAVE_ONEBITSOUND stays off. Commit 7eaaabfed8.

closed 2026-11-09 by feilipu