z88dk/z88dk #2144: [bug?][mtx][psg] Either bug or bad emulator

Fabrizio-Caruso, 2022-12-19:
If I compile
```
#include <psg.h>

int main(void)
{
    unsigned char i;
    unsigned char j;

    psg_init();
    psg_volume(0, 15);
    //psg_channels(1, 0); // set first channel to tone generation, reset noise channels

    for(i=0;i<16;i++)
    {
        psg_tone(0, 500-(i*16));
        for(j=0;j<150;++j) {}
    }
    psg_volume(0, 0);
    return 0;
}
```
with
```
$ zcc.exe +mtx generic_psg.c -lndos -create-app
```
which I load with `LO.""` and run with `RUN` I get a crash

[image]

On the other hand if I do the same with `sms`, I get a correctly working rom image.

I am testing with MEMU emulator.

suborb, 2023-01-06:
Sorry @Fabrizio-Caruso I’ve not had a chance to get onto this one yet.

suborb, 2023-05-30:
Sorry for taking so long to get to this one...

I think you're testing on a MTX512, for a which a couple of commands are needed before loading the binary, see the section here: https://github.com/z88dk/z88dk/wiki/Platform---Memotech-MTX#quick-start

```
POKE 64122,0
NEW
LOAD ""
```

suborb, 2023-07-08:
Whilst you're here @Fabrizio-Caruso there's this one as well!

Fabrizio-Caruso, 2023-07-08:
Without the real hardware we cannot be sure. Let us assume Mame is accurate.

closed 2023-07-08 by Fabrizio-Caruso

zx70, 2023-07-14:
IIRC This is the way the expanded MTX512 had to be prepared also when running m/c commercial software.