z88dk/z88dk #2147: "common" module with shared state in library

empathicqubit, 2022-12-29:
I'm probably missing something, but is it possible to share a variable from one common C file to another file without also exporting the variable from the library?

desertkun, 2022-12-29:
As in any C program

file1.c
    int variable = 10.

file2.c
    extern int variable;
    variable = ...

empathicqubit, 2022-12-29:
The problem is that if I do that and I try to redefine it in the main program, it breaks because it's already defined

desertkun, 2022-12-29:
You define it in only one place, e. g. main program, everywhere else you declare it with extern.

empathicqubit, 2022-12-29:
I don't want to define internal state for my library in the main program.

desertkun, 2022-12-29:
Typical pattern of doing this

library.h

    extern int variable;

library.c

    int variable = 0;

main.c

    #include "library.h"
    // you have access to variable here

empathicqubit, 2022-12-29:
I don't want access in main. I want it completely invisible when the library gets linked, so that main can use that name without it clashing.

lst file:
```

src/gdb_set_step_toggle
src/gdb_set_get_char
src/gdb_set_put_char
src/gdb_step
src/gdb_set_hwbreak_toggle
src/gdb_nmi
src/gdb_print
src/gdb_set_swbreak_toggle
src/gdb_set_enter
src/lib
src/gdb_hwbreak
src/gdb_swbreak
src/gdb_exception
src/gdb_int
src/utils
```

src/lib.c:
```c
byte state[NUMREGBYTES];
```

src/lib.h:
```
extern byte state[NUMREGBYTES];
```

Then if I reference the resulting library in another project:

```LDFLAGS?=$(BASE_CFLAGS_DEBUG) -Llib/build -lgdb```

main.c:
```
char state = 'h';
```

This line breaks:
```build/src/main.o: error: duplicate definition: X_home_empathicqubit_z88dk_ti8xp_template_src_main_c::_state```

feilipu, 2022-12-29:
Also, if you need to know where something is, but want to hide or ignore the contents, FreeRTOS does [this](https://github.com/feilipu/z88dk-libraries/blob/master/freertos/source/portable/port.c#L42).

empathicqubit, 2022-12-29:
@feilipu Thanks. I'm not sure that's what I'm after though

empathicqubit, 2022-12-29:
Something like this: https://www.techopedia.com/definition/11663/internal-c

desertkun, 2022-12-29:
z88dk has a very primitive linker (it's just z80asm that does the linking), I am afraid there is no way to do that, that I am aware of. You can redirect the symbol, but not rename it.

you seem to be working on a gdserver? So it seems you want to guarantee that  the gdb library does not interfere with the main program in any way? You can force the users to leave a chunk of memory at pre-defined location for you, and then you can declare `static char __at(0xXXXX) variable;` in a header file of your library. `__at` doesn't support non-constants so you can't hide a variable in that fashion. What about coming up with a prefix (namespace) for all of your variables?

Also, take a look this [this project](https://github.com/speccytools/spectranet-gdbserver) its a gdbserver that uses spectranet

empathicqubit, 2022-12-29:
Thanks for linking that. Will definitely take a look. I based my library on Sergey's Z80 stub code in the main GDB project. It's basically working now as a non-library, but it's quite complicated because it has a bunch of #defines to turn different features on and off, and I've added another layer on top so that the user can pass in their own functions without worrying about the internals too much.

empathicqubit, 2022-12-29:
I have namespaced the variables I intended to export, I guess I was just hoping not to have to do it for every non-static in the project. I guess _gdb or something, idk.

desertkun, 2022-12-29:
Are you intending to make it compatible with z88dk-gdb? Just curious how is it supposed to work?

suborb, 2022-12-29:
So, C itself doesn't support proper namespaces which I think is what you actually need.

Everyone just ends up prefixing variables with a conceptual "module" name or manually mangling the name in such a way that it's unlikely to collide.

empathicqubit, 2022-12-29:
@desertkun You supply different functions for setting breakpoints and such, and the get and put char commands, and it does everything else. For me on a TI calculator, the get and put char send bytes over serial link, and for breakpoints I poke NOP holes in my program when I compile, and the break calls get put in those holes. The only thing that gets added into main.c is an initial call to break at the beginning of the program. I don't know if there's a better way to do that on my platform. I've read that the TI Z80 is pretty limited.

empathicqubit, 2022-12-29:
It currently works with z88dk-gdb. Currently it's off by one line, and very slow to step.

desertkun, 2022-12-29:
The only thing gdbserver is supposed to do, is provide access to memory/registers and place/trip breakpoints. z88dk-gdb should do the rest. 

How are you doing stepping? You follow every instruction with a rst?

If you can modify rom, you can just replace the current byte with `rst 08` ([see](https://github.com/speccytools/spectranet-gdbserver/blob/master/src/server.c#L293)), and upon execution of reset, you restore the original instruction. Only problem is, you have to tweak the ret address by one (to essentially undo one rst), and then somehow place the rst back after the original code is executed. That could be next to impossible, or you can simply accept breakpoints being "one time" break only. But stepping won't work this way if your hardware has no way to do an interrupt after one instruction. Spectranet has a special mechanism to trap on hardware level, when an instruction on a specific address has concluded execution.

empathicqubit, 2022-12-30:
@desertkun All the RSTs are part of flash so I don't think it's possible. I use a regular CALL, so three bytes, and the "holes"are between each line, not each instruction, because obviously between each instruction would probably get too heavy. It's based on whether the C_LINEs are there in the file, so if the debugging is turned off it should just not do it at all.

empathicqubit, 2022-12-30:
This isn't what we're talking about exactly, but this is the bridging application for the GDB server, if you're interested. It basically just looks at the packets to figure out who's turn it is, since the link is half duplex. I want to link you the other project, but it's still private. I wanted to figure out this library thing to isolate the GDB stub from the rest of the project, since it's based on GPL code. https://github.com/empathicqubit/ticables-gdb-bridge/blob/master/src/tibridge.c

empathicqubit, 2022-12-30:
The original code: https://sourceware.org/git/?p=binutils-gdb.git;a=blob;f=gdb/stubs/z80-stub.c;h=0ec128fbe6a67c1c6d59df62ee61c767ad96b2a2;hb=3dab9e15d3ab1cde555dfbf4a958eb17eec2ecb3

empathicqubit, 2023-01-01:
Happy new year!
The following code:
```c
  1 #include "../3rdparty/z88dk-gdbstub/src/gdb.h"
  2 #include "../lib/debug-impl.h"
  3 #include <conio.h>
  4 
  5 unsigned char main (void) {
  6     // This is needed to set up the debugger
  7     // To disable it, comment out GDB=1 in the Makefile
  8     // With this option can leave debug.h references in your code
  9     // If debugging is enabled, you MUST attach a debugger to continue
 10     puts("running");
 11     debug_init();
 12     gdb_swbreak();
 13 
 14     puts("hello world");
 15     puts("goodbye world");
 16 
 17     while(1);
 18 
 19     return 0;
 20 }
```
Produces this output:
```asm
; Function main flags 0x00000000 __stdc 
; unsigned char main()
	C_LINE	5,"/home/empathicqubit/z88dk-ti8xp-template/src/main.c::main::0::1"
._main
	call	l_debug_push_frame
	C_LINE	5,"/home/empathicqubit/z88dk-ti8xp-template/src/main.c::main::0::1"
	C_LINE	6,"/home/empathicqubit/z88dk-ti8xp-template/src/main.c::main::1::2"
	C_LINE	7,"/home/empathicqubit/z88dk-ti8xp-template/src/main.c::main::1::2"
	C_LINE	8,"/home/empathicqubit/z88dk-ti8xp-template/src/main.c::main::1::2"
	C_LINE	9,"/home/empathicqubit/z88dk-ti8xp-template/src/main.c::main::1::2"
	C_LINE	10,"/home/empathicqubit/z88dk-ti8xp-template/src/main.c::main::1::2"
	C_LINE	10,"/home/empathicqubit/z88dk-ti8xp-template/src/main.c::main::1::2"
	C_LINE	11,"/home/empathicqubit/z88dk-ti8xp-template/src/main.c::main::1::2"
	ld	hl,i_1+0
	push	hl
	call	puts
	pop	bc
	C_LINE	11,"/home/empathicqubit/z88dk-ti8xp-template/src/main.c::main::1::2"
	C_LINE	11,"/home/empathicqubit/z88dk-ti8xp-template/src/main.c::main::1::2"
	call	_debug_init
	C_LINE	12,"/home/empathicqubit/z88dk-ti8xp-template/src/main.c::main::1::2"
	C_LINE	12,"/home/empathicqubit/z88dk-ti8xp-template/src/main.c::main::1::2"
	call	_gdb_swbreak
	C_LINE	14,"/home/empathicqubit/z88dk-ti8xp-template/src/main.c::main::1::2"
	C_LINE	14,"/home/empathicqubit/z88dk-ti8xp-template/src/main.c::main::1::2"
	C_LINE	15,"/home/empathicqubit/z88dk-ti8xp-template/src/main.c::main::1::2"
	ld	hl,i_1+8
	push	hl
	call	puts
	pop	bc
	C_LINE	15,"/home/empathicqubit/z88dk-ti8xp-template/src/main.c::main::1::2"
	C_LINE	15,"/home/empathicqubit/z88dk-ti8xp-template/src/main.c::main::1::2"
	C_LINE	16,"/home/empathicqubit/z88dk-ti8xp-template/src/main.c::main::1::2"
	ld	hl,i_1+20
	push	hl
	call	puts
	pop	bc
	C_LINE	17,"/home/empathicqubit/z88dk-ti8xp-template/src/main.c::main::1::2"
	C_LINE	17,"/home/empathicqubit/z88dk-ti8xp-template/src/main.c::main::1::2"
.i_2
	jp	i_2	;EOS
.i_3
	C_LINE	19,"/home/empathicqubit/z88dk-ti8xp-template/src/main.c::main::1::2"
	C_LINE	19,"/home/empathicqubit/z88dk-ti8xp-template/src/main.c::main::1::2"
	ld	hl,0	;const
	call	l_debug_pop_frame
	ret
```
The above is unmodified. You can see with some lines it wants to generate the correct line number then puts the next one after it:
```asm
	C_LINE	14,"/home/empathicqubit/z88dk-ti8xp-template/src/main.c::main::1::2"
	C_LINE	14,"/home/empathicqubit/z88dk-ti8xp-template/src/main.c::main::1::2"
	C_LINE	15,"/home/empathicqubit/z88dk-ti8xp-template/src/main.c::main::1::2"
	ld	hl,i_1+8
	push	hl
	call	puts
	pop	bc
```

I've been trying to figure out why it does it for some lines but not others, but I'm not really sure yet

desertkun, 2023-01-01:
Line number generation has always been a struggle. In particular, resolving an address to a line. The ultimate algorithm z88dk-gdb ended up with, I think, is to resolve address to a last C_LINE that points to it. So break on `ld hl, 99` would resolve to line 7. That seems incorrect, however putting breakpoint at line 6 still works and the problem is only with address-to-line and not line-to-address. After after several attempts of going back and forth, this is something I ended up with.

Also, as of your "blanks" and stuff. As long as you're willing to accept not being able to break on library function calls,  you might be better off just modifying the `l_debug_push_frame`, as it's being called on every function entrance with `-debug` flag.

empathicqubit, 2023-01-05:
That might be a good idea to replace the entry point. I'll have to think about that.

empathicqubit, 2023-01-05:
The full example is here: https://github.com/empathicqubit/z88dk-ti8xp-template

The TI-specific parts of the debugger are in here: https://github.com/empathicqubit/z88dk-ti8xp-template/blob/master/lib/debug-impl.c

It actually works much faster than I originally experienced. Something about my VM was jammed up and it was transferring at several dozen baud. Not the calc's fault but the host? Turning it all off and on again seemed to fix it.