z88dk/z88dk #2148: Resetting flags operant does not work on sccz80

desertkun, 2023-01-03:
main.c
```
#include "stdio.h"
#define THE_FLAG (0x10)
int main()
{
	int a = 0;
	a |= THE_FLAG;
	a &= ~THE_FLAG;
	printf("the flag = %d\n", a);
}
```
I have no idea what's causing this, but it's something somewhat recent. My program stopped working out of sudden for no reason and during extensive debugging I would the `x &= ~(value)` does nothing to the `x`.
```
zcc +test -m -debug -create-app -lndos main.c
z88dk-ticks a.bin
```
Prints out `the flag = 16` while gcc reports 0.

desertkun, 2023-01-04:
Looking at disassembly,
```
    $0131    (_main+0)> lis
     1: #include "stdio.h"
     2: #define THE_FLAG (0x10)
     3: int main()
>    4: {
     5: 	unsigned char c = ~THE_FLAG;
     6: 	printf("the flag = %d", c);
     7: }

    $0131    (_main+0)> dis
_main:
                    call    l_debug_push_frame              ;[0131] cd 36 06
                    dec     sp                              ;[0134] 3b
                    pop     hl                              ;[0135] e1
                    ld      l,$ff                           ;[0136] 2e ff
                    push    hl                              ;[0138] e5
                    ld      hl,$06d9                        ;[0139] 21 d9 06
                    push    hl                              ;[013c] e5
                    ld      hl,$0002                        ;[013d] 21 02 00
                    add     hl,sp                           ;[0140] 39
                    ld      l,(hl)                          ;[0141] 6e
```
Seems like the constant is generated incorrectly.

Preprocessing works okay:
```
unsigned char c = ~ (0x10) ;
```

While sccz80 fails miserably:
```
	call	l_debug_push_frame
	C_LINE	4,"main.c::main::0::1"
	C_LINE	5,"main.c::main::1::2"
	C_LINE	5,"main.c::main::1::2"
	dec	sp
	pop	hl
	ld	l,255
	push	hl
	C_LINE	6,"main.c::main::1::2"
	C_LINE	6,"main.c::main::1::2"
	C_LINE	7,"main.c::main::1::2"
```

suborb, 2023-01-04:
However, on Linux arm64 I can.

Wonderful....

suborb, 2023-01-04:
I've been here before, try removing ` _M_ARM64` from here: https://github.com/z88dk/z88dk/blob/master/src/sccz80/define.h#L14

desertkun, 2023-01-04:
Thank you, it does fix it. I've pushed a direct commit 42bd07561f29ced482b79d8d8a6e44b5de67570f

suborb, 2023-01-20:
@desertkun lets have the conversation here.

The fix here broke the test for #543

desertkun, 2023-01-20:
I guess even if I wanted to I can't simply update the test since then the intel cpus etc will fail?

suborb, 2023-01-20:
Correct, it's a bug. Somewhere I've assumed the wrong thing with type conversions..

suborb, 2023-01-21:
Test is broken on armhf

desertkun, 2023-01-21:
On top of that, doing `./build.sh -t` on arm64 fails here:

```
Starting suite Right shift Tests (9 tests)
Running test test_rshift16_const.....failed rshift.c:146 (>>1)
Running test test_rshift16_var.....failed rshift.c:14 (>>1)
9 run, 7 passed, 2 failed
```

suborb, 2023-01-21:
Arghh. That suite isn’t run as part of the snap build.

suborb, 2023-01-22:
Curiously, I get failures on test_rshift64_const_unsigned on my rpi4

suborb, 2023-01-22:
The ticks tests are now running in snapcraft and the arm64 failure there matches my failure:

```
 10:21:42  Test completed 
zcc +test -vn  -I../../framework -DNO_LOG_RUNNING -DNO_LOG_PASSED ../../framework/test.c rshift.c -o test_rshift.bin -lndos  -m
 10:21:46  Starting test 
Starting suite Right shift Tests (9 tests)
Running test test_rshift64_const_unsigned.....failed rshift.c:462 (>>0)
9 run, 8 passed, 1 failed
```

Will update with the armhf failure when that finishes.

suborb, 2023-01-23:
I should be able to have access to an M1 Mac later on this week so I can continue debugging.

suborb, 2023-01-24:
Wow..I know it's been a while since I used recent hardware, but the M1 performance is pretty outstanding.

So, the actual issue here is fixed, but I spy another one:

```
rshift.c:525:31: warning: On this host, 64 bit constants may not be correct
```

Which doesn't happen on Linux/arm64

We use a `long double` to hold constant values, falling back to a double means that we're missing 10/11 bits of mantissa precision for a lossless conversion to a `uint64_t/int64_t`

Given how little 64 bit values are used, it's not massively important

desertkun, 2023-01-24:
If you have access to M1 only temporarily, I can spin up a virtual box vm on my m1 with remote access.

suborb, 2023-01-24:
Thanks, but I should be good now - I picked up a cheap secondhand mini as a late birthday present to myself.

commit c7dc4eab (2023-01-21): Merge pull request #2165 from z88dk/type_issues

Fix #2148 and keep #543 working