z88dk/z88dk #2154: "Full" Windows path with backslashes confuses #include ""

empathicqubit, 2023-01-09:
```
/c/z88dk/bin/zcc +ti8x -Wall -debug -g -O0 -S -o "build/src/main.i.asm" "Z:\UserProfiles\EmpathicQubit\tyzzy\src\main.c"

Z:\UserProfiles\EmpathicQubit\tyzzy\src\main.c: line 1: file '../3rdparty/z88dk-gdbstub/src/gdb.h' not found
Z:\UserProfiles\EmpathicQubit\tyzzy\src\main.c: line 2: file '../lib/debug-impl.h' not found
make: *** [Makefile:189: build/src/main.i.asm] Error 1
```

But this works:

```
/c/z88dk/bin/zcc +ti8x -Wall -debug -g -O0 -S -o "build/src/main.i.asm" "Z:\UserProfiles\EmpathicQubit\tyzzy\src/main.c"
```

Note the forward slash in the last path part

suborb, 2023-01-11:
Just from code inspection I spot:

https://github.com/z88dk/z88dk/blob/master/src/ucpp/cpp.c#L679
https://github.com/z88dk/z88dk/blob/master/src/ucpp/cpp.c#L684

I think both the MSDOS and the non-AMIGA case need to be activated.

empathicqubit, 2023-01-11:
so `if defined(MSDOS) || defined(WIN32)` ?

empathicqubit, 2023-01-11:
No, actually there would need to be a third option for Windows since both / and \ can work

PR #2161 (2023-01-14): Fix path separator on Windows. Correctly clear symbol memory.
Fixes #2154 .
closed 2023-01-14 by suborb