z88dk/z88dk #2156: The cygwin build in GitHub Actions takes forever

empathicqubit, 2023-01-11:
You are probably already aware of this, but the cygwin build is very slow. There is one still running after two hours. With that in mind I think it would be better not to run it on every PR.

desertkun, 2023-01-11:
Kind of defeats the purpose. The idea is to make sure everything compiles there.

empathicqubit, 2023-01-11:
I had mentioned it because you ignored it before merging. Kind of defeats the purpose if you merge before it finishes.

Could you also try doing something like this? It will cut down on duplicate builds: https://github.com/python/cpython/pull/98009/files

empathicqubit, 2023-01-11:
Okay, actually I read that wrong. Sorry. That one is on master.

empathicqubit, 2023-01-11:
Still take a look at the concurrency stuff. I can add it if it makes sense to you.

desertkun, 2023-01-11:
Either way that's something for @suborb to look into since I can only manage z88dk-gdb/ticks and maybe some sccz80, but that's it.

closed 2023-01-13 by empathicqubit