z88dk/z88dk #2163: SQRT of negative numbers

zx70, 2023-01-20:
In C the result of SQRT of negative numbers is often forced to zero.
It appears to be the preferred behavior, e.g. it fixes this: https://github.com/z88dk/z88dk-ext/blob/master/graphics/hat.c

The default high precision math library works fine, the target specific ones and the alternative ones might require a calibration.

feilipu, 2023-01-20:
For `math32` the `sqrt()` of negative numbers is a [NaN](https://en.wikipedia.org/wiki/NaN), which is as designed.

But I'm not sure NaNs are handled consistently by libraries and compilers if provided as input to a function chain. I'll need to think about that one.

zx70, 2023-01-20:
I had to do something for the Sinclair ROMs, it simply interrupted the runtime by jumping to BASIC and giving an "Invalid argument" error.
The example program was originally for the MS BASIC which IMO put zero as well.

feilipu, 2023-01-21:
Well a bit of a rough check through shows that NaN values are not generated, except for the `sqrt(-n)` already discussed.

The `log(0)` or `log(-n)` should return a NaN, but is [currently returning negative infinity](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/math/float/math32/c/m32_logf.c#L58). Perhaps that could be fixed to return a NaN?

There is no NaN included in the [C header files to return](https://github.com/z88dk/z88dk/blob/master/include/math/math_math32.h#L30). Perhaps that should be added (below for IEEE754), and then returned as needed?

```asm
#define NAN_POS_F32    ((unsigned long)0x7FFFFFFF)
#define NAN_NEG_F32    ((unsigned long)0xFFFFFFFF)
```

The `acos()` and `asin()` functions should return a NaN when the operator is greater than 1. This could be tested and added too.

But otherwise the `fsclassify()` function is properly implemented, so if that is used from C then things can work as intended.

feilipu, 2023-01-21:
Resolved some weaknesses with #2166.

closed 2023-03-30 by zx70