z88dk/z88dk #2169: Extend actions pipeline to build on different architectures

suborb, 2023-01-22:
I'm currently having some "fun" with arm behaviour differences. The snap build does build on armhf,arm64,ppc64el but isn't obviously integrated into build status.

There's a project here: https://github.com/marketplace/actions/run-on-architecture which does allow building on different architectures. Unfortunately it does use qemu so may be too slow, but it's probably worth a try.

dfandrich, 2024-03-26:
There is also https://github.com/cross-platform-actions/action, which also uses QEMU (on Linux).

suborb, 2024-05-11:
I'm going to put this one on ice - I tried a multi-target docker build but it just was just too slow to be viable.

closed 2024-05-11 by suborb

dfandrich, 2024-05-11:
Does the speed really matter if it's running in the background on a CI server somewhere?

suborb, 2024-05-11:
There’s a job limit of 6 hours. 

If I remember correctly, the docker job I tried had almost finished building the binaries after 2. On a regular Linux build building the binaries is <10% of the build time.

suborb, 2024-05-11:
I remembered slightly incorrectly, it was this build: https://github.com/z88dk/z88dk/actions/runs/8862460206/job/24335438236

So 2h25m and up to 'G' in the libraries.