z88dk/z88dk #2173: 8085 optimisation rules break code

suborb, 2023-01-23:
I've got an md5 program that I've been using for years for benchmarking purposes.

Unfortunately it doesn't work on the 8085 target anymore:

**Correct result:**

```
zcc +test md5sum.c -clib=8085 -O0
md5sum.c:369:28: warning: Converting integer type to pointer witout cast. From unsigned int  to char * fmt*  [-Wincompatible-pointer-types]
md5sum.c:369:31: warning: Assigning 'v', type: unsigned char * v from char * fmt*  [-Wincompatible-pointer-types]
dom@ermintrude md5 % z88dk-ticks -w 30 -m8085 a.bin -- testfile.bin
argc 2
Fname: testfile.bin
996d17ca7b233f87a94325b21762dc35
Ticks: 60963950
```

**Bad result:**

```
dom@ermintrude md5 % zcc +test md5sum.c -clib=8085
md5sum.c:369:28: warning: Converting integer type to pointer witout cast. From unsigned int  to char * fmt*  [-Wincompatible-pointer-types]
md5sum.c:369:31: warning: Assigning 'v', type: unsigned char * v from char * fmt*  [-Wincompatible-pointer-types]
dom@ermintrude md5 % z88dk-ticks -w 30 -m8085 a.bin -- testfile.bin
argc 2
Fname: testfile.bin
b92937e0887e80e1e851d1fa39ebe383%
```

I'm pretty certain it's some of the rules in z80rules.1 that are causing problems.

[md5sum.c.txt](https://github.com/z88dk/z88dk/files/10483506/md5sum.c.txt)

desertkun, 2023-01-23:
[l_gintXsp](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/l/sccz80/9-common/l_gint4sp.asm) functions shift by X+2 amount of stack. That' because at that point, stack has moved back by 2 due to the fact it's being within l_gint function. Could that be the problem? What happens when you ditch the *sp shortcuts?

suborb, 2023-01-23:
I don't think it's anything to do with your rules - my hunch is that it's the `ld hl,(de)` rules and that `de` is actually in use at the call sites.

feilipu, 2023-01-23:
> I don't think it's anything to do with your rules - my hunch is that it's the ld hl,(de) rules and that de is actually in use at the call sites.

Just to confirm this is with the current rule set, not including the PR currently open?

Assuming yes. That is something that I was afraid of. I couldn't find anything in sccz80 code that definitively kept `de` or `bc` free during `l_gint` or `l_gchar` calls. I had to rely on reading a lot of compiled assembly looking for examples. It is good that you've found an example so we can understand this.

[I had previously found `de` in use during unwinding of the stack, which is why a previous attempt to use `ld de,sp+n` to unwind the stack frame failed.]

The question is then, under what circumstances will sccz80 hold an `int` or `char` in the secondary variable rather than pushing it onto the stack?

Or is there a way to inform sccz80 that the secondary register is always alive, and not to occupy it during calls?

feilipu, 2023-01-23:
Just digging around, and found rules for the [8085_rules.1](https://github.com/z88dk/z88dk/blob/master/lib/arch/8085/8085_rules.1) too, which seem to only be used for one target.
So probably this should be sorted out too.

feilipu, 2023-01-24:
> I've just done a quick side-by-side comparison (-O2 vs -O0), and it looks like an addition has gone missing which might explain the result.

Very odd that a whole section of code is missing, with nothing to replace the addition. Almost looks like there's a broken rule syntax somewhere that is capturing the addition, and not actually replacing it with "optimised" code.

feilipu, 2023-01-24:
I'm not at my usual place, so debugging is a bit challenging, but comparing two builds produces some strange things.

1. only the test target is using the `arch/8085/8085_rules.1` file.
2. comparing `-O0` with `-O2` means that with `-O2` the `z80_rules.1` is being run before the `8085_rules.1` file, so there are different 8085 optimisations being done.

Just focussing on some simple issues, in the `-O0` build there's some strange optimisation around `.i_12`.
Where `pop bc` is being done, although it is not being used in the section of code.
```asm
   179                          .i_12
   179  0337  3804              	ld	de,sp + 4
   179  0339  ed                	ld	hl,(de)
   179  033a  23                	inc	hl
   179  033b  d9                	ld	(de),hl
   179  033c  2b                	dec	hl
   179  033d  210200            	ld	hl,2	;const
   179  0340  39                	add	hl,sp
   179  0341  cd0000            	call	l_gint	;
   179  0344  010400            	ld	bc,4
   179  0347  09                	add	hl,bc
   179  0348  d1                	pop	de
   179  0349  c1                	pop	bc
   179  034a  e5                	push	hl
   179  034b  d5                	push	de
   179                          .i_14
```

The `-O2` build (that ran `z80_rules.1` first) doesn't do that, but rather recovers a long from the stack.
```asm
   179                          .i_12
   179  024f  3804              	ld	de,sp+4
   179  0251  d5                	push	de
   179  0252  ed                	ld	hl,(de)	;l_gint
   179  0253  23                	inc	hl
   179  0254  d1                	pop	de
   179  0255  d9                	ld	(de),hl	;l_pint
   179  0256  d1                	pop	de
   179  0257  e1                	pop	hl
   179  0258  010400            	ld	bc,4
   179  025b  09                	add	hl,bc
   179  025c  e5                	push	hl
   179  025d  d5                	push	de
   179                          .i_14
```
Odd.

With such a lot of code, it is a bit difficult to see into specifics.
But I'll have a detailed look at the weekend (back at home), and slowly over the next few days.

suborb, 2023-01-24:
Those bits of code look equivalent - increment sp+4 and add 4 to sp+2.

My example above is the result of 0 >> 13 and adding to a long. Which results in a noop.

So there's somewhere else that's wrong.

When you get back, it might be worth running on real hardware - we could have an emulator issue.

suborb, 2023-01-24:
Well:

```
zcc +test -clib=8080 md5sum.c
md5sum.c:369:28: warning: Converting integer type to pointer witout cast. From unsigned int  to char * fmt*  [-Wincompatible-pointer-types]
md5sum.c:369:31: warning: Assigning 'v', type: unsigned char * v from char * fmt*  [-Wincompatible-pointer-types]
dom@ermintrude md5 % z88dk-ticks -m8080 -w 30 a.bin -- testfile.bin
argc 2
Fname: testfile.bin
996d17ca7b233f87a94325b21762dc35
Ticks: 57814930
dom@ermintrude md5 % z88dk-ticks -m8085 -w 30 a.bin -- testfile.bin
argc 2
Fname: testfile.bin
d41d8cd98f00b204e9800998ecf8427e%
```

I wasn't expecting the 8080 binary not to work on an 8085

suborb, 2023-01-24:
It's the flag calculation in ticks that's causing the trouble: https://github.com/z88dk/z88dk/blob/master/src/ticks/ticks.c#L645

suborb, 2023-01-24:
There were some push/pop af down in the fcntl handling. Which means the return address was slightly off, so skipping out bits of the algorithm.

closed 2023-01-24 by suborb