z88dk/z88dk #2175: Arm64 macOS doesn't support long double

suborb, 2023-01-24:
Which results in a bunch of warnings when using 64 bit integers in sccz80:

Wow..I know it's been a while since I used recent hardware, but the M1 performance is pretty outstanding.

So, the actual issue here is fixed, but I spy another one:

```
rshift.c:525:31: warning: On this host, 64 bit constants may not be correct
```

This is an apple thing, since floats larger than 64 bits are available on Linux/arm64

We use a long double to hold constant values, falling back to a double means that we're missing 10/11 bits of mantissa precision for a lossless conversion to a uint64_t/int64_t

Given how little 64 bit values are used, it's not massively important, but I should fix it.

suborb, 2023-10-05:
Also related, to recent armhf failures on snap craft:

```
zcc +test -vn  -I../../framework -DNO_LOG_RUNNING -DNO_LOG_PASSED ../../framework/test.c mult.c -o test_mult.bin  -m
mult.c:37:50: warning: On this host, 64 bit constants may not be correct
 [-Wlimited-range]
mult.c:53:33: warning: On this host, 64 bit constants may not be correct
 [-Wlimited-range]
mult.c:54:33: warning: On this host, 64 bit constants may not be correct
 [-Wlimited-range]
mult.c:56:29: warning: On this host, 64 bit constants may not be correct
 [-Wlimited-range]
mult.c:57:29: warning: On this host, 64 bit constants may not be correct
 [-Wlimited-range]
 06:39:09  Starting test 
Starting suite Multiplication Tests (7 tests)
Running test test_mult_unsigned_longlong.....failed mult.c:53 (3 * 0x2AAAAAAAAAAAAAAB)
7 run, 6 passed, 1 failed

Ticks: 579522
 06:39:09  Test failure 
```

feilipu, 2026-11-08:
Fixed in 0798471ae6 (`80cc: preserve exact integer literal semantics for both compilers (#2175)`) and 98541fc53e (`sccz80 - keep a long long multiply constant at full width`).

An integer literal is accumulated in a `uint64_t` and kept in `int_const_val`. A constant wider than a 53-bit mantissa is not stored in the host `long double`. The warning "On this host, 64 bit constants may not be correct" is gone.

`3 * 0x2AAAAAAAAAAAAAAB` loads `ab aa aa aa aa aa aa 2a` and multiplies with `l_i64_mult_u`. `x * 4294967297ULL` loads `01 00 00 00 01 00 00 00`. `test/suites/sccz80/literal.c` and `mult.c` cover both.

closed 2026-11-08 by feilipu