z88dk/z88dk #2178: Optimisation hunt

desertkun, 2023-01-26:
So I made this script [cosize.py.txt](https://github.com/z88dk/z88dk/files/10512881/cosize.py.txt) that detect duplicate assembly

Used like so:
```
z88dk-dis -x a.map -o 32768 a.bin > dump.txt && python3 tools/cosize.py dump.txt > top.txt
```

I wonder if you can help me look at "top hits" on my project and see if a few of them can be further optimized:

```
148 x 4 = 592, 4 bytes, 592 overall:
1; pop     bc
1; pop     hl
1; push    hl
1; push    bc

140 x 4 = 560, 4 bytes, 560 overall:
1; add     hl,sp
1; ld      l,(hl)
2; ld      h,$00

67 x 7 = 469, 7 bytes, 469 overall:
2; ld      h,$00
1; dec     sp
1; ld      a,l
1; pop     hl
1; ld      l,a
1; push    hl

68 x 6 = 408, 6 bytes, 408 overall:
2; ld      h,$00
1; dec     sp
1; ld      a,l
1; pop     hl
1; ld      l,a

73 x 5 = 365, 5 bytes, 365 overall:
1; dec     sp
1; ld      a,l
1; pop     hl
1; ld      l,a
1; push    hl

90 x 4 = 360, 4 bytes, 360 overall:
1; ld      l,(hl)
2; ld      h,$00
1; push    hl

68 x 5 = 340, 5 bytes, 340 overall:
2; ld      h,$00
1; dec     sp
1; ld      a,l
1; pop     hl

74 x 4 = 296, 4 bytes, 296 overall:
1; dec     sp
1; ld      a,l
1; pop     hl
1; ld      l,a

14 x 21 = 294, 21 bytes, 294 overall:
3; ld      hl,$0003
1; add     hl,sp
1; push    hl
1; dec     hl
1; dec     hl
3; call    l_gint
3; call    strlen
1; pop     de
3; call    l_pint
3; ld      hl,$0005
1; add     hl,sp

73 x 4 = 292, 4 bytes, 292 overall:
1; ld      a,l
1; pop     hl
1; ld      l,a
1; push    hl

12 x 24 = 288, 24 bytes, 288 overall:
3; call    l_pint
3; ld      hl,$0003
1; add     hl,sp
1; push    hl
1; dec     hl
1; dec     hl
3; call    l_gint
3; call    strlen
1; pop     de
3; call    l_pint
3; ld      hl,$0005
1; add     hl,sp

14 x 20 = 280, 20 bytes, 280 overall:
3; ld      hl,$0003
1; add     hl,sp
1; push    hl
1; dec     hl
1; dec     hl
3; call    l_gint
3; call    strlen
1; pop     de
3; call    l_pint
3; ld      hl,$0005

12 x 23 = 276, 23 bytes, 276 overall:
3; call    l_pint
3; ld      hl,$0003
1; add     hl,sp
1; push    hl
1; dec     hl
1; dec     hl
3; call    l_gint
3; call    strlen
1; pop     de
3; call    l_pint
3; ld      hl,$0005

55 x 5 = 275, 5 bytes, 275 overall:
1; add     hl,sp
1; ld      l,(hl)
2; ld      h,$00
1; push    hl

68 x 4 = 272, 4 bytes, 272 overall:
2; ld      h,$00
1; dec     sp
1; ld      a,l

14 x 18 = 252, 18 bytes, 252 overall:
1; add     hl,sp
1; push    hl
1; dec     hl
1; dec     hl
3; call    l_gint
3; call    strlen
1; pop     de
3; call    l_pint
3; ld      hl,$0005
1; add     hl,sp

12 x 20 = 240, 20 bytes, 240 overall:
3; call    l_pint
3; ld      hl,$0003
1; add     hl,sp
1; push    hl
1; dec     hl
1; dec     hl
3; call    l_gint
3; call    strlen
1; pop     de
3; call    l_pint

34 x 7 = 238, 7 bytes, 238 overall:
3; ld      hl,$0002
1; add     hl,sp
1; ld      l,(hl)
2; ld      h,$00

14 x 17 = 238, 17 bytes, 238 overall:
3; ld      hl,$0003
1; add     hl,sp
1; push    hl
1; dec     hl
1; dec     hl
3; call    l_gint
3; call    strlen
1; pop     de
3; call    l_pint

14 x 17 = 238, 17 bytes, 238 overall:
1; add     hl,sp
1; push    hl
1; dec     hl
1; dec     hl
3; call    l_gint
3; call    strlen
1; pop     de
3; call    l_pint
3; ld      hl,$0005

14 x 17 = 238, 17 bytes, 238 overall:
1; push    hl
1; dec     hl
1; dec     hl
3; call    l_gint
3; call    strlen
1; pop     de
3; call    l_pint
3; ld      hl,$0005
1; add     hl,sp

57 x 4 = 228, 4 bytes, 228 overall:
1; ex      de,hl
1; and     a
2; sbc     hl,de

14 x 16 = 224, 16 bytes, 224 overall:
1; push    hl
1; dec     hl
1; dec     hl
3; call    l_gint
3; call    strlen
1; pop     de
3; call    l_pint
3; ld      hl,$0005

```

For example, I see right away that 
```
3; call    l_gint
3; call    strlen
```
can be replaced by having a prefix strlen_l_gint: call l_gint before strlen to save 3 bytes on each of those

zx70, 2023-02-04:
I know this it tempting  :)
The problem is that this approach tend to help in reducing the size of big programs but may affect the size of the small ones.    Moreover it usually work on specific code sets while other programming styles would require different optimizations. 
Last but not least  we often bumped into unexpected side effects, e.g. different status of some obscure flag, a relative jump shift...  they are mostly unpredictable and show up only when an issue is opened. 
I suggest to test the optimization on few samples before opening a pull request. 
Another possible approach is to keep an optimization set local or to provide is as an option, like here:

https://github.com/z88dk/z88dk/tree/master/examples/clisp

feilipu, 2023-05-26:
With `sccz80` I often see long `pop bc` and `pop af` sequences for [cleaning up the stack](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/math/float/am9511/c/8085/am9511_fmax.asm). I guess there's no other way to do this if `de` and `hl` are to be preserved for return.

suborb, 2023-05-26:
> With `sccz80` I often see long `pop bc` and `pop af` sequences for [cleaning up the stack](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/math/float/am9511/c/8085/am9511_fmax.asm). I guess there's no other way to do this if `de` and `hl` are to be preserved for return.

8085? I think on z80 it'll use the exx set to adjust the stack if it's more than 6 words on the stack.

closed 2024-05-10 by suborb