z88dk/z88dk #2179: non-z80 - optimisation rules

feilipu, 2023-01-27:
I'm looking at this 8085 (and also gbz80) optimisation question.

For reference, I find this situation.

- z80_rules.9 is always run

- -01 - z80_rules.1 is run

- -02 - z80_rules.2 then z80_rules.1

- -03 - z80_rules.2 then z80_rules.1 then z80_rules.0

- --opt-code-speed - z80_rules.8 is always run last (except arch file) irrespective of Ox.

- -m8085 - 8085_rules.1 is always run last.

For gbz80 there are lots of rules using `ex de,hl` which are not excluding gbz80. Which I guess means that sccz80 will not generate `ex de,hl` for gbz80, otherwise things would be broken.

To Do @feilipu: If that is the case, then the few remaining rules `%notcpu gbz80` protection for `ex de,hl` can be removed (noting there are other opcodes that it also can't do).

For 8085 the `--opt-code-speed` `z80_rules.8` rules remove all of the potential `l_gint` `l_gchar` improvements as it is run before the architecture specific file.

To Do @suborb: This should to be changed, so that "speed" optimisation `rules.8` is run last, or the `--opt-code-speed` optimisation file is associated with architecture and run last.

feilipu, 2023-01-27:
> This should to be changed, so that "speed" optimisation rules.8 is run after the architecture file, or the --opt-code-speed optimisation file is associated per architecture and also run last.

It looks like zcc is already set up to handle an additional CPU related optimisation step, before speed optimisation. I'll have a look at putting the arch files to run there as CPU optimisation, which solves this ordering issue perfectly.

It also leaves the user optimisation step open too.

feilipu, 2023-01-28:
I think for the gbz80 there is an opportunity for adding `LD HL,SP+int8` instruction optimisation, which at 12 ticks would be faster than `LD HL,int16` `ADD HL,SP` with 20 ticks.

Also `ADD SP,int8` at 16 ticks is still better than `LD HL,int16` `ADD HL,SP` `LD SP,HL` with 28 ticks.

feilipu, 2023-01-30:
If the gbz80 `add sp+int8` is really an integer, then it would be worth changing the boundaries on the rule checks to allow negative number results, -128 through 127.

This would be useful for establishing the stack frame, if for no other reason.

suborb, 2023-01-30:
I thought that bit was already done by sccz80?

feilipu, 2023-01-30:
IDK whether it is done by sccz80, but the rules are all definitely `%check 0 <= %1 <= 127` which, if they're being used at all, should be changed to match match %check -128 <= %1 <= 127` (if `init8` is actually a true fact, which I don't know).

suborb, 2023-01-30:
> (if `int8` is actually a true fact, which I can't confirm).

I think we've implemented it as signed and rgbds docs indicate it as signed. Though `ld hl,sp+dd` would be probably be more useful as unsigned.

So yes, go ahead and change it to be correct.

commit 8140c290 (2023-01-30): sccz80 gbz80 rule range - closes #2179
closed 2023-01-30 by feilipu