z88dk/z88dk #2186: Test current sdcc (before 4.3.0 release)

spth, 2023-02-04:
SDCC developers intend to do a 4.3.0 release soon. While the official release process with code freezes, etc hasn't been started yet, this seems like a good time to test if z88dk works well with SDCC from current svn, and report any found SDCC issues upstream.

feilipu, 2023-02-06:
The only thing I'm noticing is that `SECTION` statements before each function seem to be omitted where they are redundant.

That might be effective, but for human readability it might actually be better to reiterate the function `SECTION` to save scrolling back through a thousand lines to find the last  `SECTION code_compiler` statement. Just a thought.

feilipu, 2023-02-07:
@spth in our builds we're using the actual executable for `sdcpp`, found here `support/cpp/gcc/cpp`, rather than the script found as `sdcc/bin/sdcpp` which calls that executable.

I guess something is getting confused, because the C preprocessor is then asking for a `cc1` command to follow, which produces following [build test error](https://github.com/z88dk/z88dk/actions/runs/4106825152/jobs/7085540277).

`z88dk-zsdcpp: fatal error: cannot execute 'cc1': execvp: No such file or directory`

Can you give a bit of background as to why sdcc decided not to put the actual executables in the `sdcc/bin` directory, but rather execute from the compilation directories?

And, how to avoid this issue above?

spth, 2023-02-09:
I can't give much background on why it was done this way when the upgrade from the GCC 4-based preprocessor to the GCC 12-based one happened. @felix-salfelder can.

felix-salfelder, 2023-02-09:
> Can you give a bit of background as to why sdcc decided not to put the actual executables in the sdcc/bin directory, but rather execute from the compilation directories?

I can give it a go.

First of all, there is no good reason for storing compilation output in a nonstandard place. And then it caused problems during the cpp update, and that's why I had to revisit it.

I put the wrapper script into $builddir/bin to provide some amount of backwards compatibilty: It works if you don't change the build output prior to running $builddir/bin/sdcpp. Much more than that is hard to specify and likely difficult to maintain or even guarantee.

To avoid the issue (and similar future issues), consider to run "make install" -- it will create the sdcpp executable that you are looking for in the place where you should be looking for it. C.f. the 'install' target [1] in the GNU Makefile conventions and its purpose.

PS0: I am aware of the bug in sdcc/src/clean.mk (see mailing list). I will fix it within the next days.
PS1: I am also aware that the PATH to cc1 should be hardwired somewhere in the binaries, but isn't.

[1] https://www.gnu.org/prep/standards/html_node/Standard-Targets.html#Standard-Targets

feilipu, 2023-02-11:
@felix-salfelder 
> PS1: I am also aware that the PATH to cc1 should be hardwired somewhere in the binaries, but isn't.

Could you please enlighten as to why there is a `cc1` program necessary now, where it wasn't in the past for `sdcpp`?

felix-salfelder, 2023-02-11:
On Sat, Feb 11, 2023 at 02:39:58AM -0800, Phillip Stevens wrote:
> Could you please enlighten as to why there is a `cc1` program
> necessary now, where it wasn't in the past for `sdcpp`?

The split into cpp & cc1 is how gcc does it (I won't speculate as to
why). The cpp executable is the basis for sdcpp. The more we deviate
from gcc, the harder it will be to follow them. So we decided to leave
it like that (This may change, don't depend on cc1!).

Why it wasn't there in the past... Maybe cc1 did not exist in the early
days. Somehow sdcc/cpp ended up as a mix from various gcc versions.

Maybe we should only use cc1, and merge the rest of cpp into sdcc. Or
just depend on gcc? There will be side effects no matter how we do it.

HTH
felix

feilipu, 2023-02-11:
> Somehow sdcc/cpp ended up as a mix from various gcc versions... There will be side effects no matter how we do it.

Sounds like this is all still to be resolved. So imho we'll just wait until it is done.

Z88dk doesn't seem to have any issues with r13854 `sdcc`, so that looks clean to go forward.

And the `sdcpp` solution hopefully will be a single binary that can be renamed as desired, and doesn't have hard or soft links to other stages (aka behaviour as it is in v4.2.0).

feilipu, 2023-02-19:
Watching these SDCC binutils configuration or build bugs [#3491](https://sourceforge.net/p/sdcc/bugs/3491/) and [#3555](https://sourceforge.net/p/sdcc/bugs/3555/) as they affect our build.

feilipu, 2023-03-31:
See #2216 for `__smallc` issue.

spth, 2023-05-24:
> 
> We'll move to using `ucpp` for the sdcc compiler from Release 4.3.0.
> 
> The [`ucpp` preprocessor](https://github.com/z88dk/z88dk/tree/master/src/ucpp) is used for `sccz80` too, so using a single preprocessor solution for both compilers will make things simpler for us. So we are no longer dependent on the integration path for `sdccp`.

On the SDCC side, migrating sdcpp to a current upstream cpp has given us some troubles. cpp is now more tightly integrated into gcc than before. We are considering moving to a Wave-based sdcpp after the 4.3.0 release. SDCC needs a modern (i.e. C2X, with a perspective towards C2Y) preprocessor; I don't think ucpp would do for us (it is C99 only). On the other hand, the preprocessors from GCC and clang are too tightly integrated into their respective projects, making it hard for us to maintain our fork.
Given that Wave is meant to the base for custom C/C++ preprocessors, and is in boost (a library widely available), it looks like a good choice for us (unfortunately, it was unmaintained and C99 only at the time we decided to go for newer cpp from GCC, but that changed since).

closed 2023-07-20 by suborb